1technophile/OpenMQTTGateway #1519: Not transmitting with Heltec 433 (HTIT-WS)

ryanm101, 2023-03-10:
I've installed oMG using the prebuilt binaries. 

publishing `omgheltec433/OpenMQTTGateway_heltec_rtl_433_ESP/commands/MQTTto433` `{"value": 2573456787}` does nothing.

I am seeing other traffic arriving on `omgheltec433/OpenMQTTGateway_heltec_rtl_433_ESP/RTL_433toMQTT` so i know receive is working. 

honestly not sure how to troubleshoot this any help appreciated. 
Doorbell capture attached just in case it helps. 
[g004_433.92M_250k.zip](https://github.com/1technophile/OpenMQTTGateway/files/10945689/g004_433.92M_250k.zip)

NorthernMan54, 2023-03-10:
@ryanm101 With the SX127X chipset ( used by the Heltec and LilyGo boards ) as you found transmitting is not supported, also the RF, RF2 and PiLight ( MQTTto433) modules are not supported.  The support was brought in by the rtl_433_ESP project as they where looking for all in one boards ( no soldering required ), and at this time is not integrated into the other RF receiver modules.

ryanm101, 2023-03-11:
Ahh damn it.. Thanks, that would explain why i'm banging my head against a wall. Might be a nice feature to add a compatibility  table to show which boards are RX only vs TX/RX..

NorthernMan54, 2023-03-16:
The CC1101 Module is support for both receive and transmit.

Also the next release will include updated documentation for the various modules and what is supported

1technophile, 2023-04-09:
Compatibility table added

closed 2023-04-09 by 1technophile