1technophile/OpenMQTTGateway #1529: Enable BLE and 433 for Boards like the LilyGo

Andyriod, 2023-03-13:
The lilygo-rtl_433 has both BLE and 433 on board.
After the PR from @SmittyHalibut there is no compile problem anymore, but I tried to enable both in environments.ini and it doesn't work. It only publishes BT-Deveices and is in some kind of bootloop.

Here is what I did:
    [env:lilygo-rtl_433]
    platform = ${com.esp32_platform}
    board = ttgo-lora32-v21
    ; ~/.platformio/packages/framework-arduinoespressif32/variants/.../pins_arduino.h
    board_build.partitions = min_spiffs.csv
    lib_deps =
      ${com-esp.lib_deps}
      ${libraries.wifimanager32}
      ${libraries.ssd1306}
      ${libraries.rtl_433_ESP}
      ${libraries.ble}
      ${libraries.decoder}
    build_flags =
      ${com-esp.build_flags}
    ; *** OpenMQTTGateway Config ***
      ;'-UZmqttDiscovery'          ; disables MQTT Discovery
      '-DvalueAsATopic=true'    ; MQTT topic includes model and device
      '-DGateway_Name="OpenMQTTGateway_lilygo_rtl_433_ESP"'
    ; *** OpenMQTTGateway Modules ***
      '-DZgatewayRTL_433="rtl_433"'
      '-DZradioSX127x="SX127x"'
      '-DZgatewayBT="BT"'
    ; *** ssd1306 Display Options ***
      '-DZdisplaySSD1306="LilyGo_SSD1306"'
    ;  '-DLOG_TO_OLED=true'         ; Enable log to OLED
    ;  '-DJSON_TO_OLED=true'
    ;  '-DLOG_LEVEL_OLED=LOG_LEVEL_NOTICE'
    ;  '-DDISPLAY_IDLE_LOGO=false'
    ;  '-DDISPLAY_BRIGHTNESS=80'
    ;  '-DDISPLAY_METRIC=false'
    custom_description = For ESP32, Gateway using RTL_433_ESP and RadioLib
    custom_hardware = ESP32 LILYGO LoRa32 V2.1

1technophile, 2023-03-13:
We are working on it below:
https://github.com/1technophile/OpenMQTTGateway/tree/heltec-bt-rtl-env

Not finished yet

SmittyHalibut, 2023-03-13:
I think what you're seeing is https://github.com/NorthernMan54/rtl_433_ESP/pull/62

Apply this change to OpenMQTTGateway (to use my branch of rtl_433_ESP instead) and try again, tell me if its still happening.

```
[13:49:08] mark@crossthread:~/src/OpenMQTTGateway$ git diff
diff --git a/platformio.ini b/platformio.ini
index 7b99171..b48c222 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -145,7 +145,8 @@ smartrc-cc1101-driver-lib = SmartRC-CC1101-Driver-Lib@2.5.7
 stl = https://github.com/mike-matera/ArduinoSTL.git#7411816
 shtc3 = https://github.com/sparkfun/SparkFun_SHTC3_Arduino_Library
 somfy_remote=Somfy_Remote_Lib@0.3.0
-rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.1.6
+;rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.1.6
+rtl_433_ESP = https://github.com/SmittyHalibut/rtl_433_ESP.git
 emodbus =  miq19/eModbus@1.0.0
 gfSunInverter = https://github.com/BlackSmith/GFSunInverter.git#v1.0.1
 decoder = https://github.com/theengs/decoder.git
[13:49:11] mark@crossthread:~/src/OpenMQTTGateway$ 
```

NOTE: This is only a temporary change until rtl_433_ESP accepts my PR, and OpenMQTTGateway pulls in a new version of rtp_433_ESP that has my change.  I will NOT be working to keep my branch up-to-date with upstream (at least, not reliably), so this is most definitely a compromise.  Hopefully my PR to rtl_433_ESP will be accepted soon, and OpenMQTTGateway will bump to the new version of rtl_433_ESP soon after that.

SmittyHalibut, 2023-03-13:
Thanks to @NorthernMan54 it looks like https://github.com/NorthernMan54/rtl_433_ESP/pull/62 has been accepted and will be released soon.  🥳

wastlmuc, 2023-12-13:
Possibly getting a bit off-topic, but are there plans to combine ["RF","RF2","BT","rtl_433"] in an official package for ESP. This setup is working fine for me, since I used above https://github.com/NorthernMan54/rtl_433_ESP/pull/62.
Thanks.

NorthernMan54, 2023-12-13:
Spent a fair bit of time and effort on this, and eventually abandoned the effort as getting something that was stable due to resource constraints between the various packages.

palto42, 2024-11-16:
Is anyone still working on this or will this not be solved due to technical constraints?

NorthernMan54, 2024-11-16:
@palto42 no nothing has changed, the boards have insufficient ram resources.