1technophile/OpenMQTTGateway #1539: Help needed debugging RCSwitch protocol

redahb, 2023-03-16:
I have a couple of Chuango DWC-102 door sensors. They are a few years old and they have had several revisions. There's a DWC-102 with board revision 2.0 that only sends 'open' signals, and there's a DWC-102 with board revision 4.1 that also sends a 'closed' signal, along with a 'low battery' signal. But apparently, there are also differences between revision 4.1 boards, let's call them 'old' and 'new'.

I'm using a **ESP32** with **CC1101** and **esp32dev-multi_receiver**, hooked it up to a Serial Monitor for debugging.

### Old 4.1 boards
Working fine with RCSwitch module. When openend, the signal looks like this:
```
T: Hey I got a callback openmqttgw/OpenMQTTGateway_multi_receiver/433toMQTT
T: isAdupl?
T: no pub. dupl
N: [ MQTT->OMG ]: {"value":16156202,"protocol":1,"length":24,"delay":541,"tre_state":"-","binary":"111101101000011000101010","raw":"16771,1644,619,1595,622,1596,603,1623,583,549,1669,1629,583,1630,587,524,1668,1633,587,557,1671,560,1649,561,1651,560,1651,1637,561,1659,562,553,1649,583,1647,560,1650,1638,590,560,1647,1636,567,581,1623,1659,561,554,1620,","mhz":433.92}
```
When closed, the signal looks like this:
```
T: Hey I got a callback openmqttgw/OpenMQTTGateway_multi_receiver/433toMQTT
T: isAdupl?
T: no pub. dupl
N: [ MQTT->OMG ]: {"value":7767596,"protocol":11,"length":23,"delay":820,"tre_state":"-","binary":"11101101000011000101100","raw":"4350,1646,596,1617,599,1621,579,1648,558,550,1668,1629,583,1628,560,551,1669,1632,587,557,1669,561,1645,563,1646,558,1650,1636,563,1634,587,553,1671,533,1670,561,1649,1639,589,559,1649,1636,561,1633,585,554,1645,","mhz":433.92}
```

### New 4.1 boards
Not working with RCSwitch module (tested 4 devices with same results). When openend, the signal looks like this:
```
T: Hey I got a callback openmqttgw/OpenMQTTGateway_multi_receiver/433toMQTT
T: isAdupl?
T: no pub. dupl
N: [ MQTT->OMG ]: {"value":3606775,"protocol":1,"length":24,"delay":543,"tre_state":"-","binary":"001101110000100011110111","raw":"16844,577,1669,560,1672,1642,564,1662,581,536,1676,1637,581,1636,556,1664,583,560,1654,584,1657,556,1662,556,1658,1643,593,559,1656,564,1658,560,1656,1661,575,1656,570,1641,594,1643,578,546,1672,1629,585,1653,561,1629,558,","mhz":433.92}
```
However, the 'close' signal is not processed (usually). It is sent by the DWC-102 (red led flashes), it is received by the ESP32 (red led flashes) but the signal is not processed. With the **old 4.1** boards, _open/close is picked up immediately_ by the ESP32 and OMG. WIth the **new 4.1 boards**, _open is picked up immediately_ but close (almost) never. In 30 tests, the signal was picked up only once and looked like this:
```
T: Hey I got a callback openmqttgw/OpenMQTTGateway_multi_receiver/433toMQTT
T: isAdupl?
T: no pub. dupl
N: [ MQTT->OMG ]: {"value":3606780,"protocol":24,"length":24,"delay":1217,"tre_state":"-","binary":"001101110000100011111100","raw":"4378,544,1648,560,1674,1635,589,1637,579,560,1657,1637,590,1639,557,1663,583,562,1655,561,1656,584,1658,557,1659,1640,570,562,1644,570,1666,569,1651,1645,580,1650,585,1653,563,1634,587,1656,559,1657,564,554,1647,582,1625,","mhz":433.92}
```

So it looks like the new 4.1 boards use a different protocol for sending their 'close' signal. The main problem is that the RCSwitch module almost never picks up this signal and I don't know why. Serial monitor with verbose logging did not give more information.

### Signals ARE received with RTL_433 module
...But wrong. Because I know the signal is sent, and I know the ESP32 receives the signal, I wanted to try another module. So I switched OMG to RTL_433 instead of RF (RCSwitch), and this is what happens when I close the contact on the sensor:

1. Signal  is received _immediately_
2. RTL_433 doesn't seem to know what to do with it:
```
N: uniqueid: Smoke-GS558-30855
T: isAdupl?
N: Send on /RTL_433toMQTT msg {"model":"Smoke-GS558","id":30855,"unit":12,"learn":0,"code":"3f10ec","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-41,"duration":753996}
T: jsonPubl - ON
T: [ OMG->MQTT ] topic: openmqttgw/OpenMQTTGateway_multi_receiver/RTL_433toMQTT msg: {"model":"Smoke-GS558","id":30855,"unit":12,"learn":0,"code":"3f10ec","protocol":"Wireless Smoke and Heat Detector GS 558","rssi":-41,"duration":753996} 
N: uniqueid: Hyundai-WS-4-3
T: isAdupl?
N: Send on /RTL_433toMQTT msg {"model":"Hyundai-WS","id":3,"channel":4,"battery_ok":0,"temperature_C":-88.1,"button":1,"protocol":"Hyundai WS SENZOR Remote Temperature Sensor","rssi":-41,"duration":753996}
T: jsonPubl - ON
T: [ OMG->MQTT ] topic: openmqttgw/OpenMQTTGateway_multi_receiver/RTL_433toMQTT msg: {"model":"Hyundai-WS","id":3,"channel":4,"battery_ok":0,"temperature_C":-88.1,"button":1,"protocol":"Hyundai WS SENZOR Remote Temperature Sensor","rssi":-41,"duration":753996}
```

And, well, it's definitely not a Smoke Sensor or Weather Station. It's not another device that's picked up either, these two devices show up in RTL_433 _immediately_ when I close the DWC-102 contact.

### Question:
How can I debug this further so I can make the RF module (RCSwitch) accept/decode this signal?

NorthernMan54, 2023-03-16:
@redahb The RFLINK regular decode is probably the best to look at to determine why the new board is not working.  When I quickly looked at it, it appears the manufacturer changed the signal modulation dramatically hence the issue.

The Open or working signal is a serial of pulses, either ~500uSec or ~1500uSec ( 0's and 1's ), and end of row marker ~6990uSec ( next line ).  As a signal typically repeats ( repeating rows ) several times during a transmission.

The Close signal is similar, but is missing the end of row marker ~6990uSec ( next line ).

I'm wondering if these are defective from the manufacturer, or if the original application doesn't use the Close message and the manufacturer doesn't care.

redahb, 2023-03-17:
@NorthernMan54 thanks for your interest and input.

First of all: No, the devices are not defective. But you're right in that the original application doesn't use the Close message so the manufacturer probably doesn't care. Strange though they added it in later revision boards. Perhaps at one point they did want to implement it somewhere, but abandoned the idea.

Anyway, I have about 8 of these devices, 4 'old' boards and 4 'new' boards, and I checked the raw signal of the old boards as well. Here is one example:

```
20;04;DEBUG;Pulses=511;Pulses(uSec)=1620,480,450,1560,450,1560,450,1560,450,1560,1560,480,450,1560,450,1560,1530,480,1530,480,1530,480,1560,480,1560,480,450,1560,1560,480,450,1590,1530,480,1560,480,1560,480,1560,480,1530,480,1530,480,450,1560,450,1530,450,4290,1560,480,450,1560,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,1560,480,1530,480,1530,480,1530,480,1560,480,450,1560,1530,480,450,1590,1530,510,1530,480,1560,480,1560,480,1530,480,1530,480,450,1560,450,1530,450,4290,1530,480,450,1560,450,1560,450,1560,450,1560,1560,480,450,1560,450,1560,1560,480,1530,480,1530,480,1560,480,1530,480,450,1560,1560,480,450,1590,1530,480,1530,480,1530,480,1560,480,1560,480,1530,480,450,1560,450,1560,450,4290,1530,510,450,1560,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,1560,480,1560,480,1530,480,1530,480,1530,480,450,1560,1560,480,450,1590,1530,480,1530,480,1530,480,1560,480,1560,480,1560,480,450,1560,450,1560,450,4290,1530,510,450,1560,450,1560,450,1560,480,1560,1530,480,450,1560,450,1590,1560,480,1560,480,1560,480,1530,480,1530,480,450,1560,1560,480,450,1590,1530,480,1530,480,1530,480,1530,480,1560,480,1560,480,450,1560,450,1560,450,4290,1560,480,450,1590,450,1560,450,1560,450,1560,1530,480,450,1560,450,1590,1530,480,1560,480,1560,480,1530,480,1530,480,450,1560,1530,480,450,1560,1560,480,1530,480,1530,480,1530,480,1530,480,1560,480,450,1560,450,1560,450,4290,1530,480,450,1590,450,1560,450,1560,450,1560,1560,480,450,1560,450,1590,1530,480,1530,480,1560,480,1560,480,1530,480,450,1560,1530,480,450,1560,1560,480,1560,480,1530,480,1560,480,1530,480,1560,480,450,1560,450,1530,450,4290,1560,480,450,1590,450,1560,450,1560,450,1560,1560,480,450,1560,450,1590,1530,480,1530,480,1560,480,1560,480,1560,480,450,1560,1530,480,450,1590,1560,480,1560,480,1560,480,1530,480,1560,480,1530,480,450,1560,450,1530,450,4290,1560,480,450,1560,450,1590,450,1560,450,1560,1560,480,450,1560,450,1590,1530,510,1530,480,1530,480,1560,480,1560,480,450,1560,1560,480,450,1590,1530,480,1560,480,1560,480,1530,480,1530,480,1560,480,450,1560,450,1530,450,4290,1560,480,450,1560,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,1560,480,1530,480,1560,480,1530,480,1560,480,450,1560,1560,480,450,1590,1530,510,1530,480,1560,480,1560,480,1530,480,1530,480,450,1560,450,1530,450,4290,1530,480,450,1560,450,1560,450,1560,450,1560,1560;
```
Again, this is received and decoded by RCSwitch without issue. Even though these old boards also don't have an end of row marker in the 'close' signal. For comparison, this is the raw output of the new boards again:

```
20;0C;DEBUG;Pulses=511;Pulses(uSec)=540,1590,450,1590,1530,480,1560,480,480,1560,1530,480,1530,480,1530,480,450,1560,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,480,1560,1560,480,1530,480,1530,480,1530,480,1560,480,1560,480,450,1560,450,1560,450,4290,450,1590,450,1590,1530,480,1560,480,480,1560,1560,480,1530,480,1530,510,450,1560,450,1560,450,1560,480,1560,1530,480,450,1560,450,1560,480,1590,1560,480,1530,480,1530,480,1530,480,1530,480,1560,480,480,1560,480,1530,450,4290,450,1560,450,1590,1530,480,1530,480,450,1560,1560,480,1530,480,1530,510,450,1590,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,450,1590,1560,480,1560,480,1530,480,1530,480,1530,480,1530,480,450,1560,480,1530,450,4290,450,1560,450,1590,1530,480,1530,480,450,1560,1560,480,1560,480,1530,510,450,1590,450,1590,450,1560,450,1560,1560,480,450,1560,450,1560,450,1560,1560,480,1560,480,1560,480,1530,480,1530,480,1530,480,450,1560,450,1530,450,4290,450,1560,450,1560,1530,480,1530,480,450,1560,1560,480,1560,480,1560,480,450,1590,450,1590,450,1560,450,1560,1560,480,480,1560,450,1560,450,1590,1530,480,1560,480,1560,480,1530,480,1560,480,1530,480,450,1560,450,1530,480,4290,450,1560,450,1560,1560,480,1530,480,450,1560,1530,480,1560,480,1560,480,450,1560,450,1590,450,1560,450,1560,1560,480,450,1560,480,1560,450,1590,1530,480,1530,480,1560,480,1560,480,1530,480,1530,480,450,1560,450,1530,450,4290,450,1560,450,1560,1560,480,1530,480,450,1560,1530,480,1530,480,1560,480,450,1560,450,1590,450,1560,450,1560,1530,480,450,1560,480,1560,450,1590,1530,480,1530,480,1530,480,1560,480,1560,480,1530,480,450,1560,450,1560,450,4290,450,1590,450,1560,1560,480,1560,480,450,1560,1530,480,1530,480,1530,480,450,1560,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,480,1560,1530,480,1530,480,1530,480,1560,480,1560,480,1560,480,450,1560,450,1560,450,4290,450,1590,450,1560,1560,480,1560,480,450,1560,1530,480,1530,480,1530,480,450,1560,450,1560,480,1560,450,1560,1530,480,450,1560,450,1560,450,1560,1560,480,1530,480,1530,480,1530,480,1560,480,1560,480,480,1560,450,1560,450,4290,450,1590,450,1590,1530,480,1560,480,450,1560,1530,480,1530,480,1530,510,450,1560,450,1560,450,1560,450,1560,1530,480,450,1560,450,1560,450,1590,1560,480,1530,480,1530,480,1530,480,1560,480,1560,480,450,1560,450,1560,450,4290,450,1590,450,1590,1530,480,1530,480,450,1560,1560;
```

If you or anyone could point me in the right direction to solve this, it'd be much appreciated.

redahb, 2023-03-21:
So, some more information I've found. The signal is always 50 pulses, and these are the ones I have found now:

**Old sensor OPEN**, ending with 6990usec EOR:
```
540,1590,450,1560,1560,480,1560,480,450,1560,1530,480,1560,480,1560,480,480,1560,480,1560,450,1560,450,1560,1560,
480,450,1560,480,1560,480,1590,1530,480,1560,480,1530,480,1560,480,450,1560,1560,480,1560,480,1560,450,450,6990
```
**Old sensor CLOSE**, ending with 4290usec EOR:
```
1620,480,1560,480,1560,480,1530,480,450,1560,1530,480,1530,480,450,1560,1530,480,450,1560,450,1560,450,1560,450,
1560,1560,480,1530,480,450,1560,450,1590,450,1560,1530,480,450,1560,1560,480,1560,480,450,1560,450,1560,450,4290
```
**New sensor OPEN**, ending with 6990usec EOR:
```
540,1590,450,1590,1560,480,1560,480,1560,480,450,1590,450,1560,1560,510,1560,480,480,1590,450,1590,1560,480,1560,
480,1560,480,1560,480,1560,510,1560,510,1560,510,1560,480,1560,480,1560,480,480,1560,1560,480,450,1560,450,6990
```
**New sensor CLOSE**, ending with 4320usec EOR:
```
480,1590,480,1590,1560,480,1560,480,1560,480,450,1560,480,1560,1560,480,1560,510,450,1590,450,1590,1560,480,1560,
480,1560,480,1560,480,1560,510,1560,480,1560,480,1560,480,1560,480,1560,480,1560,480,450,1560,480,1560,480,4320
```
I've converted the signals to binary and was comparing the old and new close signals, untill I noticed the _new close_ signal is way more similar with the _new open_ signal than with the old close signal:

[image]
_(I've marked all similar pieces in the signal)_

Now according to RC-Switch both the old and new open signals are using protocol 1, which in [rcswitch.cpp](https://github.com/1technophile/rc-switch/blob/development/src/RCSwitch.cpp) is written as:
```
{ 350,  0, { 0, 0 }, 1, {  1, 31 }, { 1,  3 }, { 3, 1 }, false,  0 },  // 01 (Princeton, PT-2240)
```

Whereas the old close signal is using protocol 11, which is:
```
{ 500,  2, { 3, 3 }, 0, {  0,  0 }, { 1,  2 }, { 2, 1 }, false, 37 },  // 11 (Nice_FloR-S 52bit)
```
I understand why the open and close signals from the old sensors are picked up by different protocols in RC-Switch, as the binary is vastly different.

I fail to understand however, why the protocol 1 isn't picking up both the open and close signals from the new sensors. They look extremely similar to me.

Any suggestions are greatly appreciated.

redahb, 2023-03-22:
If anyone is interested, I've finally figured it out. These protocols work for any of my Chuango devices:

```
  { 560,  0, {  0,  0 },  1, {  1, 30 }, {  1,  3 }, {  3,  1 }, false, 12 }, //  Chuango old protocol
  { 560,  0, {  0,  0 },  1, {  1,  8 }, {  1,  3 }, {  3,  1 }, false,  7 }, //  Chuango new protocol
```
I'll see if I can make a PR somewhere in the near future!

closed 2023-03-22 by redahb