1technophile/OpenMQTTGateway #1564: lilygo-rtl_433 does not try to connect to broker (but dev version does)

rodpayne, 2023-03-23:
**Describe the bug**
`lilygo-rtl_433` does not try to connect to MQTT broker on a brand new `LILYGO T3 v1.6.1 20210104`.  After configuring using the Wifi Manager portal, the last thing that the display says is "Wifi connected", including after resets.

I tried uploading a couple of times (Option 1), with Erase Flash checked, without luck.

**Uploaded from Dev https://docs.openmqttgateway.com/dev/upload/web-install.html (version 85d649) and the device began operating.  (And it is wonderful.)**

To verify that Dev helped, I uploaded v1.4.0 again and the device stopped again.  Uploaded the Dev version again and it is working.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload lilygo-rtl_433 using Option 1.
2. Use Wifi Manager portal to set Wifi and MQTT parameters.
3. Verify that it stops with last message = "Wifi connected".

**Expected behavior**
"Wifi connected" should be followed by "MQTT connected".

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (v1.4.0, and 85d649 development)
* Library version related to the problem you have (lilygo-rtl_433)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

1technophile, 2023-03-23:
Hello,

You may wait for tomorrow and the dev version will not have anymore this error:
`E (3591493) gpio: gpio_set_level(226): GPIO output gpio_num error`

Also we should release soon

rodpayne, 2023-03-30:
Version v1.5.0 is connecting fine.

closed 2023-03-30 by rodpayne