1technophile/OpenMQTTGateway #1582: lilygo_rtl_433 doesn' recognize Auriol HG02832 protocol (16)

fenyvesi, 2023-03-29:
**Describe the bug**
An otherwise working v1.5 version doesn't recognize the Auriol HG02832 rain sensor protocol (16). The SDR version does, the protocol number there is 200. I compared the 2 source files (auriol_4ld5661.c) and they seems to be identical except of calculating the volume of rain.

**Environment (please complete the following information):**
* OpenMQTTGateway v1.5 burnt in from Web and compiled locally, the same.

**Additional context**
The SDR output:
time      : 2023-03-29 08:15:39
Model     : Auriol-4LD5661                         ID        : d8
Battery OK: 1            Temperature: 15.9 C       Rain      : 151.0 mm

NorthernMan54, 2023-03-29:
The increase in the number of false positive signals, is due to the receiver tuning we did to improve the overall sensitivity of the receiver.

Did the previous version receive ( v1.4.0 ) receive it ?

fenyvesi, 2023-03-29:
Unfortunately I don't know. I tried twice to upload the v1.4 precompiled files locally, but it was unsuccessfull. There was a boot loop. All the files and adresses was the same you proposed. The flash_download tool was different compared to the one you had used.
Advice?
Thanks.

NorthernMan54, 2023-03-29:
One approach to downgrade the release from a platformio build environment is to change the rtl_433_ESP library reference to the version included within v1.4.

v1.4.0
rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.1.6

v1.5.0
rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.1.9

I don't think this changed, but it is worth checking.

Have you tried moving the location of your OMG device, we have found that the reception distance is roughly 50% of what a rtl_sdr is able to achieve.

fenyvesi, 2023-03-29:
Good idea. I will try the local platformio solution with v1.4.  The sensitivity was not a problem now, because the sensor is next to the ESP32 now. I'll come back with the results.

fenyvesi, 2023-03-29:
I copied the v1.4 in the  platformio project folder. Compiled and uploaded. No WIFI AP, the program can't connect to WIFI.

environment.ini:

> [env:lilygo-rtl_433]
platform = ${com.esp32_platform}
board = ttgo-lora32-v21
; ~/.platformio/packages/framework-arduinoespressif32/variants/.../pins_arduino.h
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ;${libraries.ssd1306}
  ${libraries.rtl_433_ESP}
build_flags =
  ${com-esp.build_flags}
; *** OpenMQTTGateway Config ***
  ;'-UZmqttDiscovery'          ; disables MQTT Discovery
  '-DvalueAsATopic=true'    ; MQTT topic includes model and device
  '-DGateway_Name="OpenMQTTGateway_lilygo_rtl_433_ESP"'
; *** OpenMQTTGateway Modules ***
  '-DZgatewayRTL_433="rtl_433"'
  '-DZradioSX127x="SX127x"'
; *** ssd1306 Display Options ***
**;  '-DZdisplaySSD1306="LilyGo_SSD1306"'**
;  '-DLOG_TO_LCD=true'         ; Enable log to LCD
;  '-DJSON_TO_LCD=true'
;  '-DLOG_LEVEL_LCD=LOG_LEVEL_NOTICE'
;  '-DDISPLAY_IDLE_LOGO=false'
;  '-DDISPLAY_METRIC=false'
custom_description = For ESP32, Gateway using RTL_433_ESP and RadioLib
custom_hardware = ESP32 LILYGO LoRa32 V2.1

platformio.ini excerpt:
...
;default_envs = heltec-rtl_433
**default_envs = lilygo-rtl_433**
;default_envs = esp32dev-multi_receiver
....

The consol:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
N: 
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: version_tag
*wm:[2] Added Parameter: server
*wm:[2] Added Parameter: port
*wm:[2] Added Parameter: user
*wm:[2] Added Parameter: pass
*wm:[2] Added Parameter: secure
*wm:[2] Added Parameter: cert
*wm:[2] Added Parameter: name
*wm:[2] Added Parameter: topic
*wm:[2] Added Parameter: ota
N: Attempting Wifi connection with saved AP: 0
N: ZgatewayRTL_433 setup done 
N: Switching to RTL_433 Receiver: 433.92Mhz
N: OpenMQTTGateway modules: ["rtl_433"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
[E][WiFiClient.cpp:258] connect(): socket error on fd 55, errno: 104, "Connection reset by peer"
W: failure_number_mqtt: 1
W: failed, rc=-2
W: MQTT connection...
[E][WiFiClient.cpp:258] connect(): socket error on fd 56, errno: 104, "Connection reset by peer"
W: failure_number_mqtt: 2
W: failed, rc=-2

fenyvesi, 2023-03-29:
If I upload v1.5 from web and erasing the flash, then I can type the wifi and MQTT data and it connects to MQTT.

After it I upload V1.4 with platformio. No WIFI AP, no MQTT connection. How can I erase the flash in platformio?

1technophile, 2023-03-29:
Could you try to connect at 115200 with this tool, and erase with it:
https://espressif.github.io/esptool-js/

fenyvesi, 2023-03-30:
Flash cleared, v1.4 uploaded.  I managed to type the credentials, WIFI, MQTT are connected.
Some sensors (NEXUS-TH, 20 m away, -80dB) recognized, but the Auriol HG02832 rain sensor (50cm) not yet.

closed 2023-09-18 by github-actions[bot]