1technophile/OpenMQTTGateway #1585: TSL2561 setup freeze ESP boot after upgrade to 1.5.0

murfy76, 2023-04-02:
TSL2561 setup will complete freeze ESP boot after upgrade from OMG 1.4.0 to 1.5.0.  To run OMG v1.5.0, TSL2561 needs to be disabled. Same result on two boards - ESP32 and Lolin32 Lite. Thanks.

Steps to reproduce the behavior:
1. Start OMG v1.4.0 with TSL2561 - it's working 
2. Upgrade to OMG v1.5.0
3. ESP will freeze at boot with message:

N: OpenMQTTGateway Version: 1.5.0
N: WiFi ok with manual config credentials
N: Setup BME280/BMP280 on address: 0x76
N: Bosch BME280 successfully initialized: 0x60
N: Setup TSL2561 on adress:  
(freeze)

4. Connect ESP over USB, disable TSL2561, reflash - ESP will boot again  

**Expected behavior**
ESP will continue to run no matter of setup result of TSL2561.

1technophile, 2023-04-02:
Hello,

Do you have a development environement to try the newesr versions of the library?
It is located into platformio.ini

murfy76, 2023-04-04:
Helllo,

I've tried download recent development OMG version and compared platformio.ini in dev and 1.5.0 version but can't find any newer libraries.  After flash dev version ESP freeze and OTA not working anymore. Access to the ESP USB is not easy. I'll try to fix it tomorrow. Thank you.

murfy76, 2023-04-05:
Hello,

I found out that to reproduce this bug you don't need even sensor connected. Just ESP32 and TSL2561 enabled. I also tried Nodemcu and there is no problem, just on ESP32 boards. All other modules was disabled, just wifimanager and TSL2561. No compilation errors. Thanks.

Peter

--- [env:nodemcuv2-all-test] ---

************* WELCOME TO OpenMQTTGateway **************
N: WiFi ok with manual config credentials
N: Setup TSL2561 on adress: 
E: No TSL2561 detected      
N: OpenMQTTGateway modules: ["TSL2561"]
N: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
N: Connected to broker 

--- [env:esp32dev-all-test] ---

************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: 1.5.0-dev
N: WiFi ok with manual config credentials
N: Setup TSL2561 on adress: 
(freeze)

1technophile, 2023-04-05:
Try maybe to update the library version to the latest in platformio.ini:
`tsl2561 = Adafruit TSL2561@1.0.3`
to
`tsl2561 = Adafruit TSL2561@1.1.0`

If it doesn't work, try to go down the platform version for the ESP32
https://github.com/1technophile/OpenMQTTGateway/blob/bf004b078beb40e268ca8423e7e1e4fc4bf2a4a8/platformio.ini#L173
from 6.1.0 to 3.5.0

1technophile, 2023-04-05:
I was suspecting this, maybe open an issue into the TSL2561 repository about compatibility with the latest ESP32 platform

murfy76, 2023-04-05:
Issue opened on TSL2561 repository, thank you.

murfy76, 2023-06-20:
I was not able to explain changes needed to make on TSL2561 repository. Same issue is on OMG 1.6.0 release - need to downgrade to espressif32@3.5.0. Thank you.

closed 2023-09-18 by github-actions[bot]