1technophile/OpenMQTTGateway #1593: Ethernet not connected with Arduino / Ethernet Shield (W5500)

prosperz31, 2023-04-06:
Hi,

When I try to setup an Arduino + Ethernet Shield (W5500), i have the log message in the Serial Monitor, indicates to me that no Ethernet connection is detected.

************* WELCOME TO OpenMQTTGateway **************
T: Adv eth cfg
T: ip: s␄�� 
T: Port: 74647387
T: Mqtt server: 192.168.0.6
N: RF_EMITTER_GPIO: 4 
N: RF_RECEIVER_GPIO: 1
T: ZgatewayRF command topic: home/OpenMQTTGateway_ATMEGA_ALL/commands/MQTTto433
T: ZgatewayRF setup done
T: enableActiveReceiver: 2
N: Switching to RF Receiver
T: mqtt_max_packet_size: 1024
N: OpenMQTTGateway modules: ["RF"]
N: ************** Setup OpenMQTTGateway end **************
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:

My prod_env.ini file :

[platformio]
default_envs = 
  atmega-piscine

[env:atmega-piscine]
platform = ${com.atmelavr_platform}
board = megaatmega2560
lib_deps =
  ${com-arduino.lib_deps}
  ${libraries.irremote}
  ${libraries.rfm69}
  ${libraries.rfm69spi}
  ${libraries.rfm69_low-power}
  ${libraries.rc-switch}
  ${libraries.newremoteswitch}
  ${libraries.bme280}
  ${libraries.bmp180}
  ${libraries.htu21}
  ${libraries.ahtx0}
  ${libraries.unifiedsensor}
  ${libraries.dht}
  ${libraries.tsl2561}
  ${libraries.a6lib}
  ${libraries.ina226}
  ${libraries.fastled}
  ${libraries.onewire}
  ${libraries.dallastemperature}
  ${libraries.rfWeatherStation}
  ${libraries.stl}
build_flags =
  ${com-arduino.build_flags}
  '-DZgatewayRF="RF"'
  '-DZmqttDiscovery="HADiscovery"'
  '-DsimplePublishing=true'
  '-DGateway_Name="OpenMQTTGateway_ATMEGA_ALL"'
  '-DMQTT_USER="arduino"'
  '-DMQTT_PASS="magd_pisc_#2303"'
  '-DMQTT_SERVER="192.168.0.6"'
  '-DMQTT_PORT="1883"'
  '-DESP32_ETHERNET=true'
  '-DNetworkAdvancedSetup=false'

Is W5500 not compatible with this MQTTGateway ?

Is there some specific parameters for Ethernet Shield in User_config.h ? 
My current one is : 

[platformio]
default_envs = 
  atmega-piscine

[env:atmega-piscine]
platform = ${com.atmelavr_platform}
board = megaatmega2560
lib_deps =
  ${com-arduino.lib_deps}
  ${libraries.irremote}
  ${libraries.rfm69}
  ${libraries.rfm69spi}
  ${libraries.rfm69_low-power}
  ${libraries.rc-switch}
  ${libraries.newremoteswitch}
  ${libraries.bme280}
  ${libraries.bmp180}
  ${libraries.htu21}
  ${libraries.ahtx0}
  ${libraries.unifiedsensor}
  ${libraries.dht}
  ${libraries.tsl2561}
  ${libraries.a6lib}
  ${libraries.ina226}
  ${libraries.fastled}
  ${libraries.onewire}
  ${libraries.dallastemperature}
  ${libraries.rfWeatherStation}
  ${libraries.stl}
build_flags =
  ${com-arduino.build_flags}
  '-DZgatewayRF="RF"'
  '-DZmqttDiscovery="HADiscovery"'
  '-DsimplePublishing=true'
  '-DGateway_Name="OpenMQTTGateway_ATMEGA_ALL"'
  '-DMQTT_USER="arduino"'
  '-DMQTT_PASS="magd_pisc_#2303"'
  '-DMQTT_SERVER="192.168.0.6"'
  '-DMQTT_PORT="1883"'
  '-DESP32_ETHERNET=true'
  '-DNetworkAdvancedSetup=false'

Regards,

1technophile, 2023-04-06:
Hi,

I never tried with the W5500, you could attempt to make it work with the latest Atmel AVR platform `4.1.0`
https://github.com/1technophile/OpenMQTTGateway/blob/6e8b4632131ff9e6b1771aceb577a7cdcfdaa960/platformio.ini#L175

prosperz31, 2023-04-07:
I have just try and same result in Serial Monitor : 

W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
......

Anotre way, when I try to ping my board, it answers (so it seems to be connected) :
Envoi d’une requête 'Ping'  192.168.0.252 avec 32 octets de données :
Réponse de 192.168.0.252 : octets=32 temps<1ms TTL=128
Réponse de 192.168.0.252 : octets=32 temps<1ms TTL=128
Réponse de 192.168.0.252 : octets=32 temps<1ms TTL=128
Réponse de 192.168.0.252 : octets=32 temps<1ms TTL=128

Statistiques Ping pour 192.168.0.252:
    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),
Durée approximative des boucles en millisecondes :
    Minimum = 0ms, Maximum = 0ms, Moyenne = 0ms

I do not understand :(

prosperz31, 2023-04-07:
Seems to not coming from hardware, I used RFLink on the same board and it work...

1technophile, 2023-04-07:
Here is an example of an environment that I use with a W5100:
```
[env:atmega-rfirdht-test]
platform = ${com.atmelavr_platform}
board = megaatmega2560
lib_deps =
  ${com-arduino.lib_deps}
  ${libraries.rc-switch}
build_flags = 
  ${com-arduino.build_flags}
  '-DZgatewayRF="RF"'
  '-DGateway_Name="OpenMQTTGateway_ATMEGA_RF_TEST"'
  '-DMQTT_USER="user"'
  '-DMQTT_PASS="pwd"'
  '-DMQTT_SERVER="192.168.1.17"'
monitor_speed = 115200
```

You may have to modify the code to adapt to the W5500:
https://github.com/1technophile/OpenMQTTGateway/blob/4fa74e31ba8f70aa99ca5118f46034b68ab844c6/main/main.ino#L1536

https://forum.arduino.cc/t/arduino-mega2560-and-ethernet-w5500/654798

prosperz31, 2023-04-08:
I try to update the connection to Ethernet code with one I have and working perfectly on another project on Arduino :
#else // Arduino case
void setup_ethernet() {
  const byte IP[] = {192, 168, 0, 252};
  const byte MAC[] = {0xA8, 0x61, 0x0A, 0xAE, 0x96, 0xB4}; //W5100 ethernet shield MAC address

  Log.trace(F("Setup ETHERNET." CR));
  if(Ethernet.begin(MAC) == 0)
  {
    Log.trace(F("Erreur DHCP." CR));
    if (Ethernet.hardwareStatus() == EthernetNoHardware) { Log.trace(F("Erreur Shield." CR)); } 
    else if (Ethernet.linkStatus() == LinkOFF) { Log.trace(F("Cable RJ45 deconnecte." CR)); }

    Ethernet.begin(MAC, IP);
  }
  Log.trace(F("ip: %s " CR), Ethernet.localIP());
}
#endif

I also try to update my environment tio the one you provide me with. 

But same result... It displays Network connected in loop... Even if it seems to be connected, cause I can ping it. Bu nothing is received from MQTT...

I think I will let down my project... It's not works with my config and I do not understand, that's all...

1technophile, 2023-04-09:
I would prefer to have the shield to debug this, unfortunately, I don't.

closed 2023-09-18 by github-actions[bot]