1technophile/OpenMQTTGateway #1597: Compilation error

eddieari, 2023-04-07:
Hello, 

get the following error output when I compile with Arduino IDE for the ESP8266.
Can anyone give me a tip on how to fix the error?

Greetings
Thomas

Error output Arduino IDE:

`Arduino: 1.8.19 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

main:291:11: error: declaration of template parameter 'T' shadows template parameter
  291 | void Config_update(JsonObject& data, const char* key, T& var);
      |           ^~~~~~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/main.ino:290:11: note: template parameter 'T' declared here
  290 | template <typename T> // Declared here to avoid pre-compilation issue (missing "template" in auto declaration by pio)
      |           ^~~~~~~~
main:291:27: error: too many template-parameter-lists
  291 | void Config_update(JsonObject& data, const char* key, T& var);
      |                           ^~~~~~~~~~~~~
main:291:55: error: 'T' has not been declared
  291 | void Config_update(JsonObject& data, const char* key, T& var);
      |                                                       ^
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/main.ino: In function 'void setup_wifimanager(bool)':
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/main.ino:1468:26: warning: NULL used in arithmetic [-Wpointer-arith]
 1468 |       while (*cert_in != NULL) {
      |                          ^~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/main.ino: In function 'void stateMeasures()':
main:1858:20: error: 'ENV_NAME' was not declared in this scope
 1858 |   SYSdata["env"] = ENV_NAME;
      |                    ^~~~~~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino: In function 'void RFtoMQTTdiscovery(uint64_t)':
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:90:29: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   90 |   char* switchRF[2] = {val, "RF"};
      |                             ^~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:103:7: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  103 |       "", "",
      |       ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:103:11: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  103 |       "", "",
      |           ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:105:7: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  105 |       "", "", "", "");
      |       ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:105:11: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  105 |       "", "", "", "");
      |           ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:105:15: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  105 |       "", "", "", "");
      |               ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZgatewayRF.ino:105:19: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  105 |       "", "", "", "");
      |                   ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino: In function 'void pubMqttDiscovery()':
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:25: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                         ^~~~~~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:35: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                                   ^~~~~~~~~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:48: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                                                ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:52: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                                                    ^~
In file included from /Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/main.ino:155:
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/config_mqttDiscovery.h:195:27: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  195 | #  define jsonVal         "{{ value_json.value | is_defined }}"
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/config_mqttDiscovery.h:195:27: note: in definition of macro 'jsonVal'
  195 | #  define jsonVal         "{{ value_json.value | is_defined }}"
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:65: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                                                                 ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:69: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                                                                     ^~
/Users/thomas/Downloads/OpenMQTTGateway-1.5.0 2/main/ZmqttDiscovery.ino:920:73: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  920 |   char* gatewayRF[8] = {"sensor", "gatewayRF", "", "", jsonVal, "", "", ""};
      |                                                                         ^~
exit status 1
declaration of template parameter 'T' shadows template parameter

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

NorthernMan54, 2023-04-07:
Personally I use VSCODE with platformio ( and so does the project team )

eddieari, 2023-04-07:
> 

Thanks, with that its works!

closed 2023-04-09 by 1technophile