1technophile/OpenMQTTGateway #1608: Implement the ESP's task watchdog timer

AlexMKX, 2023-04-16:
**Is your feature request related to a problem? Please describe.**
Sometime the gateways getting offline and returns back only with powercycle. Which means - it requires user interaction to get it back online.

**Describe the solution you'd like**
Use the watchdog timer with some long timeot like a 10-15 minutes on the MQTT command. If there is no MQTT command received from the MQTT (some chatty topic should be subscribed, or ping-like command issued by the OMG) the device will be reset with WDT automatically.

**Describe alternatives you've considered**
Actually, the watchdog is a silver-bullet or last resort for such situations. Software should be rock-solid itelf, but there is always could be a bug. Also there is always could be a bug on wifi/ethernet hardware, which requires to poweroff client adapter.

murfy76, 2023-04-24:
This could be great. I'm experiencing little different behavior on v1.5.0. After short, maybe 2s power outage (switch from main supply to ups) ESP looks like all good - LWT is online but no more new values received. Sending ESP reset via MQTT not helping. It's need to be powered off an on again. Thanks.

1technophile, 2023-04-24:
> Sometime the gateways getting offline and returns back only with powercycle. Which means - it requires user interaction to get it back online.

@AlexMKX @murfy76  Could you share the configuration/environment used please and your router model? Also if you could connect a serial monitor would it be possible to share the logs.

murfy76, 2023-04-24:
Sure, here is mine environment for OMG v1.5.0:

[env:esp32dev-all-test]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ble}
  ${libraries.decoder}
  ${libraries.rc-switch}
  ${libraries.bme280}
  ${libraries.pms}
  ${libraries.sgp40}
  ${libraries.unifiedsensor}
  ${libraries.tsl2561}
  ${libraries.ina226}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayRF="RF"'
  '-DZgatewayBT="BT"'
  '-DZsensorTSL2561="TSL2561"'
  '-DZsensorINA226="INA226"'
  '-DZsensorBME280="BME280"'
  '-DZsensorPMS="PMS"'
  '-DZsensorSGP40="SGP40"'
  '-DGateway_Name="ESP32B1"'
  '-DvalueAsATopic=true'  ; MQTT topic includes model and device (rtl_433) or protocol and id (RF and PiLight)
  '-UZmqttDiscovery'          ; disables MQTT Discovery
  '-DvalueAsATopic=true'      ; MQTT topic includes model and device
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="AP"'
  '-Dwifi_password="***"'
  '-DMQTT_SERVER="192.168.*.*"'
...
Wifi is powerline TP-LINK TL-WPA4220. I'm afraid that I wont be able to connect serial debug for this case, because this ESP is powered through Vin. And if there will be USB power I cant reproduce this behavior. Thanks.

1technophile, 2023-04-24:
Quite huge environment. Are you using all these sensors?
Also what is your free memory (published into SYStoMQTT )?

murfy76, 2023-04-24:
Yes, I'm using all of these sensors. Free mem values are - Min: 134944, Mean: 139744, Max: 140816 (in two days period). Thanks.

AlexMKX, 2023-04-25:
Hello, mine is here 

 https://github.com/AlexMKX/OMGOTA/blob/master/templates/prod_env.ini.j2

I do provision and OTA with Ansible. No special flags added besides the template and credentials. 
Probably this repo deserves to be mentioned at OMG ;)

Mine are simle 433 and BT gateways. Working in pair to cover the area. I'm using the esp32 and SRX882 receiver - that's all.

Wifi is 2014's ubiquity APs 
While digging kibana, I see no connection attempts after the offline. They not that often go offline, so I don't beleive serial log would help a lot, moreover they are in production and mounted inside box, which is inside the DIN box :) 

I do beleive the watchdog is a silver bullet for this anyway. When I put this functionality to my fork of ZigStar gateway, it made my ZigBee network steady.

AlexMKX, 2023-04-25:
Memory info. Note the gap between April 8 and April 12. This is when they went offline. After that I've updated to the new build and it seems memory usage is squeezed.

[image]

1technophile, 2023-04-25:
Yes the new esp32 platform is taking more memory unfortunately

closed 2023-09-10 by github-actions[bot]