1technophile/OpenMQTTGateway #1609: SilverCrest RTL433 Remotes parsing got lost between version 1.3.0 and 1.5.1

Dattel, 2023-04-16:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I have some removecontrolls from Silvercrest, each with 4 Channels on/off and 2 main on/off. Version 1.3 parses these device as Smoke-GS558. It is not the right device eigther but i was able to work with the parsed informations than.

Now with version 1.5.1 these device is no longer parsed. The log throws a "unknown msg"

**To Reproduce**
Steps to reproduce the behavior:
1. flashing omg1.3
2. pressing some buttons on the remotecontrol
3. watching the console and see the parsed signal
4. flashing omg1.5.1
5. pressing some buttons on the removecontrol
6. watching the console pointing out unknown msg

**Expected behavior**
The signal should be recognized as known device as of 1.3 already did

**Screenshots**
-no screenshots available :-D

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V1.3 development, 1.5.1 development)
* CC1101 attached
* using own prod_env.ini for device specifications - just copied between v1.3 and 1.5.1 

**Additional context**
Console messages from 1.3
```
N: [ MQTT->OMG ]: {"value":4557898,"protocol":5,"length":24,"delay":504,"tre_state":"-","binary":"010001011000110001001010","raw":"7052,504,1027,987,506,514,1035,496,1033,480,1048,986,527,499,1032,1005,524,1001,536,495,1057,491,1002,478,1048,989,551,977,542,489,1027,483,1065,481,1048,972,544,483,1026,518,1021,984,544,489,1027,990,558,494,1024,"}
N: [ MQTT->OMG ]: {"value":4913786,"protocol":5,"length":24,"delay":506,"tre_state":"-","binary":"010010101111101001111010","raw":"7151,478,1118,904,503,526,1050,467,1053,957,567,478,1060,958,556,492,1064,941,555,974,548,995,601,915,599,919,567,478,1033,1003,549,437,1095,477,1014,982,566,974,560,946,557,1000,558,443,1074,944,549,475,1091,"}

N: [ MQTT->OMG ]: {"value":4503261,"protocol":3,"length":23,"delay":101,"tre_state":"-","binary":"10001001011011011011101","raw":"11366,937,604,416,1110,419,1108,455,1074,930,602,432,1081,424,1112,928,604,436,1090,930,588,945,600,425,1092,924,593,931,615,413,1103,929,587,947,603,420,1089,945,601,931,597,907,614,434,1120,917,598,"}
N: [ MQTT->OMG ]: {"value":9215034,"protocol":5,"length":24,"delay":515,"tre_state":"-","binary":"100011001001110000111010","raw":"7297,952,565,458,1078,422,1106,422,1103,933,598,946,601,409,1093,438,1089,949,584,444,1083,443,1104,929,563,972,598,928,565,454,1080,435,1094,422,1103,430,1107,926,600,930,585,948,584,441,1101,922,623,398,1104,"}
```

Console messages from 1.5.1

```
rtl_433_ESP(6): Unparsed Signal length: 443000, Signal RSSI: -20, pulses: 150
rtl_433_ESP(6): RAW (443000): +2535-7056(-24)+489-1038(-23)+988-544(-22)+478-1040(-23)+492-1030(-22)+496-1021(-23)+995-536(-22)+482-1053(-23)+984-532(-22)+1011-512(-22)+507-1016(-22)+513-1030(-22)+481-1048(-22)+988-539(-22)+970-535(-22)+504-1026(-22)+483-1055(-23)+489-1034(-22)+996-525(-22)+485-1046(-22)+486-1044(-23)+992-517(-22)+497-1033(-23)+997-527(-22)+509-1020(-23)+3010-7066(-26)+486-1042(-21)+986-538(-22)+494-1018(-22)+489-1045(-22)+493-1029(-22)+977-547(-22)+500-1020(-22)+989-542(-22)+987-552(-22)+482-1043(-22)+495-1014(-23)+492-1044(-22)+979-551(-22)+979-545(-22)+490-1038(-23)+491-1048(-22)+465-1044(-23)+978-552(-22)+493-1024(-23)+485-1042(-22)+990-545(-22)+494-1038(-22)+966-534(-22)+493-1040(-22)+3019-7046(-24)+494-1044(-22)+986-531(-22)+482-1041(-22)+494-1036(-23)+495-1040(-22)+982-530(-22)+488-1047(-22)+967-561(-22)+978-545(-22)+480-1054(-23)+470-1055(-22)+486-1022(-23)+1001-523(-22)+994-529(-22)+489-1049(-22)+487-1043(-22)+501-1020(-22)+988-542(-22)+495-1028(-22)+478-1044(-22)+989-545(-22)+491-1020(-22)+990-537(-22)+485-1045(-22)+2997-7073(-25)+509-1018(-25)+994-524(-22)+485-1047(-23)+487-1043(-22)+495-1009(-22)+988-551(-22)+490-1027(-22)+983-567(-22)+982-524(-22)+499-1044(-22)+488-1025(-22)+485-1044(-22)+994-516(-21)+1015-512(-21)+508-1022(-22)+510-1026(-22)+504-1015(-21)+997-531(-21)+488-1048(-21)+488-1022(-22)+990-543(-21)+498-1030(-22)+995-526(-21)+488-1048(-22)+2991-7060(-25)+498-1046(-21)+969-542(-21)+498-1030(-21)+492-1029(-21)+485-1049(-21)+987-539(-21)+495-1014(-21)+994-562(-21)+970-546(-21)+491-1032(-21)+498-1018(-22)+488-1049(-21)+987-543(-21)+988-540(-21)+479-1044(-22)+490-1048(-21)+482-1045(-21)+963-563(-21)+482-1020(-21)+501-1040(-21)+973-539(-21)+493-1038(-21)+987-529(-21)+493-1039(-21)+2995-7070(-24)+503-1031(-20)+986-532(-21)+482-1045(-21)+495-1037(-21)+468-1053(-21)+979-538(-20)+507-1025(-21)+976-543(-21)+993-549(-20)+488-1046(-21)+469-1047(-20)+485-1023(-21)+993-539(-20)+995-535(-20)+483-1039(-20)+486-1036(-21)+496-1037(-20)+987-545(-20)+482-1040(-21)+491-1032(-20)+995-525(-20)+486-1045(-20)+985-544(-20)+485-0(-20)
N: Send on /RTL_433toMQTT/unknown msg {"model":"unknown","protocol":"signal parsing failed","duration":443000,"rssi":-20,"pulses":150}

rtl_433_ESP(6): Unparsed Signal length: 446000, Signal RSSI: -35, pulses: 150
rtl_433_ESP(6): RAW (446000): +2806-7039(-40)+489-1038(-39)+974-563(-35)+461-1065(-36)+461-1051(-36)+487-1045(-36)+467-1062(-37)+980-539(-36)+486-1044(-37)+967-562(-36)+469-1041(-37)+996-533(-36)+992-545(-36)+969-547(-36)+484-1043(-37)+473-1058(-36)+983-558(-36)+963-570(-36)+473-1045(-37)+964-546(-36)+993-538(-36)+983-548(-36)+467-1048(-36)+987-541(-36)+472-1078(-36)+2980-7056(-39)+489-1038(-36)+987-533(-36)+493-1041(-37)+490-1047(-36)+464-1051(-37)+478-1043(-36)+986-544(-36)+493-1045(-36)+985-541(-36)+486-1026(-36)+987-541(-36)+995-535(-36)+991-544(-36)+467-1048(-36)+486-1047(-36)+966-562(-36)+975-540(-36)+494-1039(-37)+986-531(-36)+995-537(-36)+988-542(-36)+480-1054(-37)+971-547(-36)+484-1032(-37)+3006-7064(-39)+488-1044(-35)+979-537(-36)+483-1037(-36)+483-1046(-37)+500-1039(-37)+467-1052(-37)+980-549(-37)+466-1075(-37)+974-540(-37)+471-1051(-37)+968-572(-37)+965-542(-37)+978-542(-37)+484-1057(-37)+483-1042(-38)+962-560(-37)+996-537(-37)+472-1055(-37)+985-532(-37)+984-544(-37)+983-554(-37)+467-1048(-37)+992-538(-37)+484-1045(-37)+2985-7065(-39)+489-1045(-37)+997-537(-37)+474-1054(-38)+485-1025(-38)+491-1034(-38)+495-1045(-38)+973-536(-38)+484-1053(-39)+987-556(-38)+468-1039(-39)+994-535(-38)+970-566(-38)+957-575(-38)+457-1047(-39)+499-1027(-38)+991-551(-38)+966-568(-38)+457-1054(-38)+992-526(-38)+996-538(-38)+984-548(-38)+467-1047(-38)+987-541(-38)+470-1076(-38)+2985-7052(-41)+504-1021(-39)+992-534(-38)+516-1015(-38)+497-1040(-39)+490-1029(-39)+480-1052(-39)+986-525(-39)+489-1041(-38)+1003-536(-39)+489-1023(-39)+998-541(-39)+995-508(-39)+1009-520(-39)+501-1030(-39)+496-1040(-39)+985-542(-39)+993-530(-38)+500-1020(-39)+999-543(-39)+989-537(-39)+985-520(-39)+501-1029(-39)+1001-524(-39)+506-1021(-39)+3018-7061(-43)+496-1027(-41)+996-526(-39)+490-1047(-39)+489-1017(-39)+494-1036(-39)+483-1054(-39)+986-532(-39)+486-1040(-39)+995-538(-39)+490-1021(-39)+1010-524(-39)+989-540(-39)+1003-518(-39)+483-1055(-39)+468-1056(-40)+985-528(-39)+997-534(-39)+504-1024(-39)+984-545(-39)+987-545(-39)+970-544(-39)+495-1039(-39)+989-538(-39)+495-0(-39) 
N: Send on /RTL_433toMQTT/unknown msg {"model":"unknown","protocol":"signal parsing failed","duration":446000,"rssi":-35,"pulses":150}

rtl_433_ESP(6): Unparsed Signal length: 1286258, Signal RSSI: -71, pulses: 172
rtl_433_ESP(6): RAW (1286258): +196640-200766(-79)+347-149990(-71)+288-19637(-43)+786-1019(-43)+494-1099(-45)+482-2173(-46)+155-1355(-46)+200-1348(-46)+194-1352(-44)+956-675(-43)+866-812(-43)+719-2148(-45)+927-2206(-45)+119-2543(-46)+936-2207(-46)+96-1378(-46)+191-1386(-46)+929-722(-46)+789-3752(-46)+869-3812(-48)+794-1064(-49)+477-971(-50)+588-3690(-51)+170-1383(-50)+156-1432(-50)+865-824(-47)+702-882(-47)+665-2276(-49)+819-2188(-49)+124-2534(-51)+921-2252(-46)+72-1479(-47)+73-1428(-48)+876-800(-48)+735-3743(-49)+881-2185(-48)+120-1389(-49)+924-814(-48)+717-843(-49)+697-2179(-51)+154-1361(-50)+177-1361(-50)+183-1383(-48)+927-778(-46)+755-836(-48)+688-2220(-50)+865-2187(-49)+149-2497(-49)+974-705(-45)+54-1384(-47)+168-1405(-47)+144-1384(-48)+920-703(-46)+827-516(-35)+265-1250(-36)+287-1224(-37)+1062-473(-36)+299-1221(-37)+308-1225(-36)+1060-492(-36)+1046-475(-36)+1043-496(-37)+267-1245(-36)+312-1216(-37)+295-1225(-36)+307-1226(-36)+1072-452(-36)+1067-479(-37)+1060-469(-37)+286-1260(-37)+1052-482(-37)+263-1248(-37)+2971-7257(-39)+939-587(-36)+442-1109(-36)+417-1122(-37)+418-1114(-37)+924-613(-37)+917-645(-37)+390-1153(-38)+388-1136(-39)+938-610(-38)+414-1121(-40)+411-1119(-39)+930-662(-38)+900-641(-38)+909-643(-39)+390-1170(-39)+3 [... cut ...] 59-1185(-37)+361-1148(-38)+417-1153(-38)+399-1129(-38)+413-1145(-38)+898-679(-38)+358-1163(-39)+911-646(-38)+901-645(-39)+895-666(-39)+373-1175(-39)+386-1119(-39)+932-637(-39)+382-1166(-39)+909-645(-39)+391-1146(-39)+919-623(-39)+918-657(-39)+889-631(-39)+916-673(-39)+373-1145(-39)+913-644(-39)+387-0(-39) 
N: Send on /RTL_433toMQTT/unknown msg {"model":"unknown","protocol":"signal parsing failed","duration":556000,"rssi":-71,"pulses":175}
```

is there anything i can further provide?

Thanks :-D

Dattel, 2023-04-17:
i have two different setups.. both attached to the same workflow in node-red for better RF-coverage and fallback (but listening on different mqtt topics with diffenent algorythm)

[image]

-ESP8266 with a simple MK-RM Receiver using RF/RC-Switch
-the second is a ESP32 DevBoard with RTL and rcswitch 

Recently i wanted to update to latest version (1.5.1) but stopped migration and reverted back to 1.3, because the ESP32 doesnt working properly

Dattel, 2023-04-17:
just updated my issue.
I tried v1.4.0 and it still works.. The break is between 1.4.0 and 1.5.0 - maybe that helps?

NorthernMan54, 2023-04-19:
We have identified two recent changes within rtl_433_ESP since rtl_433_ESP release 0.1.6 that impact the ability of receivers to receive signals.

And will be releasing a new version to resolve.

Dattel, 2023-04-23:
RemoteControl is parsed again with RTL_0.2.1... thanks for your quick support.

seems to work more stable now... flashed omg1.5.1 dev with RTL0.2.1 yesterday and it's still online and receiving.

closed 2023-04-23 by Dattel

hb9eue, 2023-05-18:
I wonder if I have missed something. My omg1.5.1 still does not receive my LaCrosse weather sensor anymore. Has there bin a fixed re-release of 1.5.1 without bumping the version so it need manual re-flashing?

hb9eue, 2023-05-19:
Thank you. I compiled the actual development branch and can confirm, receptions works again. But I get a lot of unexpected restarts (sometimes after  2-3 minutes, sometimes after some hours).