1technophile/OpenMQTTGateway #1616: Requesting 915MHz config for lilygo-rtl_433

lachesis, 2023-04-22:
**Is your feature request related to a problem? Please describe.**
I have two Lilygo LORA32 boards that I bought for use with rtl_433 via OpenMQTTGateway. One is on 433MHz, which works great with the out of the box config, while the other is on 915MHz, for use with a weather station in the US. I tried flashing both with lilygo-rtl_433 config. The 433MHz one sees my cheap sensors, while the 915MHz one sees nothing, and claims to be on `"mhz": 433.92`.

**Describe the solution you'd like**
Please add another config to the web flashing UI that sets the frequency of the board to 915MHz.

**Describe alternatives you've considered**
I could probably build this software myself, but it looks like it has a lot of dependencies and it scares me. :) The web flash utility is so easy to use, so I'd love to use it with the 915MHz device as well.

**Additional context**
Not sure, let me know if anything else is needed.

NorthernMan54, 2023-04-23:
To change the frequency you can use a MQTT command, details are here

https://docs.openmqttgateway.com/use/rf.html#rtl-433-device-decoders

Please ensue to use the correct gateway name

lachesis, 2023-04-23:
Oh interesting, I did not see that option. I am having an issue getting it to work however. I tried publishing to all 3 of:

home/OpenMQTTGateway/commands/lilygo_rtl433_915
home/commands/lilygo_rtl433_915
home/lilygo_rtl433_915/commands
home/lilygo_rtl433_915/commands/mhz
home/lilygo_rtl433_915/commands/lilygo_rtl433_915

with the message:
`{"mhz": 915.0}`

But the "mhz" value in the SYStoMQTT topic did not change. I also tried a power cycle both before and after publishing. This did not help either. Here is a screenshot from my MQTT Explorer if you can tell me what I'm doing wrong?

[image]

Thank you!

lachesis, 2023-04-23:
Ah, I tried the topic `home/lilygo_rtl433_915/commands/MQTTtoRTL_433` and I now see that the board responds to the `mhz` and `status` commands, but the SYStoMQTT `mhz` value does not update and I do not see any decoded messages.

lachesis, 2023-04-23:
Oh I believe my sensor (weather station) uses FSK. It is this decoder in rtl433:
```
    [78]  Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor
```
I see that this decoder is not listed on the doc page that you linked, and I believe I've seen that FSK is not supported, only OOK. If that's the case, you can go ahead and close this issue!

Here is the dump from a run with `-A` in rtl_433:
```
Detected FSK package	2023-04-23 18:11:48
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-04-23 18:11:48                    Protocol  : 78
model     : Fineoffset-WH24                        ID        : 195
Battery   : 1            Temperature: 6.4 C        Humidity  : 61 %          Wind direction: 213       Wind speed: 2.4 m/s       Gust speed: 3.4 m/s       Rainfall  : 3138.3 mm     UV        : 54            UVI       : 0             Light     : 6696.0 lux
Integrity : CRC
Analyzing pulses...
Total count:  105,  width: 12.00 ms		(12004 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    2,  width:    3 us [3;3]	(   3 S)
 [ 2] count:   41,  width:   57 us [55;60]	(  57 S)
 [ 3] count:    8,  width:  116 us [115;117]	( 116 S)
 [ 4] count:    4,  width:  174 us [173;175]	( 174 S)
 [ 5] count:    2,  width:  232 us [232;233]	( 232 S)
 [ 6] count:    1,  width:  349 us [349;349]	( 349 S)
 [ 7] count:    4,  width:    7 us [7;9]	(   7 S)
 [ 8] count:   18,  width:    1 us [1;1]	(   1 S)
 [ 9] count:   20,  width:    2 us [2;2]	(   2 S)
 [10] count:    1,  width:   16 us [16;16]	(  16 S)
 [11] count:    2,  width:    5 us [5;5]	(   5 S)
 [12] count:    1,  width:   32 us [32;32]	(  32 S)
Gap width distribution:
 [ 0] count:    3,  width:    2 us [2;2]	(   2 S)
 [ 1] count:   37,  width:   57 us [50;59]	(  57 S)
 [ 2] count:    5,  width:  174 us [174;174]	( 174 S)
 [ 3] count:    3,  width:  231 us [231;233]	( 231 S)
 [ 4] count:    4,  width:  116 us [116;117]	( 116 S)
 [ 5] count:    3,  width:  368 us [348;408]	( 368 S)
 [ 6] count:    2,  width:  522 us [465;580]	( 522 S)
 [ 7] count:    1,  width:  290 us [290;290]	( 290 S)
 [ 8] count:    1,  width:   91 us [91;91]	(  91 S)
 [ 9] count:    2,  width:   33 us [31;35]	(  33 S)
 [10] count:    1,  width:   24 us [24;24]	(  24 S)
 [11] count:    2,  width:   11 us [11;12]	(  11 S)
 [12] count:   13,  width:    4 us [4;4]	(   4 S)
 [13] count:   18,  width:    3 us [3;3]	(   3 S)
 [14] count:    2,  width:   17 us [17;18]	(  17 S)
 [15] count:    4,  width:    6 us [6;7]	(   6 S)
Pulse period distribution:
 [ 0] count:    1,  width:    2 us [2;2]	(   2 S)
 [ 1] count:   32,  width:    5 us [5;5]	(   5 S)
 [ 2] count:   27,  width:  116 us [114;118]	( 116 S)
 [ 3] count:   10,  width:  235 us [231;265]	( 235 S)
 [ 4] count:    6,  width:  173 us [173;175]	( 173 S)
 [ 5] count:    6,  width:  309 us [290;348]	( 309 S)
 [ 6] count:    4,  width:  450 us [406;523]	( 450 S)
 [ 7] count:    2,  width:  610 us [581;639]	( 610 S)
 [ 8] count:    3,  width:   43 us [39;47]	(  43 S)
 [ 9] count:    2,  width:   31 us [31;32]	(  31 S)
 [10] count:    2,  width:   62 us [57;67]	(  62 S)
 [11] count:    3,  width:   18 us [18;19]	(  18 S)
 [12] count:    1,  width:    7 us [7;7]	(   7 S)
 [13] count:    3,  width:    3 us [3;3]	(   3 S)
 [14] count:    1,  width:   23 us [23;23]	(  23 S)
 [15] count:    1,  width:   12 us [12;12]	(  12 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   20,  width:    3 us [3;3]	(   3 S)
 [ 2] count:   78,  width:   57 us [50;60]	(  57 S)
 [ 3] count:   12,  width:  116 us [115;117]	( 116 S)
 [ 4] count:    9,  width:  174 us [173;175]	( 174 S)
 [ 5] count:    5,  width:  232 us [231;233]	( 232 S)
 [ 6] count:    4,  width:  363 us [348;408]	( 363 S)
 [ 7] count:    8,  width:    7 us [6;9]	(   7 S)
 [ 8] count:   21,  width:    1 us [1;1]	(   1 S)
 [ 9] count:   23,  width:    2 us [2;2]	(   2 S)
 [10] count:    3,  width:   17 us [16;18]	(  17 S)
 [11] count:    2,  width:    5 us [5;5]	(   5 S)
 [12] count:    3,  width:   32 us [31;35]	(  32 S)
 [13] count:    2,  width:  522 us [465;580]	( 522 S)
 [14] count:    1,  width:  290 us [290;290]	( 290 S)
Level estimates [high, low]:  15889,     90
RSSI: -0.1 dB SNR: 22.5 dB Noise: -22.6 dB
Frequency offsets [F1, F2]:   -2716,   2894	(-41.4 kHz, +44.2 kHz)
Guessing modulation: No clue...
```

NorthernMan54, 2023-04-23:
Sir you are correct FSK modulation is not currently supported.  It is something we are working on, hopefully we are successful.

Also the command structure for your setup would be

`home/lilygo_rtl433_915/commands/MQTTtoRTL_433 {"mhz":915}`

closed 2023-04-23 by NorthernMan54

lachesis, 2024-01-17:
Resolved by pull request 1812! Thanks guys.