1technophile/OpenMQTTGateway #1619: Heltec ESP32 LoRa SX1278 V3 supported, too?

sheilbronn, 2023-04-26:
After reading about the success OpenMQTTGateway on the Heltec ESP32 LoRa SX1278 *V2* I decided to get, but accidentally bought a Heltec ESP32 LoRa SX1278 *V3* (offered as: ["SX1278 ESP32 LoRa IoT Internet of WIFI Bluetooth-compatible Development Board 433mhz for Arduino 0.96 Inch OLED Digital Display"](https://www.aliexpress.com/item/1005004645530908.html)). Now, I'm wondering whether this version is supported, too?!

According to the picture on AliExpress the PIN layout is a little different as can be seen on the 6th picture of the website

Did anybody get OpenMQTTGateway top run on this board? If yes, are there docs for this? Thanks?

NorthernMan54, 2023-04-26:
As long as the lora chipset is SX1278 you should be fine.  Looking at the pictures it appears to be a clone of the V2 board.

sheilbronn, 2023-04-28:
The pictures showing the PIN layouts show differences in the layout of V2 and V3 - wouldn't that make a difference?

- V2: https://community.hiveeyes.org/t/heltec-wifi-lora-32/3125
[image]

- V3: (attached)
[image]

(Both V2 and V3 seem to be available for 433 Mhz and 868 MHz. Frequency doesn't seem to matter for the PIN layout.)

sheilbronn, 2023-04-28:
According to https://github.com/NorthernMan54/rtl_433_ESP/issues/82 and https://heltec.org/project/wifi-lora-32-v3/
 there are be more differences such as ESP32-D0 vs ESP32-S3 and SX1276 vs. SX1262:
 
[image]

ryan-makes-it, 2023-05-04:
I also ended up with a 915mhz WiFi LoRa 32 v3 with the sx1262 chip. I was hoping to use it to listen to Ecowhitt soil moisture sensors (https://www.ecowitt.com/shop/goodsDetail/19), as using a RTL usb radio set up as a mqtt bridge with home assistant works well with them. 

I'm happy to test any code updates on my v3 if that would assist in development, or help source one for the project developers?

NorthernMan54, 2023-06-21:
If the chipset on the board is sx127x, it is supported

If the board does not have this chipset, it can still be used as a BLE receiver.

pls note that the sx126x chipset will never be supported, as it does not support direct mode.

closed 2023-06-21 by NorthernMan54