1technophile/OpenMQTTGateway #1628: MQTT blasted with  {"id": "78:D8:84:8B:8D:87", "rssi": -96, "brand": "Apple", "model": "Apple Continuity", "model_id": "APPLE_CONT", "device": "Apple device", "mfr": "Apple, Inc."}

sainio, 2023-05-01:
After some update my MQTT is filled with Apple Continuity topics, but I have no Apple devices.  How to stop?

sainio, 2023-05-01:
> Hi @sainio <https://github.com/sainio>,
>
> these Apple Continuity broadcasts are actually recognised to be
> filtered out by default. Can you make sure you have
>
> |"randommacs":false|
>
> in your BTtoMQTT setting?

And where would that setting be?

DigiH, 2023-05-01:
As I stated above, you don't seem to be running OpemMQTTGateway, for which this setting would be applicable, but Theengs Gateway, as your quoted MQTT JSON in the title includes `"mfr": "Apple, Inc."`, which is unique to Theengs Gateway.

Are you possibly using the Theengs Gateway add-on for Home Assistant?

If so, the above suggestion to update to the latest version 1.0.0. should remedy the issue you are experiencing.

If not, you should log an issue against [Theengs Gateway.](https://github.com/theengs/gateway)

closed 2023-09-09 by github-actions[bot]