1technophile/OpenMQTTGateway #1642: Experience with Mi Band 7?

tomcat-xx, 2023-05-15:
Not really a bug ...

Anyone having experience with Mi Band 7? Advertisement is received and published via MQTT, but "steps" are missing. I used different band in the past and all worked.

DigiH, 2023-05-15:
Hi @tomcat-xx,

Could you post the advertisements received by MQTT, and have you also made sure that the following two options are enabled in the Zepp Life app? 

Discoverable - for steps sharing
Acticity heart rate sharing - well … ;)

[image]

tomcat-xx, 2023-05-16:
Hi @DigiH,

Thanks for pointing out different settings. The one about heart rate I did not know before. Not sure if it was in the old app (before ZEPP) when I setup my Band 4. I'm only using original app for first setup. Same in this case for the Band 7 and then only activated "Discoverable". That is also only thing that is activated for the Band 4 (still in use). I think without that setting, I would not get any data (brand, model, name, ... is received from Band 7).  

intervalacts is set to 55555, but that is not related to device. So when getting steps for the Band 4 I should also get for Band 7.

Rest I have to add later, when Band 7 "is back from vacation" ;-)

Thomas

DigiH, 2023-05-18:
> Rest I have to add later, when Band 7 "is back from vacation" ;-)

When back from vacation you should also turn on [showing advertisement and advanced data](https://docs.openmqttgateway.com/use/ble.html#advanced-advertisement-and-advanced-data-default-false) on your gateway so we can see what data the Mi Band 7 transmits and why it is not being decoded and how to add it to the decoder.

Thanks

DigiH, 2023-06-23:
Hi @tomcat-xx,

Just wondered if you had any more chance to test with the Mi Band 7?

Thanks

tomcat-xx, 2023-07-14:
Sorry for late reply. Mi Band 7 (my daughter ;-) ) was back over weekend, but then I was on vacation. I did some tests but did not post it here so far.

Unfortunately it does not look good for me. 

Without changes (just "Discoverable" on) I got:
`{
  "id": "11:22:33:44:55:66",
  "name": "Xiaomi Smart Band 7 5566",
  "rssi": -90,
  "brand": "Xiaomi/Amazfit",
  "model": "Mi Band/Smart Watch",
  "model_id": "MB/SW"
}`

With heart rate on then: 
`{
  "id": "11:22:33:44:55:66",
  "name": "Xiaomi Smart Band 7 5566",
  "rssi": -75,
  "brand": "Xiaomi/Amazfit",
  "model": "Mi Band/Smart Watch",
  "model_id": "MB/SW",
  "act_bpm": 73
}`

And with "showing advertisement and advanced data" then: 
`{
  "id": "11:22:33:44:55:66",
  "mac_type": 1,
  "adv_type": 0,
  "name": "Xiaomi Smart Band 7 5566",
  "manufacturerdata": "570102ffffffffffffffffffffffffffffffff03c20803001415",
  "rssi": -63,
  "brand": "Xiaomi/Amazfit",
  "model": "Mi Band/Smart Watch",
  "model_id": "MB/SW",
  "type": "BODY",
  "acts": true
}`

`{
  "id": "11:22:33:44:55:66",
  "mac_type": 1,
  "adv_type": 0,
  "manufacturerdata": "570102020149ffffffffffffffffffffffffff02c20803001415",
  "rssi": -84,
  "brand": "Xiaomi/Amazfit",
  "model": "Mi Band/Smart Watch",
  "model_id": "MB/SW",
  "type": "BODY",
  "acts": true,
  "act_bpm": 73
}`

It looks as if heart rate is visible in manufacturerdata, but for different step counts I could not see changes there.

All looks very similar to my Band 4, juts there is additional "steps".

DigiH, 2023-07-14:
Thanks @tomcat-xx,

Everything does indeed look identical to previous Mi Band 4-6. For the steps, which are encoded in the servicedata, not manufacturerdata like the activity-heart rate, active scanning is required.

Can you have a look at your **intervalacts** setting in the **BTtoMQTT** message. With that active scan interval (ms) you should also get the servicedata and the encoded steps information of your Mi Band 7. If you could catch a message with this servicedata and "showing advertisement and advanced data" set to true.

tomcat-xx, 2023-07-15:
The band is not here at the moment, but my own one ordered now.

What I'm wondering: After your reply I looked into logs I recorded. For the Band 4 there is indeed "servicedata" (and also "servicedatauuid"). Did not realize that before. So it should be also there for the Band 7 because for both logs were recorded at same time. But let's see when a Band 7 will be always here ;-) and I have more time to check.

(Intervalacts is set to 55555 as written above.)

By the way, getting step is just a funny option. More important is using the Bands for presence detection in home automation. So even without steps it is very useful.

tomcat-xx, 2023-07-19:
Now I have it. But even after one hour no difference. Same as stated above: For the Band 4 (and other devices) I'm getting "servicedata" and "servicedatauuid" (after a minute or so), but nothing for the Band 7.

DigiH, 2023-07-19:
Could it be that there is a slightly different setting to **Discoverable ON** for the Mi Band 7 in its settings of the Zepp Life app? 

For all the Mi Bands 4-6 this is the setting which makes them _visible to nearby devices_ and broadcasts the "servicedata" and "servicedatauuid" with the steps information. I'm wondering if this has been changed for the Mi Band 7.

tomcat-xx, 2023-07-19:
Settings page in Zepp Life app looks the same. There is also message about bluetooth address AND steps, when turning discoverable on. Band is also not visible (to OMG and BLE monitor app) with "off". With "on2 all is published what I posted above, but no servicedata/servicedatauuid.

DigiH, 2023-07-19:
Hmm, very strange, as there really is not really enough bytes left to have the steps encoded there, as you have nicely shown with your masked MAC address, which is also contained in the manufacturerdata.

If the servicedata would have a different length for Mi Band 7 it would not be recognised by the current decoder, but as you say, there is no servicedata and servicedatauuid at all :(

Could you scan your Mi Band 7 for a while with the nRF Connect app on your phone, just to see if any servicedata does show up there, which for whatever peculiar reason we might not pick up?

Other than that I really don't know what could be the issue, as everything you have tried and done so far points to an expected similar behaviour as Mi band 4-6.

tomcat-xx, 2023-07-19:
nRF Connect also does not show service data. But shows, both bands are advertising 55 byte. 7 of it are used for service data (type 0x16) on Band4. On Band 7 all 55 are used for other types.

closed 2023-07-19 by tomcat-xx

elchitico, 2023-08-10:
Good morning, I just bought my band 7 and I'm having the problems described here. I have activated the options in zeeplife for bpm and steps, but only the bpm data is displayed.
openmqttgateway version 1.6

elchitico, 2023-08-10:
> Hi @elchitico
> 
> This seems to be the issue with the Mi Band 7, that the steps are not freely transmitted in the **servicedata** any longer, whereas the activity bpm are still in the **manufacturerdata** when an activity is ongoing.
> 
> If you want to test and verify this further, could you also activate [showing advertisement and advanced data](https://docs.openmqttgateway.com/use/ble.html#advanced-advertisement-and-advanced-data-default-false), to see if you do catch any servicedata, or possibly different manufacturerdata, at all.
> 
> The best application to monitor the broadcasts over a period of time is [MQTT Explorer](https://github.com/thomasnordquist/MQTT-Explorer/releases/tag/v0.3.5) with its history.
> 
> Could you also check in the Zepp Life app if there might be a firmware update available for the Mi Band 7?

My band 7 was updated to the same link with zeeplife, it is in the latest version.
I use mqtt explorer, but there is no service data steps.
I try to activate the advanced data mode to see if I can see any more service data.
But from what you told me, you have to be aware that you will not be able to see the step data in the successive mi band, am I right?
thank you very much for your collaboration

DigiH, 2023-08-10:
Correct, it looks as if the Mi Band 7 does not share its steps data freely any longer, as all the Mi Bands 4-6 (at least from my experience, might have already stated earlier) did, unless there might be a new different setting which needs activating, or a privacy setting which might need deactivating.

As no one in our team has a Mi Band 7 we rely on feedback from users like yourself and tomcat-xx, so thank you both for your collaboration.

tomcat-xx, 2023-08-10:
Update from my side: I reported issues to Xiaomi (at least message in app about steps is wrong). They requested to use Mi Fitness App instead. Since I had to switch from Mi Fit to ZEPP Life not that long ago, I did not even know that there is new/alternative app. Was sceptic about result anyway. And yes, is the same, same options to activate and same result. Reported back and waiting for feedback.

elchitico, 2023-08-10:
> Update from my side: I reported issues to Xiaomi (at least message in app about steps is wrong). They requested to use Mi Fitness App instead. Since I had to switch from Mi Fit to ZEPP Life not that [...]

I just installed the app you mention, My Fitness and I see that only the option to share bpm appears, no option appears for the steps

tomcat-xx, 2023-09-01:
Short update: As expected (at least by me) Mi Fitness App did not change anything. After reporting that back, I was asked to send my Band to repair it ;-) Now again wrote back ...

@elchitico: There is an was no separate switch for steps (in all apps), it was just send when "discoverable" is switched on.

DigiH, 2023-09-01:
Thanks for the update @tomcat-xx 

Let's hope that with a future firmware update the Mi Band 7 will show the steps again in its broadcast servicedata when Discoverable is turned ON.

michi1972-dev, 2024-04-28:
I have the same problem. No steps get transfered freely. Must be a Xiaomi Band 7 firmware issue. Any idea?

michi1972-dev, 2024-04-28:
Why is this issue closed? It is still a problem.

DigiH, 2024-04-28:
> Why is this issue closed? It is still a problem.

Because this seems to be the status quo for the Mi Band 7 and possibly onwards - that the steps are no longer freely broadcast in the advertising data, therefore currently not receivable nor decodable with Theengs Decoder for OpenMQTTGateway.

michi1972-dev, 2024-04-28:
Got it. Thank you!