1technophile/OpenMQTTGateway #1644: RTL_433_ESP Battery Auto Discovery Missing Unit of Measurement

mwolter805, 2023-05-17:
**Describe the bug**
Home Assistant expects device class 'battery' unit of measurement to be '%' to be present in the MQTT auto discovery value

<details>

<summary>Warning in Home Assistant Logs</summary>
Entity sensor.matts_room_thgr810_1_battery (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('battery') it is using; expected one of ['%']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. Use the config below with an Olimex PoE development board and an SX127x radio module
2. Power on an Oregon-THGR810 sensor
3. Review the logs in Home Assistant or review the MQTT auto discovery topic

<details>

<summary>esp32-olimex-poe-rtl_433_env.ini device environment file</summary>

```ini
[platformio]
default_envs = esp32-olimex-poe-rtl_433-ota

[env:esp32-olimex-poe-rtl_433]
platform = ${com.esp32_platform}
board = esp32-poe
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.rtl_433_ESP}  ; when disabled loads custom library in lib/rtl_433_ESP with disabled decoders in src/decoders.cpp
  ${libraries.wifimanager32}
build_flags =
  ${com-esp.build_flags}
; *** Main Gateway Options ***
  '-DGateway_Name="OpenMQTTGateway"'  ; host name and gateway mqtt topic name
  '-DNTP_SERVER="10.10.10.12"'  ; set local NTP server
  '-DESP32_ETHERNET=true'   ; enabling ethernet disables wifi
;   '-DESPWifiManualSetup=true'
; *** MQTT Options ***
  '-DMQTT_SERVER="10.10.10.12"'
  '-DMQTT_USER="username"'
  '-DMQTT_PASS="password"'
  '-Dota_password="password"'   ; ota password for espota.py
  '-Dsensor_Retain=true'      ; retain mqtt sensor value (sensor values survive HA reboot)
  '-DvalueAsATopic=true'       ; MQTT topic includes model and device
; *** OpenMQTTGateway Modules ***
  '-DZgatewayRTL_433="rtl_433"' ; set RTL_433 module mqtt topic name
  '-DZradioSX127x="SX127x"'
; *** rtl_433_ESP Options ***  
;  '-DRTL_DEBUG=4'           ; rtl_433 verbose mode
;  '-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
;  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'   ; display memory usage information
;  '-DDEMOD_DEBUG=true'  ; display signal debug info
;	 '-DMY_DEVICES=true'		; subset of devices
;  '-DPUBLISH_UNPARSED=true'   ; publish unparsed signal details
;  '-DRSSI_THRESHOLD=12'         ; Apply a delta of 12 (default 9)
;  '-DOOK_FIXED_THRESHOLD=0x50'  ; Inital OOK Threhold - Only for SX127X (default 15, old default 5A) 
;  '-DAVERAGE_RSSI=5000'     ; Display RSSI floor ( Average of 5000 samples )
;  '-DSIGNAL_RSSI=true'             ; Display during signal receive
;  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
;  '-DMINRSSI=-102'                  ; default is -82
; *** RF Module Pins ***
  '-DRF_SX1278="SX1278"'   ;
  '-DRF_MODULE_DIO0=15'    ; SX1278 pin DIO0
  '-DRF_MODULE_DIO1=4'     ; SX1278 pin DIO1
  '-DRF_MODULE_DIO2=16'    ; SX1278 pin DIO2
  '-DRF_MODULE_RST=14'     ; SX1278 pin RST
; *** Used non-standard SPI pins due to ethernet module ***
   '-DRF_MODULE_CS=13'     ; SX1278 pin NSS
   '-DRF_MODULE_MOSI=2'    ; SX1278 pin MOSI
   '-DRF_MODULE_MISO=5'    ; SX1278 pin MISO
   '-DRF_MODULE_SCK=32'    ; SX1278 pin SCK
; *** RadioLib Options ***
;  '-DRADIOLIB_DEBUG=true'
;  '-DRADIOLIB_VERBOSE=true'
custom_description = RTL_433 gateway using ethernet and AI-Thinker Ra-02 SX1278 LoRa Module
custom_hardware = OLIMEX ESP32-POE with SX1278
board_upload.speed = 460800
; board_monitor.speed = 460800

[env:esp32-olimex-poe-rtl_433-ota]
extends = env:esp32-olimex-poe-rtl_433  ; load main environment above
; *** OTA Options ***
upload_protocol = espota    ; use espota.py
upload_port = 10.10.21.51   ; IP address or hostname of esp32
upload_flags =
  --auth=password   ; ota_password entered above
  --port=8266
  --host_port=40000    ; custom port allowed through firewall from ESP32 to host performing upload
; monitor_port = socket://10.10.21.51:23

; Serial upload: pio run --target upload --environment esp32-olimex-poe-rtl_433
; OTA upload: pio run --target upload
```

</details>

<details><summary>Received Auto Discovery Value</summary>

```json
{
	"stat_t": "+/+/RTL_433toMQTT/Oregon-THGR810/2/48",
	"dev_cla": "battery",
	"name": "battery",
	"uniq_id": "Oregon-THGR810-2-48-battery_ok",
	"val_tpl": "{{ float(value_json.battery_ok) * 99 + 1 | is_defined }}",
	"state_class": "measurement",
	"device": {
		"identifiers": [
			"Oregon-THGR810-2-48"
		],
		"connections": [
			[
				"mac",
				"Oregon-THGR810-2-48"
			]
		],
		"model": "Oregon-THGR810",
		"name": "Oregon-THGR810-2-48",
		"via_device": "OMG-RACK"
	}
}
```

</details> 

**Expected behavior**
The battery auto discovery MQTT value should include 
```json
"unit_of_measurement": "%"
```

**Environment:**
* OpenMQTTGateway version 1.5.1
* RTL_433_ESP version 0.2.1

**Additional context**
Below is an example of the correct unit of measurement from the MQTT auto discovery script used by rtl_433
https://github.com/merbanan/rtl_433/blob/93f0f30c28cfb6b82b8cc3753415b01a85bee91d/examples/rtl_433_mqtt_hass.py#L180-L190

mwolter805, 2023-05-27:
@NorthernMan54 Would you mind taking a look at this? Thank you

NorthernMan54, 2023-05-28:
On the list for the next update

1technophile, 2023-06-12:
Should be fixed with v1.6.0

closed 2023-06-12 by 1technophile