1technophile/OpenMQTTGateway #1652: Compilation BLE + RTL433 fails claiming type TheengsDecoder not a type.

AlmightyFrog, 2023-05-22:
**Describe the bug**
Compilation fails claiming TheengsDecoder not a type (which is indeed available in decoder.h):
`TheengsDecoder' does not name a type; did you mean 'UseExtDecoder'?`
Include structure seems fine, defines seem to be set correct so that decoder.h is included; it is a little hard to tell cause sadly this project still uses ino instead of cpp and therefore the build in PIO is not as clear as it could be.

All i want is ESP32 + Pilight + RTL433 + Somfy + BLE; in theory offload to PC over MQTT would be fine as HomeAssistant is running 24/7.

**To Reproduce**
Working config without RTL433: 
```
[env:esp32dev-ble-pilight-somfy]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.decoder}
  ${libraries.ble}
  ${libraries.smartrc-cc1101-driver-lib}
  ${libraries.esppilight}
  ${libraries.somfy_remote}
build_flags =
  ${com-esp32.build_flags}
  '-DZgatewayPilight="Pilight"'
  '-DZactuatorSomfy="Somfy"'
  '-DZgatewayBT="BT"'
  '-DGateway_Name="OMG_ESP32_433"'
; *** RF Module Options ***
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
  '-DRF_MODULE_CS=5'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=12'   ; CC1101 pin GDO0
  '-DRF_MODULE_GDO2=27'   ; CC1101 pin GDO2
custom_description = ESP32 with BLE, Somfy and ESPilight
```

Problematic config with RTL433 added following multi rf example:
```
[env:esp32dev-ble-pilight-somfy-rtl]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.decoder}
  ${libraries.ble}
  ${libraries.smartrc-cc1101-driver-lib}
  ${libraries.rtl_433_ESP}
  ${libraries.esppilight}
  ${libraries.rc-switch}
  ${libraries.newremoteswitch}
  ${libraries.somfy_remote}
build_flags =
  ${com-esp32.build_flags}
  '-DvalueAsATopic=true'       ; MQTT topic includes model and device
  '-DGateway_Name="OMG_ESP32_433"'
  '-DCORE_DEBUG_LEVEL=4'
; *** 433 gateways ***
  '-DZgatewayRF="RF"'
  '-DZgatewayPilight="Pilight"'
  '-DZactuatorSomfy="Somfy"'
  '-DZgatewayRTL_433="rtl_433"'
  '-DDEMOD_DEBUG=true'         ; display signal debug info
; *** BT gateways ***
  '-DZgatewayBT="BT"'
; *** RF Module Options ***
  '-DZradioCC1101="CC1101"'
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
  '-DRF_MODULE_CS=5'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=12'   ; CC1101 pin GDO0
  '-DRF_MODULE_GDO2=27'   ; CC1101 pin GDO2
[... 188 lines cut ...]
                       ^~~~~~~~~~~~~~
/just/some/folder/OpenMQTTGateway/main/ZgatewayBT.ino: In function 'void MQTTtoBTAction(ArduinoJson::JsonObject&)':
/just/some/folder/OpenMQTTGateway/main/ZgatewayBT.ino:1357:26: error: 'TheengsDecoder' has not been declared
                          TheengsDecoder::BLE_ID_NUM::SBS1, 1);
                          ^~~~~~~~~~~~~~
/just/some/folder/OpenMQTTGateway/main/ZgatewayBT.ino:1406:24: error: 'TheengsDecoder' has not been declared
                        TheengsDecoder::BLE_ID_NUM::UNKNOWN_MODEL,
                        ^~~~~~~~~~~~~~
Compiling .pio/build/esp32dev-ble-pilight-somfy-rtl/lib37d/WiFiManager/WiFiManager.cpp.o
Archiving .pio/build/esp32dev-ble-pilight-somfy-rtl/lib58b/libWebServer.a
Indexing .pio/build/esp32dev-ble-pilight-somfy-rtl/lib58b/libWebServer.a
Compiling .pio/build/esp32dev-ble-pilight-somfy-rtl/lib31f/TheengsDecoder/decoder.cpp.o
Compiling .pio/build/esp32dev-ble-pilight-somfy-rtl/lib31f/TheengsDecoder/decoder_c.cpp.o
*** [.pio/build/esp32dev-ble-pilight-somfy-rtl/src/main.ino.cpp.o] Error 1
================================================================================================================= [FAILED] Took 9.63 seconds =================================================================================================================

```
Note: I you comment out '-DZgatewayBT="BT"' compile works again.

**Expected behavior**
Compile should run fine, maybe linking would fail due to used memory (as issue 1258 claims that might be a little much).

**Environment (please complete the following information):**
* OpenMQTTGateway latest development version
* PlatformIO Core, version 6.1.7

lachesis, 2023-06-14:
I believe this is because rtl_433_ESP defines its own decoder.h file which does something different than the one in TheengsDecoder. I am not sure how to resolve this conflict.

NorthernMan54, 2023-06-27:
Please note that the current resource requirements ( heap and CPU ) of BLE and RTL433 make this configuration not feasible.

closed 2023-09-09 by github-actions[bot]