1technophile/OpenMQTTGateway #1655: lilygo-rtl_433 crash after chacon remote control button pressed

Xenon69, 2023-05-23:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
lilygo-rtl_433 crash after chacon remote control button pressed

**To Reproduce**
Steps to reproduce the behavior:
1. compile and upload lilygo-rtl_433
2. press a button on a Chacon remote
3. the device reboots

**Expected behavior**
a MQTT message published with remote button details

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

NorthernMan54, 2023-05-23:
In order to further troubleshoot the issue, is it possible to share the serial output from the device when the reboot occurs.  And to the output easier to debug, is it possible to add this setting to your platformio build environment ?

```
monitor_filters = esp32_exception_decoder
```

This will decode where the exception occurs within the codebase.

Tks

Xenon69, 2023-05-30:
Hello,

here is the trace 'Guru meditation':

N: Send on /RTL_433toMQTT/Acurite-606TX/207 msg {"model":"Acurite-606TX","id":207,"battery_ok":1,"temperature_C":15.8,"mic":"CHECKSUM","protocol":"Acurite 606TX Temperature Sensor","rssi":-64,"duration":818996}
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (rtl_433_Decoder) 
Core  1 register dump:
PC      : 0x4008df07  PS      : 0x00060636  A0      : 0x4008dd16  A1      : 0x3ffe1af0  
A2      : 0x00000008  A3      : 0x4008aa61  A4      : 0x00000008  A5      : 0x3ffe1bc0  
A6      : 0x00000008  A7      : 0x3ffe1dc0  A8      : 0x3ffe1d9c  A9      : 0x00000008  
A10     : 0x3ffdfab0  A11     : 0x3ffe1ec0  A12     : 0x40084f92  A13     : 0x00005000  
A14     : 0x00000008  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008aa61  LEND    : 0x4008aa6e  LCOUNT  : 0xfffffffe  

Backtrace: 0x4008df04:0x3ffe1af0 0x4008dd13:0x3ffe1ec0 0x401c6a7a:0x3ffe1f50 0x40123539:0x3ffe1f90 0x400fa393:0x3ffe1fe0 0x400fa645:0x3ffe2000 0x400fa6a1:0x3ffe2020 0x400fa77b:0x3ffe2040 0x40122bda:0x3ffe2090 0x401d0a7a:0x3ffe20d0 0x401122fd:0x3ffe20f0 0x401241b3:0x3ffe3b30 0x40124b95:0x3ffe3b50 0x40122f26:0x3ffe3b90 0x400f8dff:0x3ffe3bc0

  #0  0x4008df04:0x3ffe1af0 in _xt_context_save at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_context.S:126
  #1  0x4008dd13:0x3ffe1ec0 in _frxt_int_enter at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:112
  #2  0x401c6a7a:0x3ffe1f50 in vsnprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
  #3  0x40123539:0x3ffe1f90 in abuf_printf at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/abuf.c:59
  #4  0x400fa393:0x3ffe1fe0 in format_jsons_int at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/data.c:540
  #5  0x400fa645:0x3ffe2000 in print_value at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/data.c:398
  #6  0x400fa6a1:0x3ffe2020 in format_jsons_object at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/data.c:450
  #7  0x400fa77b:0x3ffe2040 in data_print_jsons at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/data.c:557
  #8  0x40122bda:0x3ffe2090 in data_acquired_handler at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/r_api.c:977
  #9  0x401d0a7a:0x3ffe20d0 in decoder_output_data at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/decoder_util.c:41       
  #10 0x401122fd:0x3ffe20f0 in nexa_callback at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/devices/nexa.c:68 (discriminator 4)
  #11 0x401241b3:0x3ffe3b30 in account_event at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/pulse_slicer.c:34
  #12 0x40124b95:0x3ffe3b50 in pulse_slicer_ppm at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/pulse_slicer.c:303
  #13 0x40122f26:0x3ffe3b90 in run_ook_demods at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/rtl_433/r_api.c:543
  #14 0x400f8dff:0x3ffe3bc0 in rtl_433_DecoderTask(void*) at .pio/libdeps/lilygo-rtl_433/rtl_433_ESP/src/signalDecoder.cpp:482    

ELF file SHA256: 52312a0d32e452e7

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
N:
************* WELCOME TO OpenMQTTGateway **************

Have a nice day!!

Thank you

Xenon69, 2023-05-31:
Thanks, now it works.

I changed 8500 to 10000 in the code below, as OOK_MODULATION is defined to true:
`#  if OOK_MODULATION
#    define rtl_433_Decoder_Stack 10000  <-----------Here
#  else
#    define rtl_433_Decoder_Stack 20000
#  endif
`
Will that be updated in the next version?

Thank you again.

NorthernMan54, 2023-06-27:
Resolved with release 1.6.0

closed 2023-06-27 by NorthernMan54