1technophile/OpenMQTTGateway #1656: MopekaGas Sensor Pro: Add x/y-Position

Joey-1970, 2023-05-25:
Hello!

Thank you very much for this great app.

Is it possible to add the x/x-Postion to the MQTT-String?
MA MA HW BAT TEMP Q  Q  MAC MAC MAC XACEL YACEL
  1    2     3     4       5      6  7    8        9      10      11         12
I mean the last two bytes of the data field...

It is wrote in TwosComplement...

Thank you

Joachim

DigiH, 2023-05-25:
Hi @Joey-1970 

Are you able to make your own test build in PlatformIO with an updated custom decoder branch?

Joey-1970, 2023-05-25:
Hello @DigiH,

thank you for your reply.

I think I'm not able to do it myself. If you show me the original code (I don't find it), maybe I can write something here with more details...

Joachim

DigiH, 2023-05-25:
Nothing major - In the `platformio.ini` file, just change the current line 156

`decoder = https://github.com/theengs/decoder.git`

to

`decoder = https://github.com/DigiH/decoder#mopeka`

then rebuild and upload to your gateway.

Please let us know if it corresponds with the x and y position in the Mopeka app.

Thanks

Joey-1970, 2023-05-25:
...I find something here:
https://github.com/DigiH/decoder/blob/development/src/devices/Mopeka_json.h
...my try:

 x must be beginn at manufacturerdata 16, y at 18...
      "xacel":{
         "decoder":["value_from_hex_data", "manufacturerdata", 16, 2, ....
         "post_proc":[twoscomplement...

 "yacel":{
         "decoder":["value_from_hex_data", "manufacturerdata", 18, 2, ....
         "post_proc":[twoscomplement...

I fear this ist not my program language...maybe some other can complete it and merge it?
Sorry...

Joachim

DigiH, 2023-05-25:
No worries, with the manufacturerdata including the leading two company ID octets at the beginning, there must be four indexes added to your suggestion, as I submitted it to the test decoder file

https://github.com/DigiH/decoder/blob/mopeka/src/devices/Mopeka_json.h#L53

 and the _twoscomplement_ is being defined with the booleans in the "decoder" section.

You can also have a look at the extended test cases in the PR to see if the raw data would match your results as well.

**https://github.com/theengs/decoder/pull/340/files**

Joey-1970, 2023-05-26:
THANK YOU VERY MUCH!

DigiH, 2023-05-26:
@Joey-1970 You're welcome 😃 

You can now download and install pre-built binaries of the latest OpenMQTTGateway development version, with the extended Mopeka decoder included, at

https://docs.openmqttgateway.com/dev/upload/web-install.html

DigiH, 2023-05-30:
Hi @Joey-1970 were you able to test and confirm with the above test development builds?

If all is fine please close this issue.

Thanks

DigiH, 2023-05-31:
Hi Joachim,

Yes, `esp32dev-ble-mqtt-undecoded` only send data undecoded to be decoded by a central Theengs Gateway installation.

The binary you want for direct decoding and publishing to MQTT on your ESP32 is

**esp32dev-ble**

Let us know how you get on.

Thanks

Joey-1970, 2023-05-31:
Yes, yes, yes!!!

Look all good!  :-)

31.05.2023, 21:11:14 |      MQTT:RX:PUBLISH | Topic: home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/FD84BADB79C2, Payload: {"id":"FD:84:BA:DB:79:C2","rssi":-85,"brand":"Mopeka","model":"Pro Check Sensor","model_id":"M1017","type":"UNIQ","tempc":18,"tempf":64.4,"lvl_cm":8.571037,"lvl_in":3.374424,"sync":false,"volt":2.59375,"batt":60.57692,"quality":2,"accx":45,"accy":-46}

Many thanks!

closed 2023-05-31 by DigiH