1technophile/OpenMQTTGateway #1659: Flash (web) on Linux Mint with Chrome

jerome6994, 2023-05-27:
Good morning
I come to ask a question in order to help me flash an esp32 on chrome for BLE presence on Home Assistant.
I am on Linux Mint recently and by connecting the ESP32 to the computer I do not see the ESP on a serial port by clicking on Install!
I see ttys0, ttys1 ttysX...... but never ttyUSB0 while I have it on my terminal
[image]
I don't see where it can come from!
[image]
Thanks

ColinShorts, 2023-05-27:
You might not be part of the dialup group, check with `groups` (run as your
normal user without sudo). You can check the group ownership of the device
node with `ls -l /dev/ttyUSB*`.

Should be as simple as `sudo gpasswd -a <username> dialup`, then logout and
log back in again.

jerome6994, 2023-05-28:
Good morning
Thanks for the help I actually did an ls -l /dev/ttyUSB* and I didn't have access!
I did :
```
ls -l /dev/ttyUSB0 --> Failed
sudo adduser <USER> dialout
sudo usermod -a -G dialout <USER>
sudo reboot
ls -l /dev/ttyUSB0 --> OK
```

closed 2023-09-09 by github-actions[bot]