1technophile/OpenMQTTGateway #1660: 868mhz weather station

artl3x, 2023-05-28:
hello just changing the frequency from 433 to 868 does nothing. Is there anything I can do or is the modulation unsupported

NorthernMan54, 2023-05-28:
Which board/transceiver module and what device are you trying to receive?

If it is using FSK modulation it isn’t supported

artl3x, 2023-05-28:
I am using lilygo lora 32 868mhz:

€19.67 | LILYGO® TTGO LoRa32 V2.1_1.6 Version 433/868/915Mhz ESP32 LoRa OLED 0.96 Inch SD Card Bluetooth WIFI Wireless Module ESP-32  SMA
https://a.aliexpress.com/_mOC6XgU

Weather station is garni 750 : 

https://www.garnitechnology.com/garni-750

I cant find any other imformation than transmiting frequency 868mhz... 

Then i will have to wait  :(

NorthernMan54, 2023-05-28:
I googled 'garni 750 rtl_433' and did not have any hits at all.  So no clues in regards to the type of modulation.

baden03, 2023-06-12:
I also am trying to connect to a weather station broadcasting on the 868mhz.

I first bought the 433mzh LilyGo board and was able to receive many sensors from around the neighbourhood.
Then I got the same 868mhz version linked above. I was able to send [the MQTT command](https://docs.openmqttgateway.com/use/rf.html#rtl-433-device-decoders) to change the mhz from 443 to 868, but am not receiving anything.

I then tried to compile OpenMQTTGateway with the mhz set to 868 as default as this reverts back to 433 on restart, but could not find any mhz flags in any of the files, so gave up on that for now.

I am able to receive info on 868 using a standard SDR dongle, so my only option now seems to be to play around with [the different transceiver modules](https://docs.openmqttgateway.com/use/rf.html#changing-active-receiver-modules)

neoatomic, 2023-10-07:
@baden03 Having exactly the same issue as you describe. All fine with 433mhz board and noting on the 868mhz board.
And when i use rtl_433 and a SDR usb stick all sensors are received fine on 868mhz.
But i did notice something, when i set the frequency to 868mhz the symbol-rate is set to 10000k instead of 2500k .

Wondering how this works with the OMG firmware, does it auto change the symbol-rate when changing to the 868mhz frequency range? If not it could explain bad reception or devices not being found.

@1technophile @NorthernMan54 Is there a symbol-rate setting that i can use during building?

NorthernMan54, 2023-10-11:
The rtl_433_ESP receiver logic is signal level triggered and not based on sampling.  So there is no sampling rate setting.

neoatomic, 2023-10-28:
@NorthernMan54 Thank you.
I have the LILYGO LoRa32 V2.1 on 868mhz working with OOK signals. But when i add ``` '-DOOK_MODULATION=false' ``` i only receive undecoded signals. For troubleshooting i added ``` '-DRTL_ANALYZER=true' ``` and log result is ```Guessing modulation: No clue... ```
For testing i used a Ecowitt WH51 (868mhz) Soil Moisture Sensor that should be supported according to the FSK Signal Device Decoders list number 19. So i'm currently out of ideas how to solve this... 

Log : https://pastebin.com/Hiub2g9s 
(Removed the antenna and placed WH51 next to the LoRa32, than inserted the battery. Have more logs, also of the other weather station sensors if needed.)

NorthernMan54, 2023-10-30:
That is the same setup I have, I just did not disconnect the antenna.

Did you see this tuning discussion ? - https://github.com/NorthernMan54/rtl_433_ESP/issues/87

neoatomic, 2023-11-01:
Changing the ```MINIMUM_SIGNAL_LENGTH``` setting didn't work for me. 
But with the help of that discussion i found that my LILYGO gives complete random codes for all my sensors when i compare them to my RTL_433 Windows setup.

There the Fineoffset-WH51 gives the below code and barely changes over time. :
```
codes     : {181}aaaaaaaaaaa2dd4510dc1f7117f00f834ffffff8958208
codes     : {181}aaaaaaaaaaa2dd4510dc1f7117f00f834ffffff8958210
codes     : {180}aaaaaaaaaaa2dd4510dc1f7b17f0ef87dffffff9d6342
```

On my LILYGO i get only random codes that make no sense, and non of the strings do even have a similarity to the above :

```codes     : [{260}4514a2a8a1113218004610a42415113b088be30c4118ce8b1151354290d4a4420]  ```

Output with RTL_433 on Windows : https://paste.07f.de/?3ebafe527ecda56a#13s9MwwDvt4tfqqG34r5efcrBijEgskDKmC6JAD9zhbn

I can't wrap my head around the fact that OOK is working fine, but as soon as i switch to FSK it doesn't pickup a single device. Not sure if someone is willing to build a test firmware file for the LILYGO LoRa32 V2.1 868mhz with FSK? Just to exclude that that there is a problem with my PlatformIO IDE installation. 

In the process i also made some captures of my sensors, think some are not yet in the rtl_433 example repo so will try to upload them.

1technophile, 2024-01-31:
Closing as the wh51 is now supported by the v1.7.0 with fsk binary

closed 2024-01-31 by 1technophile