1technophile/OpenMQTTGateway #1662: A stack overflow in task rtl_433_Decoder

hb9eue, 2023-05-31:
Hi Technophile

I managed to capture a spontaneous reboot:

Board: Helltec ESP32 Lora V2
Branch: development at: commit b3a6559b876379ab9b1e859ad03dd882abad85a5

```
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/98008 msg {"model":"LaCrosse-TX141W","id":98008,"channel":0,"battery_ok":0,"temperature_C":22.9,"humidity":43,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, 
TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-54,"duration":640996}
N: Send on /RTL_433toMQTT/LaCrosse-TX141W/0/98008 msg {"model":"LaCrosse-TX141W","id":98008,"channel":0,"battery_ok":0,"wind_avg_km_h":4.7,"wind_dir_deg":0,"test":0,"mic":"CRC","protocol":"LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3
, TX141W, TX145wsdth, (TFA, ORIA) sensor","rssi":-54,"duration":640996}

***ERROR*** A stack overflow in task rtl_433_Decoder has been detected.

Backtrace: 0x40083c69:0x3ffe07f0 0x4008dce9:0x3ffe0810 0x400913c5:0x3ffe0830 0x4008f89f:0x3ffe08b0 0x4008ddf8:0x3ffe08e0 0x4008dda8:0x00000000 |<-CORRUPTED

```

I have about one or two hours of serial log prior to the reboot. Can also attach if this would help to figure the cause

-Benoît-

NorthernMan54, 2023-05-31:
What build/version are you running?

NorthernMan54, 2023-05-31:
Duplicate of #1655

NorthernMan54, 2023-06-27:
Resolved with release 1.6.0