1technophile/OpenMQTTGateway #1664: Support for bthome.io?

Heiko-L, 2023-06-01:
There is a new advertising format in town:

https://bthome.io/

It integrates directly with home assistant so I could uninstall Passive BLE monitor integration for sensors in range of my raspberry pi.

I would like to use this format with pvvx flashed xiaomi devices

https://github.com/pvvx/ATC_MiThermometer#readme

Heiko-L, 2023-06-01:
Thank you
I changed the title accordingly

koenvervloesem, 2023-06-01:
Can you show an example of the advertised data by this device?

Heiko-L, 2023-06-01:
I will flash a sensor with the new firmware and dump it.
Is it also possible to process encrypted data (bthome supports AES with pre-shared key https://bthome.io/encryption/)?

koenvervloesem, 2023-06-01:
Encryption is not supported yet, we'll need to look into it how to implement that first.

DigiH, 2023-06-01:
Hi @Heiko-L 

While I generally understand your request for BTHome decoding in OpenMQTTGateway, and with your comment

> It integrates directly with home assistant so I could uninstall Passive BLE monitor integration for sensors in range of my raspberry pi.

Using the native PVVX protocol should already be supported by the HA integration, for devices near your RPi Bluetooth adapter, as well as with OpenMQTTGateway for devices further away.

So apart from a protocol fondness the scenario should already be working nicely.

Heiko-L, 2023-06-01:
> Using the native PVVX protocol should already be supported by the HA integration

Tell me more. I was only able to add my pvvx flashed LYWSD03MMCs with pvvx advertisements via HACS->Passive BLE monitor.

DigiH, 2023-06-01:
Admittedly, I do not have any LYWSD03MMCs, hence the _should_ in the above statement ;) but from what I understood with the shift from ble-monitor to the native HA BLE integration, was that the same format decodings would be ported over, so that the [xiaomi-ble](https://github.com/Bluetooth-Devices/xiaomi-ble/tree/main) part would also be able to decode the Xiaomi (MiBeacon), ATC1441 and PVVX native protocols along with the BTHome protocol.

Please correct me if this is not the case, as I will then have to totally reevaluate my understanding of the native HA BLE integration.

Heiko-L, 2023-06-01:
If I try to add the Xiaomi->Xiaomi BLE or BTHome integration I get "No devices found on the network".
Or do I have to disable Passive BLE monitor first?

DigiH, 2023-06-01:
I'm afraid I really am not sure. Or could be really that the native HA BLE integration only does support BTHome, which would be a surprise for me. Also then the [Xiaomi BLE](https://www.home-assistant.io/integrations/xiaomi_ble/) page also only seem to talk about the MiBeacon protocol, not even BTHome. Even more confusing now :(

DigiH, 2023-06-01:
This seems to indicate that it should work with the PVVX protocol with the Xiaomi HA integration, but

> … it works but not as well as Passive BLE Monitor. If your BLE signal strength is low you get gaps in your data, rather than using the previous temperature the sensors show “unavailable” or similar. I have both running, but I only use the data from Passive BLE Monitor, which also provide signal strength.
> …
> As of HA 2022.09 and later, just wait for BTHome to discover the sensors, give you a notification, at which point you can add them to HA.

https://community.home-assistant.io/t/home-asssistant-xiaomi-mijia-lywsd03mmc-temperature-and-humidity-sensor-tutorial-september-2022/456403

So now being mid 2023 I would assume it should still work as described, and possibly has been improved since then.

Heiko-L, 2023-06-02:
You are right- I also read a thread where Ernst79 stated that he implemented the ATC and pvvx format in BTHome, so my underlying problem seems to be a non working Bluetooth/BTHome integration so I opened an issue in home assistant for this.

I hope I have the time to get a "BTHome" dump of a LYWSD03MMC this weekend...

DigiH, 2023-06-02:
I'm also very curious as to what the issue might be, and the resolution.

Nonetheless, as @koenvervloesem pointed out above, the BTHome protocol and decryption is something we're also looking into for the future.

Heiko-L, 2023-06-06:
> I'm also very curious as to what the issue might be, and the resolution.

pvvx advertisement format seems to be dropped, as soon I flashed the LYWSD03MMC to use BTHome advertisements BTHome created a new device in HA.

> Can you show an example of the advertised data by this device?

It is using the legacy format: https://bthome.io/v1/ with packet ids:

020106 11 161C18 020045 23024107 0303C316 020157 0B 094154435F43656C6C6172
020106 0D 161C18 020044 021001   030C000B        0B 094154435F43656C6C6172

DigiH, 2023-06-06:
I have seen you closing the other issue ;) real shame though, that the native integration seems to have dropped the PVVX broadcast decoding, while ble-monitor still supports it, and also seemingly only then supporting the **legacy** format for the LYWSD03MMC. Also have seen someone else having he same issue in the other issue.

So with the legacy format page also stating

> This is the documentation for BTHome v1. This API is considered legacy and should not be used for new projects. 

Unfortunately I then don't currently see much point in us putting effort in adding this format, and concentrate on the current format in the future, which PVVX will likely switch to sooner or later as well; or does it already have the option for both BTHome formats?

Heiko-L, 2023-06-06:
> Unfortunately I then don't currently see much point in us putting effort in adding this format, and concentrate on the current format in the future, which PVVX will likely switch to sooner or later as well; or does it already have the option for both BTHome formats?

An issue is opened already:
https://github.com/pvvx/ATC_MiThermometer/issues/350

lachesis, 2023-07-02:
PVVX has declined to implement BTHome v2. I am using my sensors with the custom PVVX format, but I configured them by following [this HomeAssistant guide](https://community.home-assistant.io/t/home-asssistant-xiaomi-mijia-lywsd03mmc-temperature-and-humidity-sensor-tutorial-september-2022/456403) which suggested BTHome v1 for HA compatibility. If anyone else finds themselves here trying to understand why their sensors are not working, just switch back to the pvvx format.

closed 2023-09-09 by github-actions[bot]

vasil-github, 2024-02-22:
I know I'm little bit late but I wanted to share one more sensor.
I did but tuya TH05 temp and hum BT sensor and fished them with custom firmware that uses BThome protocol/advertisement. The devices are detected by OMG when the following appearing in the MQTT broker:
{"id":"38:1F:8D:6D:F4:66","mac_type":0,"adv_type":0,"name":"TH05-BT-Laundry","rssi":-82,"distance":11.4949,"servicedata":"40000f015c02c00603dd120c6f0b","servicedatauuid":"0xfcd2"}

After some help this massage is translated to:
uuid: 0xfcd2 - BTHome
data: "40000f015c02c00603dd120c6f0b"
40 - BTHome V2
000F - Msg count 15
015c - Battery 0x5c = 92%
02c006 - Temp: 0x06C0 = 17.28 C
03dd12 - Humi: 0x12DD = 48.29 %
0c6f0b - Vbat: 0x0b6f = 2927 mV
Do you have some advice how to do the translation or is it possible to support this device in future releases?
The devine is Tuya TH-05

plambrechtsen, 2025-08-03:
Just wrote a [PR](https://github.com/1technophile/OpenMQTTGateway/pull/2219) adding BTHome v2 support including encrypted frames on the  LYWSD03MMC sensors with PVVX. If you wanted to try it out.