1technophile/OpenMQTTGateway #1665: CC1101 setSpi call if SPI pins are set

Mirdinus, 2023-06-06:
Hi All

**Is your feature request related to a problem? Please describe.**
I'm using [M5 Stamp-C3](https://docs.m5stack.com/en/core/stamp_c3) (ESP32-C3) and cannot use the default pins configuration as some of the pins are not available on my board
[Pinout](https://static-cdn.m5stack.com/resource/docs/products/core/stamp_c3/stamp_c3_09.webp)

**Describe the solution you'd like**
Add `ELECHOUSE_cc1101.setSPI(SCK, MISO, MOSI, CS);` call just above `ELECHOUSE_cc1101.Init();` if SPI pins are set to non-default
[All mentions](https://github.com/search?q=repo%3A1technophile%2FOpenMQTTGateway+ELECHOUSE_cc1101.Init%28%29%3B&type=code)

**Describe alternatives you've considered**
Only other solution is not using boards with limited pinout

**Additional context**
I tested it with the library and It works fine. I plan to make a PR with the changes

Mirdinus, 2023-06-06:
Might be related ti this [part](https://github.com/1technophile/OpenMQTTGateway/blob/15416e99b1a0f8ab433d8c8a1101db6f2d62ee3c/environments.ini#L630) of environments configuration

```
;  '-DRF_MODULE_MOSI=23'
;  '-DRF_MODULE_MISO=19'
;  '-DRF_MODULE_SCK=18'
```

efuturetoday, 2023-07-19:
+1 it is mandatory to support WT32-ETH01 as the SPI default pinout can not be used.

Mirdinus, 2023-07-19:
You can already 'fix' it for yourself by inserting
```
ELECHOUSE_cc1101.setSpiPin(RF_MODULE_SCK, RF_MODULE_MISO, RF_MODULE_MOSI, RF_MODULE_CS);
```
above any `ELECHOUSE_cc1101.Init();` call

efuturetoday, 2023-07-19:
Thank you - I already solved it this way. I will create a PR to introduce a new board and this change, so others can just use the defines to configure the SPI interface.

efuturetoday, 2023-07-25:
PR can be merged

closed 2023-09-09 by github-actions[bot]