1technophile/OpenMQTTGateway #1669: Not receiving any RF data using an ESP32 with MX-RM-5v and FS1000A

Heisenberg2980, 2023-06-07:
**Describe the bug**
After following the steps and connecting the ESP32 board to MX-RM-5v and FS1000A according to the instructions here https://docs.openmqttgateway.com/setitup/rf.html#esp32-hardware-setup the Bluetooth data is received by the MQTT broker but no RF data (433Mhz) is received. I have flashed the ESP32 using esp32dev-ble, and I use MQTT Explorer to analyse the MQTT data.

**Expected behavior**
I would expect to receive messages into the topic home/OpenMQTTGateway_ESP32_BLE containing 433Mhz data.

**Environment (please complete the following information):**
* OpenMQTTGateway version used: v1.5.1

NorthernMan54, 2023-06-27:
The build you are using only supports receiving Bluetooth data, to receive RF you need to select a different build.  To receive both you will need to create a custom environment.   Please note that due to the resource requirements needed to receive and decode both RF and Bluetooth messages this may not work.

Heisenberg2980, 2023-06-27:
Thank you very much, I hadn´t realised there was a esp32dev-rf option. After uploading this firmware I can see in the logs the esp32 is now setup for RF:
```
N: RF_EMITTER_GPIO: 12 
N: RF_RECEIVER_GPIO: 27 
N: Switching to RF Receiver
N: OpenMQTTGateway modules: ["WebUI","RF"]
``` 
Unfortunatelly I am still not receiving any RF data, even when pressing different buttons in a 433.92mhz blind remote controller near the receiver

closed 2023-09-08 by github-actions[bot]