1technophile/OpenMQTTGateway #1678: OTA upgrade to 1.6.0 fails with backtrace/unhandled exception

jpmorrison, 2023-06-12:
Running 1.5.1. The upgrade version is displayed in Home Assistant - clicking install tries and fails to upgrade. 

Logs
```

N: [ MQTT->OMG ]: {"version":"latest"}
N: Using system OTA url with latest version https://github.com/1technophile/OpenMQTTGateway/releases/
download/v1.6.0/esp32dev-multi_receiver-firmware.bin
W: Starting firmware update
N: Send on /RLStoMQTT msg {"release_summary":"Update in progress ..."}
N: using stored cert
rtl_433_ESP(7): Average RSSI Signal -87 dbm, adjusted RSSI Threshold -78, samples 50000
N: HTTP_UPDATE_OK
N: Send on /RLStoMQTT msg {"release_summary":"Update success !","installed_version":"v1.6.0"}
W: Client not connected, aborting the publication
{
  "mqtt_server": "x.x.x.x",
  "mqtt_port": "1883",
  "mqtt_user": "your_username",
  "mqtt_pass": "your_password",
  "mqtt_topic": "home/",
  "gateway_name": "OpenMQTTGateway",
  "mqtt_broker_secure": false,
  "mqtt_broker_cert": "",
[removed]
  "ota_pass": "OTAPASSWORD"
}ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x4012c2ac  PS      : 0x00060930  A0      : 0x80132c50  A1      : 0x3ffe3b40  
A2      : 0x3f41bd88  A3      : 0x3f423ffc  A4      : 0x3ffcd1f0  A5      : 0x00060f23  
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x8012c2ac  A9      : 0x3ffe3b20  
A10     : 0x3ffbab44  A11     : 0x3ff440b0  A12     : 0x00000000  A13     : 0x00060f23  
A14     : 0x007bed94  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x00000003  
EXCVADDR: 0x3f41bd88  LBEG    : 0x4008aa1d  LEND    : 0x4008aa2d  LCOUNT  : 0xfffffffe  

Backtrace: 0x4012c2a9:0x3ffe3b40 0x40132c4d:0x3ffe3b60 0x4012c25c:0x3ffe3b80 0x4012a8f2:0x3ffe3ba0 0x4012acbd:0x3ffe3bc0 0x400e302d:0x3ffe3be0 0x4014a787:0x3ffe3c10 0x40083206:0x3ffe3c40 0x400792ba:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 4e2584206ae5a1c6

E (786) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
N: 
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v1.5.1
```

1technophile, 2023-06-12:
Thanks for sharing this, what kind of board do you have ?

jpmorrison, 2023-06-13:
It’s an ESP-WROOM32 in an UNO form factor/pin outs. Is there anything from the boot or MQTT logs that would help ID it?

1technophile, 2023-06-15:
it seems to update:
```
N: Send on /RLStoMQTT msg {"release_summary":"Update success !","installed_version":"v1.6.0"}
W: Client not connected, aborting the publication
```
But fail on start, trying to reproduce

1technophile, 2023-06-15:
Reproducing, does not seem to be OTA related.
Investigating

jpmorrison, 2023-06-17:
With esp32dev-multi_receiver-firmware.bin via serial I have to erase flash or it won't install.
After esp32dev-multi_receiver it just gets stuck in a boot loop.

I was able to install/boot esp32dev-rtl433. I was able to configure/reconnect to wifi/mqtt/Home Assistant but it stops working when I move to a previous location/AP. It reverts to AP mode instead of joining the SSID. Looks like a separate issue.

Backtrace: 0x4012c2a9:0x3ffe3b40 0x40132c4d:0x3ffe3b60 0x4012c25c:0x3ffe3b80 0x4012a8f2:0x3ffe3ba0 0x4012acbd:0x3ffe3bc0 0x400e302d:0x3ffe3be0 0x4014a787:0x3ffe3c10 0x40083206:0x3ffe3c40 0x400792ba:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 4e2584206ae5a1c6

E (726) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x4012c2ac  PS      : 0x00060930  A0      : 0x80132c50  A1      : 0x3ffe3b40
A2      : 0x3f41bd88  A3      : 0x3f423ffc  A4      : 0x0000000c  A5      : 0x00060f23
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x8012c2ac  A9      : 0x3ffe3b20
A10     : 0x3ffbab44  A11     : 0xfffc1fff  A12     : 0x00060f20  A13     : 0x00060f23
A14     : 0x007bed94  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f41bd88  LBEG    : 0x4008aa1d  LEND    : 0x4008aa2d  LCOUNT  : 0xfffffffe

efuturetoday, 2023-07-20:
I also had also the same issue on WT32-ETH01 when using the multi_receiver config (adapted to the new board of).
So I just using RF now and it is working fine. I guess there is some issue when using all the RF protocols together with 1.6.0

1technophile, 2023-09-13:
Duplicate of #1741

closed 2023-09-13 by 1technophile