1technophile/OpenMQTTGateway #1681: Home Assistant Warning: "Template variable warning: 'dict object' has no attribute 'displaymetric' when rendering '{{ value_json.displaymetric }}'

Snuffy2, 2023-06-13:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Since upgrading to v1.6.0, I've been getting a lot of these errors in my Home Assistant logs.

```
2023-06-11 21:33:44.109 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'displaymetric' when rendering '{{ value_json.displaymetric }}'
```
[image]

I believe it is coming from OMG and may be related to #1504 although I believe that was added in v1.5.0 and seems to be [this line](https://github.com/1technophile/OpenMQTTGateway/blob/e27e055bf892a426c77cc2753d7e47724e27d70e/main/ZmqttDiscovery.ino#L511)

I'm running a Heltec WiFi LoRa 32 (V2.1) running RTL_433 which I believe uses the SSD1306 Display. I do have the display turned off.

[image]

I do not recall seeing these errors in Home Assistant with OMG v1.5.1

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development): 1.6
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)

Snuffy2, 2023-06-14:
I turned on the Display to see if that had any effect and it didn't seem to change anything. The error is still appearing. Another thing I noticed is that displayMetric is not listed as an item in the SSD1306toMQTT JSON when looking with MQTT Explorer. It is listed under WebUItoMQTT and changes when I check/uncheck the Display Metric box in the WebUI.

[image]

DigiH, 2023-06-14:
This looks like a legacy discovery for displaymetric for SSD1306, from before it was moved to the WebUI. Needs to be removed for SSD1306.

DigiH, 2023-06-30:
Hi @Snuffy2 

would you be able to verify with a test build within the next 8 hours if I start the build now?

DigiH, 2023-07-02:
@Snuffy2 

A test build which should fix the warning messages can be installed at

https://docs.openmqttgateway.com/dev/upload/web-install.html

If you could also check the correct switching between metric and imperial units, that would be great.

Snuffy2, 2023-07-07:
I installed version 9d8a96 this evening.

- [x] The Template variable warning is no longer showing
- [ ] I unchecked Display Metric in the WebUI. The change was shown in MQTT Explorer but the temperatures continue to show in °C
[image]
[image]

DigiH, 2023-07-07:
Thanks for testing @Snuffy2 

The displayMetric setting only applies to the alternating device property unit display on the WebUI main page and on the Heltec/LilyGo internal OLED display.

The rtl_433 published MQTT messages will always have temperature_C, which can/should be auto converted to Fahrenheit if so defined in HA.

I was more curious as to whether the switching also works now in the HomeAssistant Gateway Control for the different display on the Heltec and WebUI.

Snuffy2, 2023-07-07:
Gotcha. The temperature does show correctly in °F in the WebUI when I uncheck Display Metric.

DigiH, 2023-07-08:
Thanks for the confirmations.

Closing

closed 2023-07-08 by DigiH