1technophile/OpenMQTTGateway #1684: Missing Environments on  (Option 1) Upload from the web

usernamegeneratorFTW, 2023-06-14:
**Describe the bug**
On the https://docs.openmqttgateway.com/upload/web-install.html  website there are 67 Environments/ Boards listed in the table which are available via web flash. however, in the dropdown list i only count 62, which means that 5 are missing.
Among the missing boards is the lolin_c3_mini board which I am trying to get running, but fail because no AP is visible after the flash.
To eliminate me as a source of error for this it would be great if the lolin_c3_mini would be flashable on the web.

1technophile, 2023-06-14:
Fixed, you should have it in the development builds tomorrow:
https://docs.openmqttgateway.com/dev/

closed 2023-06-14 by 1technophile

usernamegeneratorFTW, 2023-06-14:
thank you very much

PR #1685 (2023-06-14): [CI] Add new boards into web upload release production and dev
## Description:
Fix #1684

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).