1technophile/OpenMQTTGateway #1689: WiFi not connecting after upgrade to 1.6.0

jpmorrison, 2023-06-17:
ESP32 running 1.5.1 connects to SSID on any channel. I have two APs, set to non overlapping channels 1 and 11.

After upgrading to 1.6.0 and configuring SSID through the captive portal it connects to AP on channel 1. When I move the ESP32 to a new location it will not connect to WiFi on the AP set to channel 11.

Workaround: on the second AP I changed it to channel 1 and it can connect. This wasn't needed in 1.5.1. OMG should be able to connect to an SSID on any available channel.

1technophile, 2023-06-22:
This is strange. I don't see what could have been changed between those two versions causing a change in Wifi connection behavior.
Note that if it was between v1.4.0 and v1.5.1 there was an update of the platform.

Which environment are you using, and what are your routers models?

jpmorrison, 2023-06-23:
It was working with 1.5.1 esp32dev-multi_receiver. I was unable to run 1.6.0 multi_receiver due to a boot loop but esp32dev-multi_receiver boots. These are release binaries, I didn't compile them myself - is that what you mean by environment?

With 1.6 it kept reverting to wifi setup/captive portal mode until I changed the rf channel.
It's like it remembers the channel it was setup on and won't scan/roam to the other channel.

WiFi AP is a Cisco - I had to change to channel 1 when I moved it downstairs. The other wifi is Technicolor /CommScope according to the Cable company.  Both are 2.4GHz g/n, 20 Mhz, WPA2-PSK/AES.

slysiou, 2023-06-24:
I also get a boot loop with 1.6 esp32dev-multi_receiver (direct flash from web or compiled with PlatformIO). 
Anyone with the same experience ?

franki29, 2023-06-24:
Hi, I have same problem too.
Using a older version it is working
Br. Frank

ilgrank, 2023-08-02:
Any update on the matter  @franki29   @jpmorrison   @slysiou ?

franki29, 2023-08-04:
I am using OMG_rtl_433_ESP , this works without problem

closed 2023-09-12 by github-actions[bot]