1technophile/OpenMQTTGateway #1690: Reboot after esp32dev-ir v1.6.0 web install fails.

NorbertHD, 2023-06-18:
After a successful web install of esp32dev-ir v1.6.0 on an ESP32 D1 mini, the board doesn't boot.
The console keeps repeating this message:

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
ets Jun  8 2016 00:22:57

1technophile, 2023-06-23:
We are very close to the max available flash of the ESP32 with this environment and its current partition scheme.
Changing the scheme to `board_build.partitions = min_spiffs.csv` gives more space available:
RAM:   [==        ]  18.6% (used 61072 bytes from 327680 bytes)
Flash: [=======   ]  66.6% (used 1310009 bytes from 1966080 bytes)

Whereas before:
RAM:   [==        ]  18.6% (used 61072 bytes from 327680 bytes)
Flash: [==========]  99.9% (used 1310009 bytes from 1310720 bytes)

And the board does not reboot anymore.

Fixed below
https://github.com/1technophile/OpenMQTTGateway/pull/1697

1technophile, 2023-06-23:
Merged, will be available in the web portal for development tomorrow:
https://docs.openmqttgateway.com/dev/upload/web-install.html

PR #1697 (2023-06-23): [IR] Increase app flash size by using min_spiffs.csv
## Description:
fix #1690
avoid ESP32 not starting with IR environment

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
commit 7b3b14e2 (2023-06-23): [IR] Increase app flash size by using min_spiffs.csv (#1697)

fix #1690
avoid ESP32 not starting with IR environment
closed 2023-06-23 by 1technophile