1technophile/OpenMQTTGateway #1691: INA226 incorrect JSON output

murfy76, 2023-06-20:
Hello,

JSON output from INA226 sensor is incorrect. Output contain space before values, e.g.:  {"volt":  3.3,...} and it should be {"volt": 3.3,...}

My workaround is to modify "ZsensorINA226.ino" to:
...   
//char volt_c[7];
//char current_c[7];
//char power_c[7];
//dtostrf(volt, 6, 3, volt_c);
//dtostrf(current, 6, 3, current_c);
//dtostrf(power, 6, 3, power_c);
INA226data["volt"] = (float)volt;
INA226data["current"] = (float)current;
INA226data["power"] = (float)power;
...

Steps to reproduce the behavior:
1. Publish INA226 to MQTT with JSON
2. Use Telegraf "mqtt_consumer" with json output to influxdb to read this topic 
3. InfluxDB/Grafana will be not able to read/graph this data

**Expected behavior**
Valid JSON output format like other sensors.

Thank you. 

Peter

1technophile, 2023-06-22:
Hello,

Feel free to submit a PR

NorbertHD, 2023-07-01:
Or set the 2. parameter of dtostrf() to 0, then no leading or trailing spaces are generated.

e.g. dtostrf(volt, 0, 3, volt_c);

PR #1714 (2023-07-31): INA226 JSON do not output leading spaces
## Description:
Closes #1691

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
closed 2023-07-31 by 1technophile