1technophile/OpenMQTTGateway #1692: Exception Loop with 1.6.0

sebbo2002, 2023-06-20:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I updated my ESP to 1.6.0, but now I get these Exceptions. Already tried the dev build (OpenMQTTGateway DEVELOPMENT SHA:561de9), but same issue.

```
…
20:39:00.528 -> N: Device detected: CD:1A:C0:33:AD:39
20:39:01.063 -> N: Device detected: 7C:2F:80:EA:FB:FD
20:39:01.424 -> N: Device detected: 7C:E0:4B:C9:D3:ED
20:39:02.264 -> N: Device detected: F8:34:B2:78:26:41
20:39:02.308 -> N: Send on /BTtoMQTT/7C2F80EAFBFD msg {"id":"7C:2F:80:EA:FB:FD","name":"Gigaset G-tag","rssi":-84}
20:39:02.334 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570F450105FF00","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
20:39:02.369 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"movingtimer":60000,"btqblck":9,"btqsum":5,"btqsnd":2,"btqavg":2.5,"bletaskstack":3028,"blecoretaskstack":3140,"blestarts":1}
20:39:02.407 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
20:39:02.407 -> 
20:39:02.407 -> Core  1 register dump:
20:39:02.407 -> PC      : 0x40090fdf  PS      : 0x00060e30  A0      : 0x801b2bdd  A1      : 0x3ffd2440  
20:39:02.441 -> A2      : 0x00000000  A3      : 0x3ffd250c  A4      : 0x0000000e  A5      : 0x00000000  
20:39:02.441 -> A6      : 0x46303735  A7      : 0x31303534  A8      : 0x00000000  A9      : 0x3ffd2400  
20:39:02.441 -> A10     : 0x3ffd2504  A11     : 0x3fff910b  A12     : 0x0000000e  A13     : 0x0000ffff  
20:39:02.441 -> A14     : 0x00000000  A15     : 0x0000ffff  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
20:39:02.441 -> EXCVADDR: 0x00000000  LBEG    : 0x40090fc0  LEND    : 0x40090fd6  LCOUNT  : 0xffffffff  
20:39:02.479 -> 
20:39:02.479 -> 
20:39:02.479 -> Backtrace: 0x40090fdc:0x3ffd2440 0x401b2bda:0x3ffd2450 0x401b2c79:0x3ffd2470 0x401b2da3:0x3ffd2490 0x400e4de5:0x3ffd24b0 0x400e4feb:0x3ffd25d0 0x400e870a:0x3ffd2610 0x400e87b3:0x3ffd29e0 0x401d2153:0x3ffd2a00 0x400ec455:0x3ffd2a20 0x400eaf1a:0x3ffd2a60 0x4011e471:0x3ffd2aa0
20:39:02.479 -> 
20:39:02.479 -> 
20:39:02.479 -> 
20:39:02.479 -> 
20:39:02.479 -> ELF file SHA256: cc40cd1dde3e0bed
20:39:02.479 -> 
20:39:02.732 -> Rebooting...
20:39:02.770 -> ets Jul 29 2019 12:21:46
20:39:02.770 -> 
20:39:02.770 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:39:02.770 -> configsip: 0, SPIWP:0xee
20:39:02.770 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:39:02.770 -> mode:DIO, clock div:2
20:39:02.770 -> load:0x3fff0030,len:1184
20:39:02.770 -> load:0x40078000,len:13160
20:39:02.770 -> load:0x40080400,len:3036
20:39:02.770 -> entry 0x400805e4
20:39:03.472 -> N: 
20:39:03.472 -> ************* WELCOME TO OpenMQTTGateway **************
…
```

**Expected behavior**
Well. No Exception.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.6.0 and SHA:561de9)
* Uploaded via Web Uploader

sebbo2002, 2023-06-21:
Downgraded to v1.5.1 for now, which fixed it for me…

1technophile, 2023-06-22:
is it not a duplicate of #1694 as both are triggered following a write command ?

sebbo2002, 2023-06-22:
@1technophile Oh, I was apparently a bit tired. Yes of course, I had overlooked the write command. That seems to have come from home automation then, sorry. I leave #1694 open, thank you 🙈

closed 2023-06-22 by sebbo2002