1technophile/OpenMQTTGateway #1694: Read BLE characteristics is not working anymore

sebbo2002, 2023-06-21:
- [x] Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**

Sending BLE characteristics over MQTT isn't working anymore for me. Unable to test it with 1.6.0 due to #1692.

```
…
21:46:32.361 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570F450105FF08","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
21:46:32.453 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"btqblck":9,"btqsum":14,"btqsnd":8,"btqavg":1.75}
21:46:32.453 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
21:46:32.453 -> 
21:46:32.453 -> Core  1 register dump:
21:46:32.453 -> PC      : 0x40090fdf  PS      : 0x00060e30  A0      : 0x801a552d  A1      : 0x3ffd0170  
21:46:32.453 -> A2      : 0x00000000  A3      : 0x3ffd023c  A4      : 0x0000000e  A5      : 0x00000000  
21:46:32.453 -> A6      : 0x46303735  A7      : 0x31303534  A8      : 0x00000000  A9      : 0x3ffd0130  
21:46:32.453 -> A10     : 0x3ffd0234  A11     : 0x3fffefe3  A12     : 0x0000000e  A13     : 0x0000ffff  
21:46:32.453 -> A14     : 0x00000000  A15     : 0x0000ffff  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
21:46:32.453 -> EXCVADDR: 0x00000000  LBEG    : 0x40090fc0  LEND    : 0x40090fd6  LCOUNT  : 0xffffffff  
21:46:32.476 -> 
21:46:32.476 -> 
21:46:32.476 -> Backtrace: 0x40090fdc:0x3ffd0170 0x401a552a:0x3ffd0180 0x401a55c9:0x3ffd01a0 0x401a56f3:0x3ffd01c0 0x400e05a5:0x3ffd01e0 0x400e07ab:0x3ffd0300 0x400e3046:0x3ffd0340 0x400e30ef:0x3ffd0710 0x401c7dbf:0x3ffd0730 0x400e4539:0x3ffd0750 0x400de6e1:0x3ffd0790 0x4010eac5:0x3ffd07b0
21:46:32.499 -> 
21:46:32.499 -> 
21:46:32.499 -> 
21:46:32.499 -> 
21:46:32.499 -> ELF file SHA256: 2ac7f811d8bb3423
21:46:32.499 -> 
21:46:32.831 -> Rebooting...
21:46:32.869 -> ets Jul 29 2019 12:21:46
21:46:32.869 -> 
21:46:32.869 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:46:32.869 -> configsip: 0, SPIWP:0xee
21:46:32.869 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:46:32.869 -> mode:DIO, clock div:2
21:46:32.869 -> load:0x3fff0030,len:1184
21:46:32.869 -> load:0x40078000,len:13160
21:46:32.869 -> load:0x40080400,len:3036
21:46:32.869 -> entry 0x400805e4
21:46:33.532 -> N: 
21:46:33.532 -> ************* WELCOME TO OpenMQTTGateway **************
21:46:33.638 -> N: OpenMQTTGateway Version: v1.5.0
21:46:33.785 -> {
21:46:33.785 ->   "mqtt_server": "***********",
21:46:33.785 ->   "mqtt_port": "1883",
21:46:33.785 ->   "mqtt_user": "***********",
21:46:33.823 ->   "mqtt_pass": "***********",
21:46:33.823 ->   "mqtt_topic": "home/",
21:46:33.823 ->   "gateway_name": "***********",
21:46:33.823 ->   "mqtt_broker_secure": false,
21:46:33.823 ->   "mqtt_broker_cert": "",
21:46:33.823 ->   "mqtt_ss_index": 0,
21:46:33.823 ->   "ota_server_cert": "\"\n-----BEGIN CERTIFICATE-----\nMIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh\nMQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\nd3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD\nQTAeFw0wNjExMTAwMDAwMDBaFw0zMTExMTAwMDAwMDBaMGExCzAJBgNVBAYTAlVT\nMRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j\nb20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IENBMIIBIjANBgkqhkiG\n9w0BAQEFAAOCAQ8AMIIBCgKCAQEA4jvhEXLeqKTTo1eqUKKPC3eQyaKl7hLOllsB\nCSDMAZOnTjC3U/dDxGkAV53ijSLdhwZAAIEJzs4bg7/fzTtxRuLWZscFs3YnFo97\nnh6Vfe63SKMI2tavegw5BmV/Sl0fvBf4q77uKNd0f3p4mVmFaG5cIzJLv07A6Fpt\n43C/dxC//AH2hdmoRBBYMql1GNXRor5H4idq9Joz+EkIYIvUX7Q6hL+hqkpMfT7P\nT19sdl6gSzeRntwi5m3OFBqOasv+zbMUZBfHWymeMr/y7vrTC0LUq7dBMtoM1O/4\ngdW7jVg/tRvoSSiicNoxBN33shbyTApOB6jtSj1etX+jkMOvJwIDAQABo2MwYTAO\nBgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUwAwEB/zAdBgNVHQ4EFgQUA95QNVbR\nTLtm8KPiGxvDl7I90VUwHwYDVR0jBBgwFoAUA95QNVbRTLtm8KPiGxvDl7I90VUw\nDQYJKoZIhvcNAQEFBQADggEBAMucN6pIExIK+t1EnE9SsPTfrgT1eXkIoyQY/Esr\nhMAtudXH/vTBH1jLuG2cenTnmCmrEbXjcKChzUyImZOMkXDiqw8cvpOp/2PV5Adg\n06O/nVsJ8dWO41P0jmP6P6fbtGbfYmbW0W5BjfIttep3Sp+dWOIrWcBAI+0tKIJF\nPnlUkiaY4IBIqDfv8NZ5YBberOgOzW6sRBc4L0na4UU+Krk2U886UAb3LujEV0ls\nYSEY1QSteDwsOoBrp+uvFRTp2InBuThs4pFsiv9kuXclVzDAGySj4dzp30d8tbQk\nCAUw7C29C79Fv1C5qfPrmAESrciIxpg0X40KPMbp1ZWVbd4=\n-----END CERTIFICATE-----\n\"",
21:46:33.937 ->   "ota_pass": "***********"
21:46:33.937 -> }*wm:[2] Added Parameter: server
21:46:33.937 -> *wm:[2] Added Parameter: port
21:46:33.975 -> *wm:[2] Added Parameter: user
21:46:33.975 -> *wm:[2] Added Parameter: pass
21:46:33.975 -> *wm:[2] Added Parameter: secure
21:46:33.975 -> *wm:[2] Added Parameter: cert
21:46:33.975 -> *wm:[2] Added Parameter: name
21:46:33.975 -> *wm:[2] Added Parameter: topic
21:46:33.975 -> *wm:[2] Added Parameter: ota
21:46:33.975 -> N: Attempting Wifi connection with saved AP: 0
21:46:34.985 -> N: Attempting Wifi connection with saved AP: 1
21:46:34.985 -> E (3415) wifi:sta is connecting, return error
21:46:34.985 -> [  1528][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
21:46:35.999 -> N: Attempting Wifi connection with saved AP: 2
21:46:37.023 -> N: Attempting Wifi connection with saved AP: 3
[... 55 lines cut ...]
21:46:41.360 -> N: ************** Setup OpenMQTTGateway end **************
21:46:41.405 -> W: MQTT connection...
21:46:41.463 -> N: Connected to broker
21:46:41.570 -> N: Send on /SYStoMQTT msg {"uptime":8,"version":"v1.5.0","discovery":true,"env":"esp32dev-ble","freemem":94484,"mqttport":"1883","mqttsecure":false,"tempc":60,"freestack":4328,"rssi":-59,"SSID":"Frittenbude","BSSID":"FC:EC:DA:1A:85:22","ip":"10.13.35.197","mac":"94:B5:55:FA:06:2C","lowpowermode":-1,"interval":55555,"intervalcnct":3600000,"scnct":0,"modules":["BT"]}
21:46:41.603 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"btqblck":0,"btqsum":0,"btqsnd":0,"btqavg":0}
21:46:41.641 -> N: Update checkN: Scan begin
21:46:42.406 -> N: Device detected: 00:7C:2D:2C:DC:6E
21:46:42.444 -> N: Device detected: EC:B9:3A:06:BF:96
21:46:42.516 -> N: Device detected: DD:A2:6E:62:B8:2F
21:46:42.516 -> N: Device detected: 66:EF:76:24:C8:D2
…
```

**To Reproduce**
Steps to reproduce the behavior:
1. Publish characteristic via MQTT, see example above

**Expected behavior**
The characteristic should be sent without an exception occurring and OpenMQTTGateway restarting.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used: 0.5.0 / 0.5.1 (both tested)

1technophile, 2023-06-21:
Let me try to reproduce it

1technophile, 2023-06-22:
Did you try with HEX type:
```
{
  "ble_write_address": "FA:83:ED:1F:81:8D",
  "ble_write_service": "cba20d00-224d-11e6-9fb8-0002a5d5c51b",
  "ble_write_char": "cba20002-224d-11e6-9fb8-0002a5d5c51b",
  "ble_write_value": "0x570F450105FF08",
  "value_type": "HEX",
  "mac_type": 1,
  "ttl": 5,
  "immediate": true
}
```

Investigating the issue with STRING

sebbo2002, 2023-06-23:
Tried it with `0x`. It doesn't hang up, but it's not working for me. It says `"success": true`, but the Switchbot Curtains don't move. 🙈

```
10:44:30.577 -> N: Device detected: 64:5A:47:C2:D9:C3
10:44:30.680 -> N: Device detected: 72:44:FA:1B:67:90
10:44:30.903 -> N: Device detected: F9:95:25:5E:75:96
10:44:31.014 -> N: Device detected: C8:54:86:B8:C0:D3
10:44:31.673 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"0x570F450105FF0F","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
10:44:31.673 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"btqblck":9,"btqsum":528,"btqsnd":445,"btqavg":1.186517}
10:44:31.777 -> N: Found 29 devices, scan number 48 end
10:44:31.777 -> N: BLE Connect begin
10:44:32.539 -> N: Send on /BTtoMQTT/FA83ED1F818D msg {"id":"FA:83:ED:1F:81:8D","service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","characteristic":"cba20002-224d-11e6-9fb8-0002a5d5c51b","write":"0x570F450105FF0F","success":true}N: BLE Connect end
10:44:32.626 -> 
10:45:11.916 -> N: Send on /SYStoMQTT msg {"uptime":3130,"version":"v1.5.0","discovery":false,"env":"esp32dev-ble","freemem":98108,"mqttport":"1883","mqttsecure":false,"tempc":59.44444,"freestack":1768,"rssi":-78,"SSID":"Frittenbude","BSSID":"FC:EC:DA:38:6F:BD","ip":"10.13.35.197","mac":"94:B5:55:FA:06:2C","lowpowermode":-1,"interval":55555,"intervalcnct":3600000,"scnct":48,"modules":["BT"]}
10:45:11.995 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"btqblck":9,"btqsum":529,"btqsnd":446,"btqavg":1.186099}
```

DigiH, 2023-06-23:
Hi @sebbo2002 

Could you try as with the original command, but with lower casing the hex value, e.g.

`{"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570f450105ff08","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}`

This has been, and is still working fine for me here. Admittedly, I use it with sending commands to different BLE devices and I also just use the 16-bit service and characteristic, e.g.

`{"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"0d00","ble_write_char":"0002","ble_write_value":"570f450105ff08","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}`

I'm also not sure if anything would have changed that your original command should not work with 1.6.0, but worth trying with the above.

sebbo2002, 2023-06-23:
## Lowercased Hex

<details>
<summary><strong>v1.5.0</strong></summary>

```
13:40:57.011 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570f450105ff08","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
13:40:57.092 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"btqblck":12,"btqsum":4,"btqsnd":1,"btqavg":4}
13:40:57.092 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
13:40:57.092 -> 
13:40:57.092 -> Core  1 register dump:
13:40:57.092 -> PC      : 0x40090fdf  PS      : 0x00060e30  A0      : 0x801a552d  A1      : 0x3ffd0200  
13:40:57.092 -> A2      : 0x00000000  A3      : 0x3ffd02cc  A4      : 0x0000000e  A5      : 0x00000000  
13:40:57.092 -> A6      : 0x66303735  A7      : 0x31303534  A8      : 0x00000000  A9      : 0x3ffd01c0  
13:40:57.120 -> A10     : 0x3ffd02c4  A11     : 0x3ffea4f3  A12     : 0x0000000e  A13     : 0x0000ffff  
13:40:57.120 -> A14     : 0x00000000  A15     : 0x0000ffff  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
13:40:57.120 -> EXCVADDR: 0x00000000  LBEG    : 0x40090fc0  LEND    : 0x40090fd6  LCOUNT  : 0xffffffff  
13:40:57.156 -> 
13:40:57.156 -> 
13:40:57.156 -> Backtrace: 0x40090fdc:0x3ffd0200 0x401a552a:0x3ffd0210 0x401a55c9:0x3ffd0230 0x401a56f3:0x3ffd0250 0x400e05a5:0x3ffd0270 0x400e07ab:0x3ffd0390 0x400e3046:0x3ffd03d0 0x400e30ef:0x3ffd07a0 0x401c7dbf:0x3ffd07c0 0x400e4539:0x3ffd07e0 0x400de6e1:0x3ffd0820 0x4010eac5:0x3ffd0840
13:40:57.156 -> 
13:40:57.156 -> 
13:40:57.156 -> 
13:40:57.156 -> 
13:40:57.156 -> ELF file SHA256: 2ac7f811d8bb3423
13:40:57.156 -> 
13:40:57.569 -> Rebooting...
13:40:57.569 -> ets Jul 29 2019 12:21:46
13:40:57.569 -> 
13:40:57.569 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:40:57.593 -> configsip: 0, SPIWP:0xee
13:40:57.593 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:40:57.593 -> mode:DIO, clock div:2
13:40:57.593 -> load:0x3fff0030,len:1184
13:40:57.593 -> load:0x40078000,len:13160
13:40:57.593 -> load:0x40080400,len:3036
13:40:57.593 -> entry 0x400805e4
13:40:58.235 -> N: 
13:40:58.235 -> ************* WELCOME TO OpenMQTTGateway **************
13:40:58.352 -> N: OpenMQTTGateway Version: v1.5.0
```
</details>

<details>
<summary><strong>v1.6.0</strong></summary>

```
13:56:10.187 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570f450105ff08","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
13:56:10.187 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"movingtimer":60000,"btqblck":10,"btqsum":8,"btqsnd":2,"btqavg":4,"bletaskstack":3028,"blecoretaskstack":3140,"blestarts":1}
13:56:10.274 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
13:56:10.274 -> 
13:56:10.274 -> Core  1 register dump:
13:56:10.274 -> PC      : 0x40090fdf  PS      : 0x00060e30  A0      : 0x801b2bdd  A1      : 0x3ffd2520  
13:56:10.274 -> A2      : 0x00000000  A3      : 0x3ffd25ec  A4      : 0x0000000e  A5      : 0x00000000  
13:56:10.274 -> A6      : 0x66303735  A7      : 0x31303534  A8      : 0x00000000  A9      : 0x3ffd24e0  
13:56:10.274 -> A10     : 0x3ffd25e4  A11     : 0x3ffecce3  A12     : 0x0000000e  A13     : 0x0000ff00  
13:56:10.274 -> A14     : 0x00ff0000  A15     : 0x40404040  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
13:56:10.274 -> EXCVADDR: 0x00000000  LBEG    : 0x40090fc0  LEND    : 0x40090fd6  LCOUNT  : 0xffffffff  
13:56:10.274 -> 
13:56:10.274 -> 
13:56:10.274 -> Backtrace: 0x40090fdc:0x3ffd2520 0x401b2bda:0x3ffd2530 0x401b2c79:0x3ffd2550 0x401b2da3:0x3ffd2570 0x400e4de5:0x3ffd2590 0x400e4feb:0x3ffd26b0 0x400e870a:0x3ffd26f0 0x400e87b3:0x3ffd2ac0 0x401d2153:0x3ffd2ae0 0x400ec455:0x3ffd2b00 0x400eaf1a:0x3ffd2b40 0x4011e471:0x3ffd2b80
13:56:10.296 -> 
13:56:10.296 -> 
13:56:10.296 -> 
13:56:10.296 -> 
13:56:10.296 -> ELF file SHA256: 758b2010a1c42e2d
13:56:10.296 -> 
13:56:10.654 -> Rebooting...
13:56:10.654 -> ets Jul 29 2019 12:21:46
13:56:10.654 -> 
[... 72 lines cut ...]
13:57:19.443 -> 
13:57:19.443 -> 
13:57:19.443 -> Backtrace: 0x40090fdc:0x3ffd2520 0x401b2bda:0x3ffd2530 0x401b2c79:0x3ffd2550 0x401b2da3:0x3ffd2570 0x400e4de5:0x3ffd2590 0x400e4feb:0x3ffd26b0 0x400e870a:0x3ffd26f0 0x400e87b3:0x3ffd2ac0 0x401d2153:0x3ffd2ae0 0x400ec455:0x3ffd2b00 0x400eaf1a:0x3ffd2b40 0x4011e471:0x3ffd2b80
13:57:19.480 -> 
13:57:19.480 -> 
13:57:19.480 -> 
13:57:19.480 -> 
[... cut ...]
13:57:19.827 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:57:19.827 -> configsip: 0, SPIWP:0xee
13:57:19.827 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:57:19.827 -> mode:DIO, clock div:2
13:57:19.827 -> load:0x3fff0030,len:1184
13:57:19.827 -> load:0x40078000,len:13160
13:57:19.827 -> load:0x40080400,len:3036
13:57:19.827 -> entry 0x400805e4
13:57:20.523 -> N: 
13:57:20.523 -> ************* WELCOME TO OpenMQTTGateway **************
13:57:20.630 -> N: OpenMQTTGateway Version: v1.6.0
```
</details>

sebbo2002, 2023-06-23:
Both lowercased and uppercased hex works with 1.4.0, just tested it:

```
14:06:44.590 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570F450105FF00","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
14:06:44.652 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"onlysensors":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"btqblck":15,"btqsum":149,"btqsnd":109,"btqavg":1.366972}
14:06:44.652 -> N: BLE Connect begin
14:06:45.633 -> N: BLE Connect end
14:06:45.633 -> N: Send on /BTtoMQTT/FA83ED1F818D msg {"id":"FA:83:ED:1F:81:8D","service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","characteristic":"cba20002-224d-11e6-9fb8-0002a5d5c51b","write":"570F450105FF00","success":true}
```

1technophile, 2023-06-23:
It seems related with the platform jump from espressif32@3.5.0 to espressif32@6.1.0, as I don't have the reset anymore if I change back the platform with the current codebase

1technophile, 2023-06-23:
Same issue with espressif32@6.3.2

1technophile, 2023-06-23:
Could you try this branch please:
https://github.com/1technophile/OpenMQTTGateway/tree/ble-fix

I have implemented a fix, note that I also updated the topic to use to be more aligned with the way we use it for other command sending.
Topic: `home/840D8E365294/commands/MQTTtoBT`
Command: 
```json
{
  "ble_write_address": "FA:83:ED:1F:81:8D",
  "ble_write_service": "cba20d00-224d-11e6-9fb8-0002a5d5c51b",
  "ble_write_char": "cba20002-224d-11e6-9fb8-0002a5d5c51b",
  "ble_write_value": "570F450105FF08",
  "value_type": "STRING",
  "mac_type": 1,
  "ttl": 5,
  "immediate": true
}
```

DigiH, 2023-06-23:
That's why it was working for me all along, up to 1.6.0, because I am using a custom build with the older platform espressif32@3.5.0, for other reasons though ;)

sebbo2002, 2023-06-24:
Thanks for the test branch, I'll have a look as soon as I get around to it. This will probably take a while, though, as I've only installed OMG via the web installer so far and now I have to get to grips with compiling it myself. I probably won't have time to do that this weekend. I'll be in touch.

1technophile, 2023-06-24:
I can have the web portal generated from this PR if you can test it today or tomorrow.
Note that it will be overwritten each night by the development branch

sebbo2002, 2023-06-25:
I would have a short time tonight, so I could test it with the web uploaded. That would be about 10 hours. Or then tomorrow, then it's no problem. So via Web Uploader would be very convenient for me, thanks 🥰

1technophile, 2023-06-25:
I have launched the build:
https://github.com/1technophile/OpenMQTTGateway/actions/runs/5369745083

When ready, you will find it here:
https://docs.openmqttgateway.com/dev

sebbo2002, 2023-06-25:
I just tested it with the provided version. Seems like there is a newline missing between `N: Send on …` and `N: BLE Connect end`, but it's working fine 🥳

```
19:44:09.091 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570F450105FF0B","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
19:44:09.177 -> N: BLE Connect begin
19:44:10.162 -> N: Send on /BTtoMQTT/FA83ED1F818D msg {"id":"FA:83:ED:1F:81:8D","service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","characteristic":"cba20002-224d-11e6-9fb8-0002a5d5c51b","write":"570F450105FF0B","success":true}N: BLE Connect end
19:44:10.233 -> 
19:44:11.397 -> N: Send on /SYStoMQTT msg {"uptime":1654,"version":"3f2705","discovery":true,"ohdiscovery":false,"env":"esp32dev-ble","freemem":89044,"mqttport":"1883","mqttsecure":false,"minfreemem":20328,"tempc":55.55556,"freestack":3140,"rssi":-76,"SSID":"Frittenbude","BSSID":"FC:EC:DA:38:6F:BD","ip":"10.13.35.197","mac":"94:B5:55:FA:06:2C","lowpowermode":-1,"interval":55555,"intervalcnct":3600000,"scnct":24,"modules":["WebUI","BT"]}
19:44:11.430 -> N: Send on /BTtoMQTT msg {"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"onlysensors":false,"randommacs":false,"hasspresence":false,"presenceTopic":"presence/","presenceUseBeaconUuid":false,"minrssi":-100,"extDecoderEnable":false,"extDecoderTopic":"undecoded","filterConnectable":false,"pubadvdata":false,"pubBeaconUuidForTopic":false,"ignoreWBlist":false,"presenceawaytimer":120000,"movingtimer":60000,"btqblck":66,"btqsum":279,"btqsnd":198,"btqavg":1.409091,"bletaskstack":3024,"blecoretaskstack":3056,"blestarts":1}
19:44:11.469 -> N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":1}
19:44:16.975 -> N: [ MQTT->OMG ]: {"ble_write_address":"FA:83:ED:1F:81:8D","ble_write_service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","ble_write_char":"cba20002-224d-11e6-9fb8-0002a5d5c51b","ble_write_value":"570F450105FF00","value_type":"HEX","mac_type":1,"ttl":5,"immediate":true}
19:44:16.975 -> N: BLE Connect begin
19:44:17.571 -> N: Send on /BTtoMQTT/FA83ED1F818D msg {"id":"FA:83:ED:1F:81:8D","service":"cba20d00-224d-11e6-9fb8-0002a5d5c51b","characteristic":"cba20002-224d-11e6-9fb8-0002a5d5c51b","write":"570F450105FF00","success":true}N: BLE Connect end
19:44:17.680 -> 
19:44:26.011 -> N: Scan begin
19:44:26.036 -> N: Device detected: 58:DB:9A:6A:43:EA
19:44:26.095 -> N: Device detected: C4:5E:E4:F8:57:5C
19:44:26.171 -> N: Device detected: FD:24:D8:C4:2E:56
19:44:26.209 -> N: Device detected: 11:0A:E3:60:1C:8E
```

1technophile, 2023-06-26:
THanks for the feedback. The new line missing is due to several tasks printing in the serial at the same time

PR #1698 (2023-06-26): [BT] Fix BLE MQTT action command trigger a reset
## Description:
Problem introduced with the bump of ESP32 platform from 3.5.0 to 6.1.0
fix #1694 

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
closed 2023-06-26 by 1technophile