1technophile/OpenMQTTGateway #1701: LilyGo Lora 32 V2.1 with rtl_433 at 915Mhz

allistermaguire, 2023-06-26:
Hi, I am trying to use a LilyGo v2.1 with the rtl_433 to decode signals from my Cavius Smoke Detectors.

I know it isn't a documented configuration, and I have cobbled my configuration together from people's posts in the repo and the rtl_433_ESP repo and am unsure where my issue might be.

According to the Cavius spec sheet my Smoke detectors are using 926.365 MHz, they are using `FSK_PULSE_PCM` modulation which is supported in the latest version of rtl_433_ESP.

This is what I can see via the terminal monitor when I test the Smoke detector, but am not sure what this means? I would assume I should also get some nice decoded text?

```
pulse_slicer_pcm: Cavius smoke, heat and water detector codes [{2139}ffffffffff43f1ff7d7ffeaff3a6fefe77fd7ff3ffcedbeffd7fff7ccfcf66ffdffefffffdfffbdffffffff721f87fdf7fdffffdd3bf9fbbfefffbffeffbcfffffde7dbfbfb3ff7ffffefffff7bffff7fffffb58fc3fefbfff77ff7bbddfb3fdfff775eff3dfffffbee9df9fcefffdf6fffff7fbffffffffffffd8fc3fef7abffffee9dfcfddff7ffdfff7fdf3fd7fff9f3ab3f77f77fffffffffffffffffffffe8fc7fdd5fffffbcfb9fdf9def36ff7ffdfafabfffffbeefefeeffffddffafffffffdfdbfbffffb0fc3ecf9effffff7b3fbf3abdfff7ffd7e7cf5ffffef33f3bbbfbafffffffffffdfffefbf5f986c3fe79ffffffefbbfbfbbfefffbffcff7dffffffcf9d7bd9dfa144a56]
```

This is what my configuration is
```
[env:ttgo-v21-rtl_433]
platform = ${com.esp32_platform}
board = ttgo-lora32-v21
; ~/.platformio/packages/framework-arduinoespressif32/variants/.../pins_arduino.h
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.wifimanager32}
  ${libraries.ssd1306}
  ${libraries.rtl_433_ESP}
build_flags =
  ${com-esp32.build_flags}
  '-DvalueAsATopic=true'                ; MQTT topic includes model and device
  '-DGateway_Name="OMG_lilygo_lora32_rtl_433_ESP"'
; *** OpenMQTTGateway Modules ***
  '-DZgatewayRTL_433="rtl_433"'
  '-DZradioSX127x="SX127x"'
  '-DCC1101_FREQUENCY=926.365'
  '-DRF_SX1276="SX1276"'
  '-DZdisplaySSD1306="LilyGo_SSD1306"'
  '-DDISPLAY_IDLE_LOGO=true'
  '-DRF_MODULE_INIT_STATUS=true'
; *** Custom Settings ***
  '-DESPWifiManualSetup=true'           ; WiFi
  '-Dwifi_ssid="*********"'
  '-Dwifi_password="*********"'
  '-DMQTT_SECURE_DEFAULT=true'          ; MQTT
  '-DMQTT_CERT_VALIDATE_DEFAULT=false'
  '-DMQTT_USER="*********"'
  '-DMQTT_PASS="*********"'
  '-DMQTT_SERVER="*********"'
  '-DMQTT_PORT="8883"'
  '-Dota_password="*********"'          ; WebUI Password
  '-UZmqttDiscovery'                    ; Disable HA discovery
  '-DOOK_MODULATION=false'              ; FSK mode
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'         ; Logging
  '-DRTL_DEBUG=1'                       ; Logging
custom_description = For ESP32, Gateway using RTL_433_ESP and RadioLib
custom_hardware = ESP32 LILYGO LoRa32 V2.1
```

NorthernMan54, 2023-06-26:
You may want to try enabling the compiler directive `DEMOD_DEBUG`, and this will log more details of the received signal.

I think the line you saw in the log is the processing logic attempting to parse the signal with the Cavius signal decoder, and since you did not see any undecoded message, decoding failed.

allistermaguire, 2023-06-27:
Thanks @NorthernMan54 for the quick response. I have enabled, but I can no longer get the previous message to show up now. So not sure what is going on now.

I am seeing additional messages, but nothing helpful. None align with the Smoke detector test either.
```
rtl_433_ESP(7): Average RSSI Signal -72 dbm, adjusted RSSI Threshold -63, samples 50000
rtl_433_ESP(7): Average RSSI Signal -95 dbm, adjusted RSSI Threshold -86, samples 50000
rtl_433_ESP(6): Ignored Signal length: 9, Time since last bit length: 6353, Gap length: 84135133, Signal RSSI: -85, Current RSSI: -93, pulses: 27, noise count: 6
rtl_433_ESP(6): Ignored Signal length: 9, Time since last bit length: 6424, Gap length: 2006422, Signal RSSI: -85, Current RSSI: -99, pulses: 28, noise count: 5
rtl_433_ESP(6): Ignored Signal length: 9, Time since last bit length: 6438, Gap length: 5243407, Signal RSSI: -84, Current RSSI: -93, pulses: 25, noise count: 7
rtl_433_ESP(6): Ignored Signal length: 9, Time since last bit length: 6329, Gap length: 4824364, Signal RSSI: -85, Current RSSI: -93, pulses: 25, noise count: 6
```

I haven't tried to understand how it works just yet, was hoping I wouldn't have to :) But it is possible that the signal isn't from the Cavius smoke alarm, and hence why it cannot decode it?

I see other signals attempted to be parsed with `fineoffset`, `lacrosse`, `honeywell` etc without successful decoding as well.

I have one with the 433Mhz SX1278 in it, and I am getting decoded results, so I know I can at least 

Trying to narrow down where the issue is, is difficult. Not sure if I have configured it correctly for the hardware, don't know if it is in the actual correct frequency, apparently the attached antenna isn't very good, could be so many things.

This is how to use in rtl_433, obviously not the NZ version.
```
rtl_433 -f 868.68M -s 250k -R 179
```

NorthernMan54, 2023-06-27:
@allistermaguire If your device is `433Mhz SX1278` it will not work at 926.365 MHz, you need one with the SX1276 chip

allistermaguire, 2023-06-27:
@NorthernMan54 Sorry that was poorly written, I have two devices one with SX1278, and one with SX1276.

I and using the SX1278 for my Ecowitt Weather Station and Sensors.

allistermaguire, 2023-07-01:
I have purchased a RTL-SDR v3 to test and get the Cavius sensors working before adding the extra variables of the ESP.

NorthernMan54, 2023-07-04:
Pls see https://github.com/NorthernMan54/rtl_433_ESP/issues/87 your issue may be the same

allistermaguire, 2023-07-06:
Thanks, @NorthernMan54, I tried it and it made no difference. I don't actually see any signal timings that correspond with triggering the smoke alarm or pairing. All I see is lots of ignored signals and then a burst of TPMS.

allistermaguire, 2023-09-01:
I will close this as I can't get it to work with rtl_433 on a RTL-SDR v3. If necessary I will open a new issue in the future.

closed 2023-09-01 by allistermaguire