1technophile/OpenMQTTGateway #1719: How to add rtl_433 module to the gateway?

michapr, 2023-07-19:
**Describe the bug**
rtl_433 module is not in list in user_config

**To Reproduce**
Steps to reproduce the behavior:
open user_config and look for rtl_433

**Expected behavior**
add method to use the module

**Environment (please complete the following information):**
* OpenMQTTGateway version used (1.6.0)

**Additional context**

If I add a line
#define ZgatewayRTL_433   "RTL433"  //ESP8266, Arduino, ESP32

will get compilation errors because of other missing libs like rtl_433_ESP.h, RadioLib.h ...
So I think here must be an other way, because ready images for rtl_433 devices are here available ;)

Maybe somebody can show the way, how the [esp32dev-rtl_433](https://docs.openmqttgateway.com/prerequisites/boards/esp32dev-rtl_433) was created - as sample?

THANKS!

habub, 2023-07-19:
You select the RTL_433 environment in the platformio.ini file from the long list of default_envs. Uncomment the configuration you want, for example: default_envs = esp32dev-rtl_433
The user_config file sets the hw configuration based on the selected environment from ini file. Look for cc1101 in the config.

michapr, 2023-07-20:
Yes, this is working for existing devices from list.
But if I want to use as sample a esp32S3 and add rtl_433 - so I can do it (like with other modules) with user_config.
This is not working for rtl_433, as far as I have seen...

NorthernMan54, 2023-07-20:
This should be pretty straight forward

1 - Create your own `test_env.ini`
2 - From environments.ini, copy and merge a one of the rtl_433 environments with a S3 environment into `test_env.ini`
3 - Restart VSC
4 - Select your new environment, then build

michapr, 2023-07-20:
@NorthernMan54 - I was thinking initial about Arduino IDE...
But will test your suggestion in PIO - I did not use it often before,... THANKS!

NorthernMan54, 2023-07-20:
I find that the Arduino IDE is like working with one arm tied behind your back ;-(

tablatronix, 2023-08-15:
For some reason mine just crashes over and over, but compiles fine, probably something with my spi..
Still working on it..
but heres an example of mine

```
[env:esp32S3BASIC-rtl_433_LOCAL]
platform = ${com.esp32_platform}
; board = esp32dev
board = esp32-s3-devkitc-1
board_build.partitions = min_spiffs.csv
lib_deps = 
	${com-esp32.lib_deps}
	${libraries.rtl_433_ESP}
build_flags = 
	${com-esp32.build_flags}
	'-DvalueAsATopic=true'
	'-DLOG_LEVEL=LOG_LEVEL_TRACE'
	'-DGateway_Name="OMG_rtl_433_ESP"'
	'-DZgatewayRTL_433="rtl_433"'
	'-DZradioCC1101="CC1101"'
	'-DDEMOD_DEBUG=false'
	'-DPUBLISH_UNPARSED=true'
	'-DSIGNAL_RSSI=true'
	'-DNO_DEAF_WORKAROUND=false'
	'-DRF_CC1101="CC1101"'
    '-DCC1101_FREQUENCY=433.950'
	'-DRF_MODULE_GDO0=7'
	'-DRF_MODULE_GDO2=6'
	'-DRF_MODULE_SCK=13'
	'-DRF_MODULE_MISO=12'
	'-DRF_MODULE_MOSI=11'
	'-DRF_MODULE_CS=10'
	'-DTRIGGER_GPIO=0'
	'-DMQTT_SERVER="192.168.100.17"'
	'-Dwifi_ssid="wifi"'
	'-Dwifi_password="password"'
	'-DNO_INT_TEMP_READING'
custom_description = Gateway using RTL_433_ESP library, need CC1101
```

NorthernMan54, 2023-08-16:
If you add the directive `RF_MODULE_INIT_STATUS` it will also log the initialization of the CC1101.

Anything interesting in the logs prior to the restart ?

tablatronix, 2023-08-16:
Seems like a memory issue https://github.com/NorthernMan54/rtl_433_ESP/issues/99

closed 2023-09-23 by github-actions[bot]