1technophile/OpenMQTTGateway #1721: Lilygo_rtl_433 compile not working

MiddleSiggy, 2023-07-21:
When I utilize the lilygo_rtl_433 precompiled firmware from the web site, my lilygo will work and receive data from 433Mhz items.

When I compile the default package from the web under option 4, and install it on the lilygo it will not receive any items.

To test I have 2 lilygo T3_v1.6.1 202110104 boards, and I installed the precompiled firmware on the first board, and the custom compiled version on the second and the second will not detect any 433 Mhz devices, while the 1st board will. 

They are sitting next to each other, so they should receive the same things.

When I compiled things, I tried the following:
- changed nothing, took the default settings
- Used this config: (some items left out because they were not changed)
  '-DZdisplaySSD1306="LilyGo_SSD1306"'
  '-DLOG_TO_OLED=false'         ; Enable log to OLED
  '-DJSON_TO_OLED=true'
  '-DDISPLAY_IDLE_LOGO=true'
  '-DDISPLAY_BRIGHTNESS=80'
  '-DDISPLAY_METRIC=false'
  '-DCC1101_FREQUENCY=433.92'

I tried many more, and have spent hours reading code.. what am I missing?

The custom compile does go on the board without issue, and it connects to my wifi and MQTT without issue.. but the RTL_433 never detects anything...

What can I check?

NorthernMan54, 2023-07-23:
Did you try building the lilygo-rtl_433 environment ?

ie

```
platformio run -e lilygo-rtl_433
```

MiddleSiggy, 2023-07-26:
> Did you try building the lilygo-rtl_433 environment ?
> 
> ie
> 
> ```
> platformio run -e lilygo-rtl_433
> ```

Yes I did do that. It still behaves different the the default binary.. :(

DigiH, 2023-07-26:
Hi @MiddleSiggy 

As the pre-compiled binary is actually a build of the **lilygo-rtl_433** environment, this is very strange. Also in your initial post you talk about 

> Used this config:
>             …
>             '-DCC1101_FREQUENCY=433.92'

Where did you get the CC1101 build_flag from? The LilyGo does not have a CC1101 radio, so this should not be in any LilyGo environment. Could you post the complete environment you are using?

Other than that, the only other thing I could think of would be a faulty board. Have you tried swapping your two LilyGo boards around, using the pre-compiled binary on this one, does it work then?

MiddleSiggy, 2023-08-02:
I did not try to swap them.. I will try that tomorrow and let you know thanks

1technophile, 2023-08-30:
Closing the issue, if it is still relevant feel free to reopen

closed 2023-08-30 by 1technophile