1technophile/OpenMQTTGateway #1729: M5Atom esp32-S3

JPFrancoia, 2023-08-08:
I have been using the M5Atom lite board for a couple of years with no problem. I wanted to extend my network so I bought two Atom S3 lite, which are basically almost the same as the M5 atom, except that they run on ESP32-S3 vs ESP32. This is the board: https://shop.m5stack.com/collections/m5-controllers/products/atoms3-lite-esp32s3-dev-kit. I tried flashing it like a M5 atom lite, something like this:

```shel
esptool.py --chip esp32-s3 --port /dev/ttyACM0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader_dio_80m.bin 0x10000 esp32-m5atom-lite-firmware.bin 0x8000 esp32-m5atom-lite-partitions.bin --force
```
But "it doesn't work", meaning when I plug the board, I see no LED turning on and the wifi network of the board doesn't turn on.

I half expected this result. Would you know how to get OpenMQTTGateway working on this board? If you could point me to the right direction, I'm not afraid to fiddle, compile, test and make a PR when I get it to work.

bruno-briner, 2024-03-19:
I would also love M5 Atom S3 support.

bruno-briner, 2024-12-31:
Is there any news for a support of the M5Stack AtomS3 Lite?

bruno-briner, 2025-01-22:
Answering my own question the image esp32s3-atomS3U works in the M5 Atom S3 Lite without any issue.