1technophile/OpenMQTTGateway #1734: NVS error and can't save settings

tiit12, 2023-08-18:
HI

I have ESP32 WT32-ETH01 module and struggle with latest firmware -- I can change the settings, but they are'nt saved. 
I see from logs:

begin(): nvs_open failed: NOT_FOUND
N: No WebUI config to load

I tried older version (1.5.6) at that one seems to work with NVS (no errors in log).

Thanks..

Tiit

1technophile, 2023-08-18:
Hello,

Did you try to erase the flash ? If yes with which tool?

With prior versions the webUI was not available so this is why you are not encountering the `N: No WebUI config to load`

tiit12, 2023-08-18:
I installed through web installer and checked Erase flash. Tried also with ESP tool and separate files -- same situation.

Tiit

1technophile, 2023-08-18:
Which environment are you using ?

tiit12, 2023-08-18:
Also from logs:

` N: [WebUI] MQTTtoSYS {"mqtt_server":"192.168.*.*","mqtt_port":"1883","mqtt_user":"****","mqtt_pass":"****","mqtt_secure":false,"gateway_name":"OMG_ESP32_WT32-ETH01","mqtt_topic":"home/"}
     [ 45310][E][vfs_api.cpp:24] open(): File system is not mounted
E: failed to open config file for writing
`

tiit12, 2023-08-18:
Windows 11 and browser is Edge. ESP tool is version 3.9.5.

1technophile, 2023-08-18:
By the way there is a template for bug report

tiit12, 2023-08-18:
A .. ok .. sorry for that :)

1technophile, 2023-08-18:
Could you share the platformio configuration or environment used ?

tiit12, 2023-08-18:
Actually I did it from your web page :

[image]

tiit12, 2023-08-18:
I did now fresh compile from gitpod -- same result:

[image]

`
N: OpenMQTTGateway URL: http://0.0.0.0/
N: ZwebUI setup                                           done
[  2950][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
N: SYS config not found
[  4464][E][Preferences.cpp:50] begin(): nvs_open failed:                                           NOT_FOUND
N: BT config not found
`

1technophile, 2023-08-18:
Do you have a link to your board ?

tiit12, 2023-08-18:
https://www.cnx-software.com/2020/11/09/6-wt32-eth01-is-a-tiny-esp32-board-with-ethernet/

It is described also on PlatformIO wiki:

https://docs.platformio.org/en/latest/boards/espressif32/wt32-eth01.html

I hope that helps.

tiit12, 2023-08-18:
I tried Tasmota on that board and Tasmota works (flash is detected and usable).

1technophile, 2023-08-18:
Which settings are you trying to change that are not saved ?

tiit12, 2023-08-18:
MQTT server settings actually, but also WIFI settings .. everything that needs save to NVS -- I see from logs, that filesystem is not mounted and settings can't save. After reboot is everything default.

closed 2023-09-26 by github-actions[bot]