1technophile/OpenMQTTGateway #1735: Disable HA discovery in pre-built images on boot

pyrodex, 2023-08-23:
**Is your feature request related to a problem? Please describe.**
I love the fact I can toggle on/off the HA discovery in OMG but it is super frustrating to go in and have to clean up MQTT after I decide to do some work or testing with boards on the pre-built images. The goal of the pre-built is to make user's lives simpler than compiling and removing flags. 

**Describe the solution you'd like**
Could we TURN off that functionality on boot-up and let people toggle it on/off at will?

**Describe alternatives you've considered**
The only alternative would be a white/black list for RF devices but that is hard.

**Additional context**
N/A

1technophile, 2023-08-23:
If HA discovery is turned off per default , HA users will not have the switch to control the discovery and will have to go through an MQTT client to turn it ON.
Note that with the latest development version the state of the discovery is saved in flash, so once it is turned OFF it will remain OFF

pyrodex, 2023-08-23:
If this new feature in the latest DEV does prevent on boot discovery if discovery was previously off, then I am fine with that since I keep it off and only toggle it when needed.

1technophile, 2023-08-23:
Yes it does exactly that.

1technophile, 2023-08-30:
Closing the issue, if it is still relevant feel free to reopen

closed 2023-08-30 by 1technophile