1technophile/OpenMQTTGateway #1738: v1.6.0 - RF compile error with Arduino IDE

lethuer, 2023-08-30:
I just downloaded the source code of v1.6.0 and wanted to compile it for RF gateway with Arduino IDE.
I'm trying to compile for ESP8266 (D1 Mini). and also downloaded required libraries from here: "[nodemcuv2-rf2-libraries.zip](https://github.com/1technophile/OpenMQTTGateway/releases/download/v1.6.0/nodemcuv2-rf2-libraries.zip)"
Afterwards I copied the files to "C:\Users\MyUsername\Documents\Arduino\libraries".

At startup Arduino IDE mentiones there are updates of some libraries.
Anyhow, doesn't matter if these updates are installed or not.
I get an error if I try to compile:
`error: declaration of template parameter 'T' shadows template parameter
  291 | void Config_update(JsonObject& data, const char* key, T& var);
      |           ^~~~~~~~`

`Compilation error: declaration of template parameter 'T' shadows template parameter`

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development) 
*      v1.6.0
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
*     ArduinoJson 6.18.3 (6.21.3 available)
*     ArduinoLog 1.0.3 (1.1.1 available)
*     esp8266_mdns 1.1.7 (1.1.8 available)
*     PubSubClient 2.8
*     rc-switch 2.6.2 (2.6.4 available)
*     WifiManager 2.0.5-beta (2.0.16-rc2 available)
* For IR and RF clarify if you tested with the basic examples given with these libraries
*     example: "ReceiveDemo_Simple.iso" from rc-switch library compiles without error

lethuer, 2023-08-30:
Also tried with 1.5.1 which has the same issue so I would expect this is a general setup issue ?!

1technophile, 2023-08-30:
I would advise using PlatformIO (it is not as difficult as it seems), Arduino IDE is more complex to maintain due to the fact that we can't explicitly state on a configuration file the platforms and the libraries versions used

lethuer, 2023-09-02:
I followed your advise, installed Mcrosoft Stduio Code with extension PlatformIO but again getting some error messages...
I'm not familiar with coding so maybe there are a few steps to get an arduino project running on platform io ?!
I created a project and added the libraries directly from there, so no copy pasting of libraries from here "[nodemcuv2-rf2-libraries.zip](https://github.com/1technophile/OpenMQTTGateway/releases/download/v1.6.0/nodemcuv2-rf2-libraries.zip)"

[image]

If I now open the project a file "platformio.ini" comes up.
If I try to build currently these problems are displayed:
[image]

1technophile, 2023-09-02:
You don't need to create a new project, you can download OpenMQTTGateway repository and open the folder with VSC and the pio extension. Also PIO will download automatically the appropriate libraries.
Instructions are below:
https://docs.openmqttgateway.com/upload/builds.html#configure-upload-with-platformio

lethuer, 2023-09-02:
Okay got it working now !
environment "rf-bridge" compiles successfully... thank you !

lethuer, 2023-09-02:
Okay got it working now and IR signals showing up in Homeasistant ! great so far ! :)

Now my question is why there aren't created new entities and devices ?
Only thing happening is that "gatewayIR" gives new states when pressing a button on my remote.
[image]

Auto discovery for new devices is activated
[image]

I can't find an option to enable discovery in mqtt configuration settings (configuration -> integration -> mqtt -> configuration), but I think it should be enabled per default...

1technophile, 2023-09-02:
You will have to define your sensors by yaml configuration as there are no auto discovery yet for infrared

closed 2023-10-18 by github-actions[bot]