1technophile/OpenMQTTGateway #1741: esp32dev-multi_receiver V1.6.0 does not work

cpramhofer, 2023-09-10:
Hi everybody,

i just tried to install the recent version esp32dev-multi_receiver (v.1.6.0)
i tried it directly from the web and a tried to compile it by myself - both ob windows and mac but always run into a boot loop with the following error:
ESP32 Dev / Guru Meditation Error: Core 0 panic’ed

for comparison i tried V 1.5.0 which went just fine.

BOARD LOLIN D32 PRO with CC1101 Chip

1technophile, 2023-09-13:
Investigating this, commenting ZgatewayPilight from the environment enable to have the multi-receiver running for information with RF, RF2 and RTL_433.

Now digging further

1technophile, 2023-09-13:
Linked with #1678

1technophile, 2023-09-13:
- Returning to RTL_433 v0.1.9 with the current development branch and some minor adaptations run well.
- RTL_433 v0.2.0 trigger the issue highlighted above
- Removing Nexus and Silvercrest protocol from Pilight and the latest development version works
protocol_header.h
```cpp
//#include "433.92/nexus.h"
//#include "433.92/silvercrest.h"
```

protocol_init.h
```cpp
//nexusInit();
//silvercrestInit();
```

This is due to a conflict related to these protocols. We don't need to have those duplicates. From this, the question is, are you using Pilight now that we have RTL_433?
Pilight could be removed from the multi-receiver setup or we could have 2 builds:
- RF RF2 PiLight
- RF RF2 RTL_433

NorthernMan54, 2023-09-15:
@1technophile The underlying issue here is a symbol conflict between PiLight and RTL_433_ESP with nexus and silvercrest signal decoders.

```
/Users//.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: warning: size of symbol `nexus' changed from 4 in .pio/build/esp32_cc1101/lib2e9/libESPiLight.a(protocol.c.o) to 112 in .pio/build/esp32_cc1101/lib395/librtl_433_ESP.a(nexus.c.o)
/Users//.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: warning: size of symbol `silvercrest' changed from 4 in .pio/build/esp32_cc1101/lib2e9/libESPiLight.a(protocol.c.o) to 112 in .pio/build/esp32_cc1101/lib395/librtl_433_ESP.a(silvercrest.c.o)
```

Any thoughts on how to simply limit the symbol scope to each library ?

NorthernMan54, 2023-09-27:
We are looking into this, and are able to recreate the issue.  Unfortunately resolving will likely require changing the upstream libraries, which will require some co-ordination.

closed 2023-12-27 by 1technophile