1technophile/OpenMQTTGateway #1744: Does not send Json messages correctly

AlonsoDeCastilla, 2023-09-14:
Hello.
I'm learning a lot thanks to this project!! 

I have two LILYGO LORA32 modules, in one I have loaded the latest version of OMG, LoRa is configured and it sends messages to an online MQTT broker. In the other a modified example of LoRa sending.
The transmission is done without problem, but it does not show me well the Json string that I sent. It shows " as \", and the broker cannot process it.

Send from node:
```c++
  StaticJsonDocument<200> doc;
  // Add an array.
  JsonArray deposito = doc.createNestedArray("deposito");
  deposito.add(temperature);
  deposito.add(distance); 
 
  LoRa.beginPacket();

  serializeJson(doc, LoRa);
  serializeJsonPretty(doc, Serial);
 
  LoRa.endPacket();

``` 
Console OMG:
[image]

Is it possible for OMG to send the Json message as it arrives?

Thank you so much.

1technophile, 2023-09-14:
Hello,

That's good synchronization. I have something for you to test below:
https://github.com/1technophile/OpenMQTTGateway/pull/1742

This enables to remove the `\` problem and merging the `message` content with the rest of the json,
there is also a sender [example ](https://github.com/1technophile/OpenMQTTGateway/tree/lora-improvements/examples/LoraTemperature) illustrating how the data should be formatted

nikito7, 2023-09-24:
This was working before:

https://github.com/nikito7/LoRa/blob/dev/nodes/LoRa-Sender-1/LoRa-Sender-1.ino

But need decoding like this:
```yaml
    - name: "LoRa ID7 Temp"
      unique_id: LoRa_ID7_Temp
      state_topic: "home/easyhan_lora_1/LORAtoMQTT"
      value_template: >
        {% set msg = value_json['message'] %}
        {% set id = (msg|from_json)['id'] %}
        {% set x = (msg|from_json)['tempc'] %}
        {% if id == "Node7" and x > -99 %}
          {{ x }}
        {% endif %}
      unit_of_measurement: "°C"
```

https://github.com/nikito7/LoRa/blob/dev/nodes/lora.yaml

1technophile, 2023-09-24:
@nikito7 you will not need this decoding and to add a sensor in yaml with the PR linked

closed 2024-01-07 by github-actions[bot]