1technophile/OpenMQTTGateway #1746: BUG: Long Passwords break connection function

neoatomic, 2023-09-15:
It seems that there is a bug if you use a long password. Specifically in the field "mqtt pass" of the webUI.

Description : 
I did setup Mosquitto on my OpenWRT device and created users with long 60 character passwords.
Tested the user with mqtt-explorer and connection was working correct. Than flashed the esp32dev-ble current build and setup everything in the webUI by copy/paste. But whatever i did i could not get the esp32 to connect. Wifi was working, but Mosquitto received no connection request or any entry's to the log. So presumed it was a vlan/routing issue at first. But finally gave creating a shorter password a try for the user (now of 22 characters) and all connected instantly.

So conclusion is that entering/storing a long password breaks the connection function.
Hopefully its an easy fix, or otherwise a warning can be added in the webUI not to exceed x amount of characters?

Also, Thank you to all who made this project possible.

NorthernMan54, 2023-09-16:
Looks like you hit the max 60

https://github.com/1technophile/OpenMQTTGateway/blob/bfca747d4bce38bad84be437cde79194a2b75f99/main/User_config.h#L153

59 Characters should have worked ( 59 characters plus the terminator )

neoatomic, 2023-09-16:
I need to buy a lottery ticket. ;) 

60 is an odd number for a limit? Seems MQTT brokers normally use a limit of 64. :

```
<username>: the user name to login to the MQTT broker, max length 64Bytes
<password>: the password to login to the MQTT broker, max length 64Bytes
```

NorthernMan54, 2023-09-17:
Looks like we need to make a change.  Bump it out to 64 plus 1

commit 8a124c88 (2023-09-24): [SYS] Increase mqtt username, base topic, otapassword and mqtt password max length (#1757)

fix #1746
closed 2023-09-24 by 1technophile