1technophile/OpenMQTTGateway #1747: Implement failure_number_ntwk for reconnects

tablatronix, 2023-09-19:
I noticed if wifi reconnect fails, nothing happens, we just get stuck here.
Reimplement counter and restart to avoid this.

Unrelated:
I am investigating the cause of my reconnects not working (ap_not_found) and think it is actually memory related (leak). There are also some ESP bugs that cause this in some IDF versions. But regardless of the reason, this would be a good failsafe to add.

1technophile, 2023-09-19:
There is already a counter used when having build time wifi credentials: `failure_number_ntwk`
we could use it when Wifi Manager is used

MichielioZ, 2023-10-14:
On a related note: on first boot, the AP requests wifi and mqtt configuration. As I just wanted to test and have no mqtt server setup at the moment, I only entered correct wifi credentials. This results in an infinite loop of trying to connect to an mqtt server and rebooting (if I didn't have the serial console open I wouldn't have noticed). Maybe store wifi credentials and mqtt credentials seperately ? I know with Wifi Manager it will return to AP mode if wifi fails, but in the case of wrong mqtt credentials I would suspect it still boots but just doesn't connect after some retries.

tablatronix, 2023-10-15:
Good point there is a seperate counter for wifi and mqtt so something could be done there.

i have been testing a fix for this for a while and it seems to be working well. Will pr it when i can test a few more issues

tablatronix, 2024-01-30:
I have not forgotten about this, I have been testing this and logging data. Unfortunately i have not figured out what causes this, but with the restart my devices stay online at least. Not sure its the best solution, but with no other ideas or info , I have no idea why reconnect just stops working. probably an sdk phy bug.

[image]

closed 2024-08-13 by github-actions[bot]