1technophile/OpenMQTTGateway #1755: Presence detection for Mi Band: wrong source_type attribute in Home Assistant

bcutter, 2023-09-22:
**Describe the bug**
In Home Assistant (HA Core 2023.3.6) the `device_tracker.xiaomi_smart_band_7` entity's attribute has `source_type: gps` even it should be `source_type: bluetooth_le`.

Interestingly, in MQTT it is correct in the device_tracker topic:
```
...
  "name": "MB/SW-tracker",
...
  "val_tpl": "{% if value_json.get('rssi') -%}home{%- else -%}not_home{%- endif %}",
  "source_type ": "bluetooth_le",
  "device": {
...
...
```

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a Xiaomi Smart Band 7
2. Activate `BT: Publish HASS presence` switch in HA
3. Wait to auto-discover the device
4. Check device_tracker entity's attribute

**Expected behavior**
Attribute `source_type` should be `bluetooth_le` in HA, like it is in MQTT.

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
* OpenMQTTGateway version used: v1.6.0 / development `a046c8`

**Additional context**
As it is fine in MQTT but not in HA I fear you might close this issue and redirect me to... well, where?

bcutter, 2023-12-22:
Not fixed. Will try to update HA to a newer version.

DigiH, 2024-01-07:
> Interestingly, in MQTT it is correct in the device_tracker topic:

Not quite ;P took some time to really look closely, but there is a trailing space in "source_type ", messing the correct recognition up :(

Fixed with
https://github.com/1technophile/OpenMQTTGateway/pull/1853

closed 2024-01-07 by DigiH

bcutter, 2024-01-07:
Oh great, so sometimes it's the little things right... 😄 
Can you expect when this fix will be shipped (in a stable release)?

DigiH, 2024-01-07:
As there are still some other issues which need addressing for the next release, a bit hard to say at the moment, but shouldn't bee too long.

bcutter, 2024-01-07:
Alright so definitely 2024 and probably until February (or even January)? :-)

bcutter, 2024-07-23:
Any update on this please? If I did not miss a release I tend to say we're still missing one, don't we? :-)