1technophile/OpenMQTTGateway #1771: disconnection of antennas on HA

jerome6994, 2023-10-10:
Good morning 1technophile

I wanted to know if what I'm encountering is normal or not!

I have several ESP32 antennas with OMG version 1.6.0 on them.
Everything works well, I have the presence of my nuts on HA. Of course it's a bit verbose and loads up the database but it works.

on the other hand what I encounter is that for classic node mcu after a certain time the entities of the device are no longer available at all and on M5 stack atom light the device is no longer there at all All !

node mcu :
[image]

M5 Stack :
[image]

For this to come back I have to restart the devices

However, I see the ping as present from my node MCU devices.
For the M5 stack I have to unplug them electrically then plug them back in!

thanks for your return

Jérôme

1technophile, 2023-10-14:
Hello Jerome,

We are working on a major rework of the publication mechanism, once done I will share it with you so that you can test if it fixes your issue

jerome6994, 2023-10-14:
Great
I love it every time I fall for a little thing in the project, I talk about it and a dev is already underway :)

1technophile, 2024-01-13:
This should have been fixed by v1.7.0

closed 2024-01-13 by 1technophile