1technophile/OpenMQTTGateway #1778: MQTT password limit to 60 characters ! and no option secure password with certificate

kwetnico, 2023-10-20:
Hi,

I am trying to configure my ESP (esp32 mini D1) which I flashed without problems with the esp32_ble firmware, however I encounter a problem connecting to my mosquitto server which tells me:
[image]

but looking at the MQTT configuration of my ESP I see that part of the password is missing to connect to the MQTT network...
I use the plugin under jeedom MQTT Manager and it generates passwords with a length of 64 characters. I realize that the OMG interface is apparently limited to 60 characters because it truncates the last letters for me, therefore the user/password is not valid and I am refused connection to the MQTT network, could you increase this limitation?

Another point I would have liked to get around this problem by using secure MQTT but it is impossible to indicate my key I do not have the field (tested under Firefox and Chrome from a Windows 10 PC):
[image]

Thanks

(sorry google translate)
OMG version 1.6

1technophile, 2023-10-20:
Hello,

The development version has fixed this and will be available in the next release. You can test it below:
https://docs.openmqttgateway.com/dev/

1technophile, 2023-10-22:
The certificates are available during onboarding if you reset/erase the esp32

kwetnico, 2023-10-30:
Hi,
I would like to know if the fix has been passed to master? Or should I keep the DEV version on my ESP32?
THANKS

PS: I have another problem, my esp32s take my IP reservations from DHCP whenever they want... is this known?

1technophile, 2023-10-30:
Hi,

> I would like to know if the fix has been passed to master? Or should I keep the DEV version on my ESP32?

It will be included in the next release, so you can keep the dev until

> PS: I have another problem, my esp32s take my IP reservations from DHCP whenever they want... is this known? Isn't possible to force adress IP interface in the ESP32 openmqttgateway?

This can be set in the code before building (not during runtime)

kwetnico, 2023-10-30:
> Hi,
>
>
> It will be included in the next release, so you can keep the dev until
>
>
> This can be set in the code before building (not during runtime)

Where specify adress ip ?

[image]

kwetnico, 2023-10-30:
Ok i seee with ide arduino..
Too complicated

kwetnico, 2023-11-01:
Bonjour, désolé en français on se comprendra mieux :)
Je me suis lancé sans connaitre avec réaliser ma configuration personnalisée avec PlatformIO sans connaitre j'ai réussi à créer mon fichier user en indiquant en dur mon IP.
[image]

[image]

Il refuse de prendre l'ip que je lui indique...

kwetnico, 2023-11-01:
hello I tried to build my configuration file but it does not work.
I have an esp32 wroom mini D1 compatible with esp32dev-ble.
I would like to force my IP address thanks for your help.

[image]

1technophile, 2023-11-01:
Hello,

You need to add also the following macro:
```
'-DNetworkAdvancedSetup=true'

DigiH, 2023-11-01:
Hi @kwetnico 

'-DESPWifiManualSetup=true'
is for defining the manual WiFi network conection with your
'-Dwifi_ssid="xxxx"'
'-Dwifi_password="xxxx"'

but for the manual IP definition you also need
'-DNetworkAdvancedSetup=true'
for your
'-DNET_IP="192.168.0.22"'
'-DNET_MASK="255.255.255.0"'
'-DNET_GW="192.168.0.1"'
'-DNET_DNS="192.168.0.1"'

kwetnico, 2023-11-01:
Thanks,

But the rest of the settings are good?
When I run it via the command it doesn't stop compiling...
"pio run --target upload"

Not easy when you discover visual studio
[image]

DigiH, 2023-11-01:
I assume you want to receive BLE device and have them decoded as well, correct?

For that you should have left everything from the original [env:esp32dev-ble] and just added your additional defines, i.e.

```
[env:esp32dev-ble]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.ble}
  ${libraries.decoder}
build_flags =
  ${com-esp32.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=2'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DGateway_Name="OMG_ESP32_BLE"' 
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="xxxx"'
  '-Dwifi_password="xxxx"'
  '-DNetworkAdvancedSetup=true'
  '-DNET_IP="192.168.0.22"'
  '-DNET_MASK="255.255.255.0"'
  '-DNET_GW="192.168.0.1"'
  '-DNET_DNS="192.168.0.1"'
  etc. …
```

of which you missed quite a few Bluetooth specific lib_deps and build_flags.

kwetnico, 2023-11-01:
> I assume you want to receive BLE device and have them decoded as well, correct?
> 
> For that you should have left everything from the original [env:esp32dev-ble] and just added your additional defines, i.e.
> 
> ```
> [env:esp32dev-ble]
> platform = ${com.esp32_platform}
> board = esp32dev
> board_build.partitions = min_spiffs.csv
> lib_deps =
>   ${com-esp32.lib_deps}
>   ${libraries.ble}
>   ${libraries.decoder}
> build_flags =
>   ${com-esp32.build_flags}
>   '-DZgatewayBT="BT"'
>   '-DLED_SEND_RECEIVE=2'
>   '-DLED_SEND_RECEIVE_ON=0'
>   '-DGateway_Name="OMG_ESP32_BLE"' 
>   '-DESPWifiManualSetup=true'
>   '-Dwifi_ssid="xxxx"'
>   '-Dwifi_password="xxxx"'
>   '-DNetworkAdvancedSetup=true'
>   '-DNET_IP="192.168.0.22"'
>   '-DNET_MASK="255.255.255.0"'
>   '-DNET_GW="192.168.0.1"'
>   '-DNET_DNS="192.168.0.1"'
>   etc. …
> ```
> 
> of which you missed quite a few Bluetooth specific lib_deps and build_flags.
> 
> I also assume you are making these changes to the existing environment definition in the file environments.ini, right? If so just use the platformio icon on the left (alien head) and expand `env:esp32dev-ble`, then select **Upload**
> 
> [image]
> If however you are defining this environment as an addition to the existing `env:esp32dev-ble`, you need to give it a new unique name, e.g. `env:esp32dev-ble-mine`, which you can then Build and Upload the same as shown above.
> 
> Actually it is the latter, as I can just about spot some .et_env.ini in your screenshot above ;) so you will have to finalise your environment with the missing lib_deps and build_flags AND give it a unique name.

I wan't just an ip adress static....  thanks for your help but it's too complicated I'm going to keep the web update method.

DigiH, 2023-11-01:
Giving up just when you were about to finalise it :(

You got everything right apart from missing the original BLE gateway specific lib_deps and build_flags, and giving your personal environment a new unique name, otherwise the build process didn't know which one of the two identically named `env:esp32dev-ble` environments it really should build and upload, which gave you the 

> doesn't stop compiling...

kwetnico, 2023-11-01:
> Giving up just when you were about to finalise it :(
>
> You got everything right apart from missing the original BLE gateway specific lib_deps and build_flags, and giving your personal environment a [...]

it's good ?

[image]

DigiH, 2023-11-01:
I can't see the very end of your environment, but I would say - Yes, looks perfect :)

Does it build and upload work fine when you select Upload on the left?

kwetnico, 2023-11-01:
> I can't see the very end of your environment, but I would say - Yes, looks perfect :)
> 
> Does it build and upload work fine when you select Upload on the left?

I don't understand it compiles and writes for all esp32-dev models

DigiH, 2023-11-01:
> My environnent :

Yes, and to only build and upload your environment, click on Upload on your expanded environment on the left.

[image]

kwetnico, 2023-11-01:
> > My environnent :
> 
> Yes, and to only build and upload your environment, click on Upload on your expanded environment on the left.
> 
> [image]
> 
> Then it will only build your environment and upload it to the ESP32 connected.

Yes ok but i have an error : 
[image]

DigiH, 2023-11-01:
> Yes ok but i have an error :

Ok, you do not have git installed on your computer, which is required for PlatformIO to fetch libraries, specifically the BLE Decoder for your environment. Some unforeseen missing element :(

Do what the prompt tells you - go to 

https://git-scm.com/downloads

and download the installer relevant for your computer OS and install git with it. Then  you are definitely good to go ;)

kwetnico, 2023-11-01:
> > Yes ok but i have an error :
> 
> Ok, you do not have git installed on your computer, which is required for PlatformIO to fetch libraries, specifically the BLE Decoder and others for your environment. Some unforeseen missing element :(
> 
> Do what the prompt tells you - go to
> 
> https://git-scm.com/downloads
> 
> and download the installer relevant for your computer OS and install git with it. Then you are definitely good to go ;)

I had installed it and I had just deleted it I thought it was mandatory with the ardino IDE application.

DigiH, 2023-11-01:
It is required for any remote library to be fetched for building, no matter for the Arduino IDE, PlatformIO … manually checking out git repos …

Having a GitHub account here it is a good idea to keep git installed on your computer ;)

kwetnico, 2023-11-01:
> It is required for any remote library to be fetched for building, no matter for the Arduino IDE, PlatformIO … manually checking out git repos …
> 
> Having a GitHub account here it is also a good idea to keep git installed on your computer ;)

thank you for your help I finally succeeded (not easy):
[image]

DigiH, 2023-11-01:
> thank you for your help I finally succeeded

You're welcome :)

> (not easy)

True, but now you know the basics of creating and uploading custom environments, so any other future environments you might want to create are all following these steps, which you now know.

closed 2023-11-01 by kwetnico