1technophile/OpenMQTTGateway #1780: esp32dev-multi_receiver no creating initial ap wifi

jbrandek, 2023-10-22:
I tried with other configs but when i try esp32dev-multi_receiver it does not create the initial wifi credentials to connect it to the networks, tried in others esp32 devices and they do the same

azzajess, 2023-10-23:
I thought i was going crazy lol. I am experiencing the same thing.
Its not broadcasting any wifi name etc.

warthog9, 2023-10-28:
Came to see if a bug pre-existed on this.  I'm seeing something similar, esp32dev board (specifically the ESP32 DEVKITv1) uploading the multi_reciever bootloader, firmware, and partition table along with the generic boot_app0.bin all via esptool (latest installed in a venv from pip) on a Linux box on two different esp32s it ends up in a crash loop (as seen on the console port) complaining about hash sums not matching.

esptool *DID* complain about flash_mode dout and some hash some so I've also confirmed the following by following esptool's vague suggestion about flash_mode keep instead.

```
ELF file SHA256: 4e2584206ae5a1c6

E (727) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x4012c2ac  PS      : 0x00060930  A0      : 0x80132c50  A1      : 0x3ffe3b40
A2      : 0x3f41bd88  A3      : 0x3f423ffc  A4      : 0x0000000c  A5      : 0x00060f23
A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x8012c2ac  A9      : 0x3ffe3b20
A10     : 0x3ffbab44  A11     : 0xfffc1fff  A12     : 0x00060f20  A13     : 0x00060f23
A14     : 0x007bed94  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f41bd88  LBEG    : 0x4008aa1d  LEND    : 0x4008aa2d  LCOUNT  : 0xfffffffe

Backtrace: 0x4012c2a9:0x3ffe3b40 0x40132c4d:0x3ffe3b60 0x4012c25c:0x3ffe3b80 0x4012a8f2:0x3ffe3ba0 0x4012acbd:0x3ffe3bc0 0x400e302d:0x3ffe3be0 0x4014a787:0x3ffe3c10 0x40083206:0x3ffe3c40 0x40079266:0x3ffe3c90 |<-CORRUPTED
```

I may try backing down the version a couple, or compiling it myself but I suspect the pre-built binaries are actively not working right now.

warthog9, 2023-10-28:
Also likely a duplicate of #1741

1technophile, 2024-01-27:
This is fixed in the [v1.7.0](https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.7.0)

closed 2024-01-27 by 1technophile