1technophile/OpenMQTTGateway #1782: Problem compiling NodeMCU with LORA

nicolasdelta, 2023-10-26:
Hello,

I'm trying to build with platformio from the [env:nodemcuv2-all-test] environnement with adding LORA function.

I've just added the following : 
   lib_deps =
     ${libraries.lora}
   build_flags =
       '-DZgatewayLORA="LORA"'

It doesn"t work and i get the following 

D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:47:1: error: 'SemaphoreHandle_t' does not name a type
   47 | SemaphoreHandle_t semaphorecreateOrUpdateDeviceLORA;
      | ^~~~~~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino: In function 'void createOrUpdateDeviceLORA(const char*, const char*, uint8_t)':
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:77:22: error: 'semaphorecreateOrUpdateDeviceLORA' was not declared in this scope; did you mean 'createOrUpdateDeviceLORA'?
   77 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(30000)) == pdFALSE) {
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                      createOrUpdateDeviceLORA
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:77:57: error: 'pdMS_TO_TICKS' was not declared in this scope
   77 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(30000)) == pdFALSE) {
      |                                                         ^~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:77:7: error: 'xSemaphoreTake' was not declared in this scope
   77 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(30000)) == pdFALSE) {
      |       ^~~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:77:82: error: 'pdFALSE' was not declared in this scope
   77 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(30000)) == pdFALSE) {
      |                                                                                  ^~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:104:18: error: 'semaphorecreateOrUpdateDeviceLORA' was not declared in this scope; did you mean 'createOrUpdateDeviceLORA'?
  104 |   xSemaphoreGive(semaphorecreateOrUpdateDeviceLORA);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  createOrUpdateDeviceLORA
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:104:3: error: 'xSemaphoreGive' was not declared in this scope
  104 |   xSemaphoreGive(semaphorecreateOrUpdateDeviceLORA);
      |   ^~~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino: In function 'void launchLORADiscovery(bool)':
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:112:22: error: 'semaphorecreateOrUpdateDeviceLORA' was not declared in this scope; did you mean 'createOrUpdateDeviceLORA'?
  112 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(1000)) == pdFALSE) {
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                      createOrUpdateDeviceLORA
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:112:57: error: 'pdMS_TO_TICKS' was not declared in this scope
  112 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(1000)) == pdFALSE) {
      |                                                         ^~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:112:7: error: 'xSemaphoreTake' was not declared in this scope
  112 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(1000)) == pdFALSE) {
      |       ^~~~~~~~~~~~~~
Compiling .pio\build\nodemcuv2-all-test\libd38\ESP8266mDNS\LEAmDNS_Structs.cpp.o
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:112:81: error: 'pdFALSE' was not declared in this scope
  112 |   if (xSemaphoreTake(semaphorecreateOrUpdateDeviceLORA, pdMS_TO_TICKS(1000)) == pdFALSE) {
      |                                                                                 ^~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:118:18: error: 'semaphorecreateOrUpdateDeviceLORA' was not declared in this scope; did you mean 'createOrUpdateDeviceLORA'?
  118 |   xSemaphoreGive(semaphorecreateOrUpdateDeviceLORA);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  createOrUpdateDeviceLORA
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:118:3: error: 'xSemaphoreGive' was not declared in this scope
  118 |   xSemaphoreGive(semaphorecreateOrUpdateDeviceLORA);
      |   ^~~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino: In function 'void LORAConfig_load()':
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:284:3: error: 'preferences' was not declared in this scope
  284 |   preferences.begin(Gateway_Short_Name, true);
      |   ^~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino: In function 'void LORAConfig_fromJson(ArduinoJson::JsonObject&)':
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:339:5: error: 'preferences' was not declared in this scope
  339 |     preferences.begin(Gateway_Short_Name, false);
      |     ^~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:357:5: error: 'preferences' was not declared in this scope
  357 |     preferences.begin(Gateway_Short_Name, false);
      |     ^~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino: In function 'void setupLORA()':
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:368:3: error: 'semaphorecreateOrUpdateDeviceLORA' was not declared in this scope; did you mean 'createOrUpdateDeviceLORA'?
  368 |   semaphorecreateOrUpdateDeviceLORA = xSemaphoreCreateBinary();
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   createOrUpdateDeviceLORA
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:368:39: error: 'xSemaphoreCreateBinary' was not declared in this scope
  368 |   semaphorecreateOrUpdateDeviceLORA = xSemaphoreCreateBinary();
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
D:/nicol/Documents/PlatformIO/Projects/OpenMQTTGateway-development/main/ZgatewayLORA.ino:369:3: error: 'xSemaphoreGive' was not declared in this scope
  369 |   xSemaphoreGive(semaphorecreateOrUpdateDeviceLORA);

I've seen that LORA supports ESP8266 https://github.com/1technophile/OpenMQTTGateway/pull/1711#issue-1795648550

Please can you tell me if i did forgot sometinh in the configuration ?

Regards
Nicolas

nicolasdelta, 2023-10-31:
SOLVED by doing clean reinstall of Vscode.
AND modifying SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS); to SPI.begin();
[image]

nikito7, 2023-11-12:
build_flags

```
  '-DLORA_BAND=868E6'
  '-DLORA_SCK=14'
  '-DLORA_MISO=12'
  '-DLORA_MOSI=13'
  '-DLORA_SS=15'
  '-DLORA_RST=16'
  '-DLORA_DI0=5'
```

nikito7, 2023-11-12:
One issue is Discovery

Another is Preferences.h

closed 2024-05-29 by github-actions[bot]