1technophile/OpenMQTTGateway #1791: Allow programatic BLE connect/disconnect

ov1d1u, 2023-11-02:
**Is your feature request related to a problem? Please describe.**
Some devices, like [Qingping CGD1](https://www.aliexpress.com/item/1005002907335593.html) require a sequence of data to be written on a characteristic. For example, to change the time this clock needs writing an auth key on a specific characteristic and then a command on the same characteristic. Writing the auth key, disconnecting and then writing the command will not work.

**Describe the solution you'd like**
To prevent breaking the current implementation, maybe a map of **[MAC : flag]** to enable/disable automatic connection management for each device should be kept. Aside from that, we will need two new endpoints to the `/commands/MQTTtoBT/config` to connect/disconnect from a specific device

**Describe alternatives you've considered**
Right now I'm using an ESPHome configuration to control this clock, but there is no way in ESPHome to read a characteristic by request.

h2zero, 2023-11-03:
Sounds like a write_multiple command would need to be created for this situation.

ov1d1u, 2023-11-03:
I don't think this is the ideal approach, as sometimes you might need to write - read - write something in the same session.

1technophile, 2023-11-03:
We should define something that could be used by other gateways like Theengs. We could have a json with a list commands for a specific mac:
1. Connect
2. Service
3. Characteristic write
4. Command
5. Read
6. Disconnect
So that this json could be reused by any gateway.
If this match your approach what would be this json for this device?

ov1d1u, 2023-11-03:
Specifically for this device something like sending a list of payload would work, e.g.:

```
[
  {
    "ble_write_address": "58:2D:34:55:8A:87",
    "ble_write_service": "22210000-554a-4546-5542-46534450464d",
    "ble_write_char": "00000001-0000-1000-8000-00805f9b34fb",
    "ble_write_value": "1101ea600e964287ea7d17894900da6174bd",
    "value_type": "HEX",
    "ttl": 4,
    "immediate": true
  },
  {
    "ble_write_address": "11:22:33:44:55:66",
    "ble_write_service": "12345678-9012-3456-7890-123456789012",
    "ble_write_char": "12345678-1234-5678-9012-123456789012",
    "ble_write_value": "0509b9414265",
    "value_type": "HEX",
    "ttl": 4,
    "immediate": true
  }
]
```

although it might not be ideal for other purposes, like when you need to read something from a characteristic between two writes.

closed 2024-02-17 by github-actions[bot]