1technophile/OpenMQTTGateway #1792: The implementation of the voltage object for BM6 battery monitoring is required

phoenixckua, 2023-11-06:
The implementation of the voltage object for BM6 battery monitoring is required . Voltage object not exists in MQTT message. 
50547B24F7B0 = {“id”:“50:54:7B:24:F7:B0”,“mac_type”:0,“adv_type”:0,“name”:“BM6”,“manufacturerdata”:“4c0002153ba29cd9a42c894856badaf2606ef777114d0000cd”,“rssi”:-55,“txpower”:0,“brand”:“GENERIC”,“model”:“BM6 Battery Monitor”,“model_id”:“BM6”,“type”:“BATT”,“track”:true,“batt”:77,“device”:“BM6 Tracker”}

DigiH, 2023-11-06:
Hi @phoenixckua 

As a quick and dirty test, can you try with test build SHA:10d7f7 at

https://docs.openmqttgateway.com/dev/upload/web-install.html

Keep in mind that the voltage might only appear in longer intervals, set by BT: Connect interval in your gateway settings.

Let us know what your results are.

phoenixckua, 2023-11-06:
BT: Connect interval is set 100m. I installed test update as update in Home Assistant and reboot HA, but problem not solved.

phoenixckua, 2023-11-06:
50547B24F7B0 = {"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"4c0002153ba29cd9a42c894856badaf2606ef777114d0000cd","rssi":-49,"txpower":0,"brand":"GENERIC","model":"BM6 Battery Monitor","model_id":"BM6","type":"BATT","track":true,"batt":77,"device":"BM6 Tracker"}

phoenixckua, 2023-11-06:
Ok, I try upload build manually to m5stack atom lite

DigiH, 2023-11-06:
> BT: Connect interval is set 100m.

That would mean that the connection is only every 100 minutes, so the voltage might only be fetched every 100 minutes.

Try setting it lower, to 1 min or so, otherwise you will have to wait 100 minutes before you can say if it might work or not.

As I said though, this is only a quick and dirty test. If it doesn't work more investigation with your help will be required to see if and how the BM6 voltage can be obtained.

phoenixckua, 2023-11-06:
esp32-atom-lite (build SHA:10d7f7)
BT: Connect interval 1m
results:

50547B24F7B0 = {"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"4c0002153ba29cd9a42c894856badaf2606ef777114d0000cd","rssi":-55,"txpower":0,"brand":"GENERIC","model":"BM6 Battery Monitor","model_id":"BM6","type":"BATT","track":true,"batt":77,"device":"BM6 Tracker"}

1technophile, 2023-11-06:
The voltage message would be a different message with just the `id`, `rssi` and `volt` fields

phoenixckua, 2023-11-06:
esp32-atom-lite (build SHA:10d7f7)
BT: Connect interval 5m
results:

I reciving two messages one after another for 50547B24F7B0. 
And that's it.

50547B24F7B0 = {"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"4c0002153ba29cd9a42c894856badaf2606ef777114c0000cd","rssi":-57,"txpower":0,"brand":"GENERIC","model":"BM6 Battery Monitor","model_id":"BM6","type":"BATT","track":true,"batt":76,"device":"BM6 Tracker"}

0547B24F7B0 = {"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"ee017627cd6db2df636b4f31f0fe5d0f","rssi":-57,"txpower":0}

1technophile, 2023-11-07:
Wait at least 10 minutes, and check the message history. BLE connection may fail sometime so it may not be every minutes that you get the voltage.

phoenixckua, 2023-11-07:
New update available . Can update using web interface HA ? 
[image]

phoenixckua, 2023-11-07:
{
  "id": "50:54:7B:24:F7:B0",
  "mac_type": 0,
  "adv_type": 0,
  "name": "BM6",
  "manufacturerdata": "4c0002153ba29cd9a42c894856badaf2606ef777114c0000cd",
  "rssi": -57,
  "txpower": 0,
  "brand": "GENERIC",
  "model": "BM6 Battery Monitor",
  "model_id": "BM6",
  "type": "BATT",
  "track": true,
  "batt": 76,
  "device": "BM6 Tracker"
  "manufacturerdata": "ee017627cd6db2df636b4f31f0fe5d0f",
  "rssi": -58,
  "txpower": 0
}
Comparing with previous message: + 3 lines, - 10 lines

phoenixckua, 2023-11-07:
{
  "id": "50:54:7B:24:F7:B0",
  "mac_type": 0,
  "adv_type": 0,
  "name": "BM6",
  "manufacturerdata": "ee017627cd6db2df636b4f31f0fe5d0f",
  "manufacturerdata": "4c0002153ba29cd9a42c894856badaf2606ef777114d0000cd",
  "rssi": -57,
  "txpower": 0
  "txpower": 0,
  "brand": "GENERIC",
  "model": "BM6 Battery Monitor",
  "model_id": "BM6",
  "type": "BATT",
  "track": true,
  "batt": 77,
  "device": "BM6 Tracker"
}
Comparing with selected message: + 9 lines, - 2 lines
07.11.2023 10:57:50 
{"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"4c0002153ba29cd9a42c894856badaf2606ef777114d0000cd","rssi":-57,"txpower":0,"brand":"GENERIC","model":"BM6 Battery Monitor","model_id":"BM6","type":"BATT","track":true,"batt":77,"device":"BM6 Tracker"}

phoenixckua, 2023-11-07:
[image]

DigiH, 2023-11-07:
> Can update using web interface HA ?

The new development build SHA:ca9ef6 does not include the voltage test changes, so it will only ever include the battery level from the newly created decoder.

As the voltage test with test build SHA:10d7f7 really only was a quick and dirty test scenario, which apparently didn't produce the desired results as you have shown with your testing, getting the voltage from BM6 devices will require more investigation, likely also requiring your help with some more nRF Connect service/Chars information.

phoenixckua, 2023-11-07:
Logs:
https://drive.google.com/file/d/13WF-4LG7dWCBUnmHXY9505GfoNLR4ZVu/view?usp=drivesdk
https://drive.google.com/file/d/13d196qOUsey03PF93y4lk6iEJ5SEVeLj/view?usp=drivesdk

phoenixckua, 2023-11-08:
UPD:
2 device updating soc information correctly, but 50547B243010 have a static data and not updating, and this one not have object tracker in MQTT integration menu.
50547B243010 = {“id”:“50:54:7B:24:30:10”,“mac_type”:0,“adv_type”:0,“name”:“BM6”,“manufacturerdata”:“eeaee3c4f8988453578074e7f1677430”,“rssi”:-59,“txpower”:0}
50547B243010 = {“id”:“50:54:7B:24:30:10”,“mac_type”:0,“adv_type”:0,“name”:“BM6”,“manufacturerdata”:“4c0002153ba29cd9a42c894856badaf2606ef777114d0000cd”,“rssi”:-61,“txpower”:0,“brand”:“GENERIC”,“model”:“BM6 Battery Monitor”,“model_id”:“BM6”,“type”:“BATT”,“track”:true,“batt”:77,“device”:“BM6 Tracker”}

What needs to be done to update the data on 50547B243010?

DigiH, 2023-11-08:
Sorry, not quite sure what you mean by 50547B243010 having static data and not bein recognised as a tracker? 

The second message seems to have the battery level as well as the “track”:true property

> 50547B243010 = [...]

phoenixckua, 2023-11-08:
I get messages on all 3 sensors.  two sensors in the mqtt integration display more entities than the third.  when i open the essence of soc 50547b243010, i see from the history that it is updated only once, when m5stack atom lite is restarted, then the graph goes along a straight line and says that it was updated 9 hours ago for the last time. 
https://drive.google.com/file/d/14Li9FPzi4C7KTLZoNbZr85LmrzLPVDw3/view?usp=drivesdk
https://drive.google.com/file/d/140oqfjwE4xwR1zZMnMYniTN90EQ8jcSm/view?usp=drivesdk
https://drive.google.com/file/d/1430WaY2QjMOWE0NjSxeJi5rzGzYMx8JR/view?usp=drivesdk
https://drive.google.com/file/d/149Vt4k_mdTrKPAoyKuwZ2YuHZSL1hv_R/view?usp=drivesdk

DigiH, 2023-11-08:
Can you paste the images here?

DigiH, 2023-11-09:
The "50:54:7B:24:30:10" seems to mainly send non-iBeacon manufacturerdata (not starting with 4c00). 
Only the data starting with 4c00 can be decoded to get the battery level, which, from what you have been saying above, seems to be fine for the other two BM6, but not this one.

Basically every iBeacon format broadcast visible in MQTT Explorer can and will be decoded, any other manufacturerdata broadcasts cannot be decoded.

Can you have a look in the battery app to see if there are any settings about the iBeacon format advertising, like the other two might have been set to?

Or it could actually be a slightly different model to the other two, only sending iBeacon formats sporadically, something not unheard of with Chinese unbranded devices being sold under the same name, but having actual different technical specifications.

ReconOperator, 2024-01-07:
Hi @DigiH,

will try to post some messages from MQTT Explorer. BTW, there is also a temperature value in the app. Maybe we can try to get that one also?

`{
  "id": "38:3B:26:B5:30:AB",
  "mac_type": 0,
  "adv_type": 0,
  "name": "Battery Guard",
  "manufacturerdata": "4c0002153ba29cd9a42c894856badaf2606ef77711550000cd",
  "rssi": -88,
  "txpower": 0,
  "brand": "GENERIC",
  "model": "BM6 Battery Monitor",
  "model_id": "BM6",
  "type": "BATT",
  "track": true,
  "batt": 85,
  "device": "BM6 Tracker"
}`

`{
  "stat_t": "+/+/BTtoMQTT/383B26B530AB",
  "dev_cla": "battery",
  "unit_of_meas": "%",
  "name": "BM6-batt",
  "uniq_id": "383B26B530AB-batt",
  "val_tpl": "{{ value_json.batt | is_defined }}",
  "state_class": "measurement",
  "device": {
    "ids": [
      "383B26B530AB"
    ],
    "cns": [
      [
        "mac",
        "383B26B530AB"
      ]
    ],
    "mf": "GENERIC",
    "mdl": "BM6",
    "name": "BM6 Battery Monitor-B530AB",
    "via_device": "OMG_ESP32_OLM_GTWE"
  }
}`

`{
  "stat_t": "+/+/BTtoMQTT/383B26B530AB",
  "name": "BM6-device",
  "uniq_id": "383B26B530AB-device",
  "val_tpl": "{{ value_json.device | is_defined }}",
  "device": {
    "ids": [
      "383B26B530AB"
    ],
    "cns": [
      [
        "mac",
        "383B26B530AB"
      ]
    ],
    "mf": "GENERIC",
    "mdl": "BM6",
    "name": "BM6 Battery Monitor-B530AB",
    "via_device": "OMG_ESP32_OLM_GTWE"
  }
}`

DigiH, 2024-01-07:
Hi @ReconOperator 

Thanks

The discovery messages are not important right now, the ones under the homeassistant topic, starting with `{ "stat_t": "+/+/BTtoMQTT/…`

For the actual device messages under the gateway topic - BTtoMQTT sub topic, have a look under the History in MQTT Explorer, and past the latest one with the manufacturerdata starting with `"4c00…`, but there should also be a recent one with the manufacturerdata starting with something else.

If you can paste these two messages, along with the actual battery level in %, the voltage, and the temperature as shown in the app at the same time these messages were received , we should be able to deduct if any further values in addition to the battery level are encoded in these advertising data.

If not tough, and if it does require a connection someone else might have to take over.

ReconOperator, 2024-01-08:
OK lets see if this one tells you anything then

First 11:29:06
`{
  "id": "38:3B:26:B5:30:AB",
  "mac_type": 0,
  "adv_type": 0,
  "name": "Battery Guard",
  "manufacturerdata": "47f51521f734b795c7a1f44aebbc6eae",
  "rssi": -90,
  "txpower": 0
}`

Second 11:30:31
`{
  "id": "38:3B:26:B5:30:AB",
  "mac_type": 0,
  "adv_type": 0,
  "name": "Battery Guard",
  "manufacturerdata": "4c0002153ba29cd9a42c894856badaf2606ef77712280000cd",
  "rssi": -80,
  "txpower": 0,
  "brand": "GENERIC",
  "model": "BM6 Battery Monitor",
  "model_id": "BM6",
  "type": "BATT",
  "track": true,
  "batt": 40,
  "device": "BM6 Tracker"
}`

[image]
[image]
[image]

DigiH, 2024-01-08:
Thanks, could it be that the MQTT reading was slightly later than the app reading, as I can see another dip going down in the green chart after 11:29 where it says 100%, with the decoded battery level being 40%, close to the dip after the reading?

Unfortunately for the voltage it might be a similar issue, with another dip going to around 12.2ish V

Unfortunately this makes it hard to try and see if there is any correlation between the braodcast advertising data and the actual values presented in the app.

I assume the dips in battery level and voltage are due to the vehicle being used, and possibly also with the cold spell you also seem to be experiencing at your location today.

Any chance of the same kind of readings as above, but possibly during a time when the battery level, voltage and temperature are more stable, without any abrupt dips?

I know, sometimes it's not that easy to catch all the data together when doing reverse engineering ;)

DigiH, 2024-01-09:
Thanks @ReconOperator, that was exactly what I needed to see if the voltage might be encoded in the remaining advertisement data, but unfortunately I couldn't find anything, so the BM6 voltage will likely require the same connection retrieval as the BM2 does.

Someone else will have to jump in and try and get the correct service/char with you. Do you have nRF Connect installed on your mobile phone?

ReconOperator, 2024-01-09:
> Do you have nRF Connect installed on your mobile phone?

Hi @DigiH 
Yes i do now =) On an Android device

DigiH, 2024-01-09:
Just as a preparation, could you post some screenshots of the different services and characteristics showing for the BM6 when connecting to it with nRF Connect?

ReconOperator, 2024-01-09:
@DigiH 

Hope this helps 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ReconOperator, 2024-01-13:
@DigiH any progress?

DigiH, 2024-01-14:
Not from my side unfortunately, as I haven't implemented getting the voltage for the BM2 either. 

And without anyone of us having a BM6 or finding any information of where the BM6 might have its voltage information encoded it is nit that easy.

One thing you could do now, to get you into the reverse engineering part of it all 😉 , is to look into any of the unnamed Service/Chars above which have the READ property, to see what you get for each, and see if any of them would make sense for a multiple of the then current voltage state. E.g. 1253 when the voltage is 12.53V

You can test this procedure with a known Service/Char like the Device Name.

superfrink, 2024-01-24:
Possibly the data is encrypted as described at https://doubleagent.net/hardware/ble/bluetooth/2023/05/23/a-car-battery-monitor-tracking-your-location-part3

DigiH, 2024-01-27:
Thanks @superfrink - this is very likely the case, but without knowing any further details about the BM6, as are available for the BM2, for which we already do this with a connection retrieval, it is hard to make the voltage also available for the BM6.

So someone with a BM6 might actually follow these instructions to see if the same is applicable to the BM6, after which we could create a similar implementation.

closed 2024-05-11 by github-actions[bot]

JeffWDH, 2024-08-13:
The BM6 uses a slightly different encryption key than the BM2. Details (and the key) can be found here: https://www.tarball.ca/posts/reverse-engineering-the-bm6-ble-battery-monitor/

ReconOperator, 2024-08-13:
@DigiH seems like this from JeffWDH could be useful